realsenseai/librealsense #6338: Ground Plane detection (Python)

sivas001, 2020-05-02:
Is there any open-source code available for ground plane detection using 3D point cloud?

MartyG-RealSense, 2020-05-02:
I researched your request extensively, and the most promising option I could find was **pydriver**, which can detect / remove the ground plane.  It is distributed under the MIT licence, meaning that it can be freely used and you just have to include an MIT Licence copyright message.

https://github.com/lpltk/pydriver

sivas001, 2020-05-03:
Thank you for getting back to me. I was looking for a ground plane recognition code in python that i could use to expand upon. The pydriver is 5 years old and am unable to execute it with my current pcl library package.

MartyG-RealSense, 2020-05-03:
Would this program that works with Python 3 be any use?

https://github.com/NVlabs/planercnn

sivas001, 2020-05-04:
no, unfortunately not, I was hoping the code will be using 3d point cloud with the RANSAC algorithm.

MartyG-RealSense, 2020-05-04:
Pre-made ground plane detection solutions for Python that have code, are usable with PCL and are newer than 3 years old are highly difficult to find, unfortunately.

There is a class in PCL called **SACSegmentation** that some have used for ground plane detection.

https://pointcloudlibrary.github.io/documentation/classpcl_1_1_s_a_c_segmentation.html

Here's an example use:

https://groups.google.com/d/topic/rgbdemo/Pg9xFW7EQtY

ev-mp, 2020-05-04:
@sivas001 , in case you're trying to identify and establish the plane from 3d Points then can try the [Plane-Fit algorithm](http://www.ilikebigbits.com/blog/2015/3/2/plane-from-points) utilized in Librealsense'.  
It is being integrated into the [`rs-depth-quality` tool](https://github.com/IntelRealSense/librealsense/blob/d39b5139ce62efffddee4cd11a87e8927ecdfa0c/tools/depth-quality/depth-metrics.h#L62-L111) to provide a baseline for the depth characteristic metrics and when you run the tool you would see it as a yellow grid in the center of FOV.

MartyG-RealSense, 2020-05-17:
@sivas001    Do you require further assistance with this question please, or is it okay to close this case?   Thanks!

sivas001, 2020-05-23:
Yes I think that is all. Thank you for your assistance. I have not been able to find something useful in the links provided but appreciate you looking.

closed 2020-05-23 by sivas001

MartyG-RealSense, 2020-05-23:
Thanks so much for the update!

netanelfar, 2020-09-07:
@saran230420 were you able to detect the ground? i am trying to get a similar if not the exact algorithem you were looking for