realsenseai/librealsense #6341: rs2.config.enable_stream(): some function signatures don't work

jurc192, 2020-05-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (D435i) | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version | Arch Linux | 
| Kernel Version (Linux Only)    |  5.6.7-arch1-1 | 
| Platform                                 | PC  |
| SDK Version                            |  2.3.3.1 | 
| Language                            |  python                        | 
| Segment			|  others | 

### Issue Description
Enabling streams using different [function signatures](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.config.html?highlight=enable_stream#pyrealsense2.config.enable_stream) (1-5) gives different results in `config.resolve(pipeline)`. Some function signatures (1,2,4) fail. Signature 3 fails only on USB2. Tested all variants of following code on both USB2 and USB3 ports.

### Minimal example
```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config   = rs.config()
# config.enable_stream(rs.stream.depth, 1, 848, 480, rs.format.z16, 15) # can't resolve
# config.enable_stream(rs.stream.depth, 1)                              # can't resolve
# config.enable_stream(rs.stream.depth, rs.format.z16, 15)              # works
# config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 15)    # USB2 fails, USB3 works
# config.enable_stream(rs.stream.depth, 1, rs.format.z16, 15)           # can't resolve

# config.enable_stream(rs.stream.depth, 848, 480)                       # works
# config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16)        # can't resolve
profile = config.resolve(pipeline)
```

It looks like a bug, or am I doing something wrong?

MartyG-RealSense, 2020-05-03:
Because the SDK supports 6, 15, 30, 60 or 90 as its most commonly used FPS modes, setting a stream to 1 FPS would not be supported:

**config.enable_stream(rs.stream.depth, 1)**

Putting a '1' after rs.stream.depth should not be necessary, as an index number like this is used for distinguishing streams of the same type, such as the left infrared (0) and right infrared (1).  There is only one depth stream, so an index number is not needed.

**config.enable_stream(rs.stream.depth, 1, rs.format.z16, 15)**

In the line below that does not resolve, please put the '15' FPS value in the bracket after z16:

 config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16)

Try replacing the profile line with this:

**profile = pipeline.start(config)**

ev-mp, 2020-05-04:
@jurc192 , most of the syntax errors were highlighted by@MartyG-RealSense. In addition - 
> (1) config.enable_stream(rs.stream.depth, ~~1~~**0**, 848, 480, rs.format.z16, 15) # can't resolve
(2) config.enable_stream(rs.stream.depth, ~~1~~**0**)                              # can't resolve
(5) config.enable_stream(rs.stream.depth, ~~1~~**0**, rs.format.z16, 15)           # can't resolve

Depth stream index is always zero, requesting another index will fail by design

>(4) config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 15)    # USB2 fails, USB3 works

848x480/15FPS for D435i is available with USB3 connection only. Use `rs-enumerate-devices` tool to list the supported configurations. See a similar question at https://github.com/IntelRealSense/librealsense/issues/6342#issue-611481473

> (7) config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16)        # can't resolve

This should work - double-check and update.

jurc192, 2020-05-04:
@ev-mp aha I see, so my mistake was using index 1 instead of 0, thanks!
It is a bit confusing that index variable normally starts with 0, but the for enabling IR streams one needs to use 1 and 2. That is why I assumed 1 as a default value. Could we add a small comment/note about this somewhere in documentation or API to avoid confusion in future?

Setting the index to zero works for (1) and (2). However, (5) and (7) still don't resolve (using USB3):
```
# config.enable_stream(rs.stream.depth, 0, rs.format.z16, 15)           # can't resolve
# config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16)        # can't resolve
```

MartyG-RealSense, 2020-05-04:
Could you run the RealSense Viewer please and check at the top of the Viewer's options side-panel whether your camera is being mis-detected as a USB2 connection when the camera is in a USB 3 port?  that could explain why 848x480 is not being recognized on a USB3 port.

jurc192, 2020-05-04:
I tried it and it recognises it as a USB3 device. Using the default settings (848x480, z16, 30 FPS) works in realsense-viewer (I can see the depth-stream). Python code fails.

MartyG-RealSense, 2020-05-04:
Okay, delete #5 and use the line below for #7 if you are not doing so already:

**config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 15)**

MartyG-RealSense, 2020-05-17:
Hi @jurc192    Do you require further assistance with this question please, or is it okay to close this case?   Thanks!

jurc192, 2020-05-18:
We can close the case, thanks

closed 2020-05-18 by jurc192

MartyG-RealSense, 2020-05-18:
Thanks so much!