realsenseai/librealsense #6343: Running the pipeline in a loop hangs after several iterations

jurc192, 2020-05-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (D435i) | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version | Arch Linux | 
| Kernel Version (Linux Only)    |  5.6.7-arch1-1 | 
| Platform                                 | PC  |
| SDK Version                            |  2.3.3.1 | 
| Language                            |  python                        | 
| Segment			|  others | 

### Issue Description
Starting and stopping the pipeline in a loop always hangs/freezes the program after several iterations (around 200).
Upon freezing, stopping the program and re-starting it, it throws: `RuntimeError: failed to set power state`. Plugging the device in and out fixes the problem.

### Minimal example
```

import pyrealsense2 as rs

def capture_depthframe():
    pipeline = rs.pipeline()
    config   = rs.config()
    config.enable_stream(rs.stream.depth)
    print("\tResolving configuration....", end='')
    profile = config.resolve(pipeline)   # hangs here after 200-ish iterations
    print("OK")
    pipeline.start(config)
    pipeline.stop()

for i in range(500):
    print(i, end='')
    capture_depthframe()
```

MartyG-RealSense, 2020-05-03:
Could you try a depth config instruction with more specific settings please and see if it makes a difference.  For example:

**config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 30)**

ev-mp, 2020-05-04:
@jurc192 , it could be that the flow produces to much stress on the USB sub-system/device.
>    pipeline.start(config)
    pipeline.stop()

Put 2-3 sec delay in between those two and update.

jurc192, 2020-05-04:
So I tried both suggestions and the problem persists - program hangs around 200-ish iterations.
@MartyG-RealSense using `config.enable_stream(rs.stream.depth, 640, 360, rs.format.z16, 30)` doesn't fix the problem.
@ev-mp adding a delay of two seconds in between doesn't fix the problem either

Could it be a memory leak somewhere, or something similar? If the problem is on my side (OS), any idea which logs should I be looking into?

ev-mp, 2020-05-04:
@jurc192 , can you generate and attach the Librealsense logs for review?
Also provide the kernel log when the freeze occurs (dmesg -wT).
Thanks

jurc192, 2020-05-04:
Using loglevel="DEBUG" I get this output: [2020-05-04-15_54_08.log](https://github.com/IntelRealSense/librealsense/files/4575067/2020-05-04-15_54_08.log)

Head and tail of the file:
```
04/05 15:54:08,394 DEBUG [140243383465792] (context.cpp:108) Librealsense VERSION: 2.33.1
 04/05 15:54:08,414 DEBUG [140243383465792] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 04/05 15:54:08,422 DEBUG [140243383465792] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 04/05 15:54:08,422 DEBUG [140243383465792] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-3-3
 04/05 15:54:08,752 DEBUG [140243383465792] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-3-3
 04/05 15:54:08,803 WARNING [140243383465792] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed. Error type: No data to return (-21).
 04/05 15:54:08,804 WARNING [140243383465792] (ds5-motion.cpp:463) IMU Calibration is not available, see the previous message
 04/05 15:54:08,804 INFO [140243383465792] (ds5-motion.cpp:225) 
 04/05 15:54:08,849 DEBUG [140243383465792] (ds5-private.h:385) Loaded Valid Table: version [mjr.mnr]: 0x0202, type 32, size 240, CRC: c84047c4
 04/05 15:54:08,986 INFO [140243383465792] (sensor.cpp:1338) Request: Z16 Depth, 
Resolved to: Z16 Depth, 
 04/05 15:54:09,039 INFO [140243383465792] (uvc-streamer.cpp:28) endpoint 82 read buffer size: 815104

....
04/05 15:56:18,235 INFO [140243383465792] (environment.cpp:85) Found 305 unreachable streams, 305 extrinsics deleted
 04/05 15:56:18,494 DEBUG [140243383465792] (rsusb-backend.cpp:34) Creating UVC Device from path: 2-3-3
 04/05 15:56:18,556 WARNING [140243383465792] (types.cpp:49) Accel Sensitivity:hwmon command 0x4f failed. Error type: No data to return (-21).
 04/05 15:56:18,556 WARNING [140243383465792] (ds5-motion.cpp:463) IMU Calibration is not available, see the previous message
 04/05 15:56:18,556 INFO [140243383465792] (ds5-motion.cpp:225) 
 04/05 15:56:18,600 DEBUG [140243383465792] (ds5-private.h:385) Loaded Valid Table: version [mjr.mnr]: 0x0202, type 32, size 240, CRC: c84047c4
 04/05 15:56:18,728 INFO [140243383465792] (sensor.cpp:1338) Request: Z16 Depth, 
Resolved to: Z16 Depth, 
 04/05 15:56:18,782 INFO [140243383465792] (uvc-streamer.cpp:28) endpoint 82 read buffer size: 815104
 04/05 15:56:18,796 DEBUG [140243383465792] (global_timestamp_reader.cpp:132) time_diff_keeper::start: _users_count = 1
 04/05 15:56:18,829 DEBUG [140243383465792] (global_timestamp_reader.cpp:143) time_diff_keeper::stop: _users_count = 0
 04/05 15:56:18,829 DEBUG [140243383465792] (global_timestamp_reader.cpp:146) time_diff_keeper::stop: stop object.
 04/05 15:56:18,829 DEBUG [140243181213440] (global_timestamp_reader.cpp:206) Notification: time_diff_keeper polling loop is being shut-down
 04/05 15:56:18,881 DEBUG [140243383465792] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.
 04/05 15:56:18,888 DEBUG [140243383465792] (ds5-factory.cpp:938) try_fetch_usb_device(...) failed.

```

dmesg showed no messages during the execution.

ev-mp, 2020-05-04:
@jurc192 , I can't see any hint in the log provided, so it should be further analyzed.
One thing to mention is that you're using `rsusb_backend`, is it by design ?

I would suggest  to rebuild and rerun the test with standard v4l uvcvideo driver to isolate the issue.

jurc192, 2020-05-04:
@ev-mp I compiled the library from source and I guess it was easier to do it that way. Am I right that this has to do with `FORCE_LIBUVC` flag when compiling? I compiled the SDK with this flag set to `True`.

Documentation for installing/compiling the SDK is too Ubuntu specific and the short snippet describing how to manage things on Arch was not too helpful, since the script for patching the kernel does not work. 

I can try recompiling the library without `FORCE_LIBUVC` flag and patching the kernel manually (never did this before, so it'll take some time). If you could provide some more explanation on how exactly to compile the library and what needs to be done with kernel (and why), it would help alot/speed things up.

MartyG-RealSense, 2020-05-04:
@jurc192   Regarding your question about Linux installation, Intel have a YouTube installation tutorial video for installing the SDK using the Debian package method.

https://youtu.be/IcCUyybZmAE

MartyG-RealSense, 2020-05-17:
@jurc192    Do you require further assistance with your question please, or is it okay to close this case?   Thanks!

jurc192, 2020-05-18:
@MartyG-RealSense I would like to investigate this further, since it seems like a bug which is annoying to work with in my scenario. I think I can recompile and run the example again sometime this week

MartyG-RealSense, 2020-05-18:
Okay, thanks for the update.  I look forward to hearing more from you after your investigations.  Good luck!

jurc192, 2020-05-21:
So, I re-compiled the library, patched the kernel and re-run the program. It still does not work, but at least now it throws a RuntimeError:
```
```

jurc192, 2020-05-21:
@ev-mp  
So I recompiled the library (without setting the RSUSB BACKEND flag), patched the kernel manually (realsense-camera-formats.patch) and re-run the example. It still breaks, but this time it throws a RuntimeError:
```
  File "confg_resolve_hang.py", line 18, in <module>
    capture_depthframe()
  File "confg_resolve_hang.py", line 11, in capture_depthframe
    profile = config.resolve(pipeline)   # hangs here after 200-ish iterations
RuntimeError: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
```
[The realsense log](https://github.com/IntelRealSense/librealsense/files/4663833/2020-05-21-19_23_25.log) with the last couple of lines:
```
 21/05 19:24:54,920 DEBUG [140041514731264] (global_timestamp_reader.cpp:212) Notification: time_diff_keeper polling loop is being shut-down
 21/05 19:24:54,927 DEBUG [140041632278336] (ds5-factory.cpp:1028) try_fetch_usb_device(...) failed.
 21/05 19:24:54,931 DEBUG [140041632278336] (ds5-factory.cpp:1028) try_fetch_usb_device(...) failed.
 21/05 19:24:54,934 INFO [140041632278336] (environment.cpp:85) Found 304 unreachable streams, 304 extrinsics deleted
 21/05 19:25:00,002 WARNING [140041632278336] (types.cpp:49) get_xu(ctrl=1) failed! Last Error: Device or resource busy
 21/05 19:25:00,005 DEBUG [140041632278336] (config.cpp:151) Iterate available devices - config can not be resolved. get_xu(ctrl=1) failed! Last Error: Device or resource busy
 21/05 19:25:00,128 DEBUG [140041632278336] (ds5-factory.cpp:1028) try_fetch_usb_device(...) failed.
 21/05 19:25:00,137 INFO [140041632278336] (environment.cpp:85) Found 3 unreachable streams, 3 extrinsics deleted
 21/05 19:25:00,138 ERROR [140041632278336] (types.h:307) set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
 21/05 19:25:00,140 WARNING [140041632278336] (rs.cpp:1745) null pointer passed for argument "profile"
```
Any news about this? I also stumbled upon [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/DSO-1369.md), which sounds similar to my original problem.

PS: This time I installed the latest SDK from github @commit bc64e52c9 (v2.34.0), and the firmware version is the same (05.12.03.00)

jurc192, 2020-05-21:
Also, I am getting a looot of those in dmesg:
 `uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024)`

Guessing it's related to #5302

MartyG-RealSense, 2020-05-21:
I went over what you have done so far from the start to get some fresh insights.  It seems like the system might have got in a tangle from multiple installation attempts by different methods, and it might be better to just do a clean-up of the computer and a fresh install of Librealsense as though it is the first time.

MartyG-RealSense, 2020-05-30:
Do you require further assistance please  or can this case be closed?   Thanks!

MartyG-RealSense, 2020-06-05:
Case closed due to no further comments.

closed 2020-06-05 by MartyG-RealSense