realsenseai/librealsense #6346: Compilation error.

malapatiravi, 2020-05-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |    D415/84141206XXXX        05.12.02.100, D415/90952206XXXX        05.12.03.00 | 
| Operating System & Version |  Ubuntu 16 | 
| Kernel Version (Linux Only)    | 4.15.0-91-generic                                      | 
| Platform                                 | PC |
| SDK Version                            |  2.33.1, 2.34.1       | 
| Language                            |  C++ / Python| 
| Segment			|  Robot        | 

### Compilation Error while compiling with Ubuntu 16
I am trying to compile the librealsense and install in ubuntu 16 and i have the following error. 
Steps i followed. 
```
cd librealsense
mkdir build && cd build
cmake ../ -DBUILD_EXAMPLES=true -DENFORCE_METADATA=true -DBUILD_PYTHON_BINDINGS=true -DFORCE_LIBUVC=true -DBUILD_WITH_CUDA=true -DIMPORT_DEPTH_CAM_FW=false
make

```
This system has two Realsense cameras already running so prerequisites is not a problem. I have reinstalled all the kernel patch and prerequisites. 
I am facing the same problem with 2.33.1 and 2.34.1. Please let me know if i am doing something wrong. Earlier i was able to build using the same configuration as above.  

```
In file included from /home/dummy/librealsense-2.34.1/src/rsusb-backend/rsusb-backend.cpp:12:0:
/home/dummy/librealsense-2.34.1/src/rsusb-backend/../tm2/tm-boot.h:11:30: fatal error: common/fw/target.h: No such file or directory
compilation terminated.
CMakeFiles/realsense2.dir/build.make:1102: recipe for target 'CMakeFiles/realsense2.dir/src/rsusb-backend/rsusb-backend.cpp.o' failed
make[2]: * [CMakeFiles/realsense2.dir/src/rsusb-backend/rsusb-backend.cpp.o] Error 1
CMakeFiles/Makefile2:991: recipe for target 'CMakeFiles/realsense2.dir/all' failed
make[1]: * [CMakeFiles/realsense2.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
```

RealSenseCustomerSupport, 2020-05-05:
----------------------------------------------

Hi malapatiravi,
please see if
either add -DBUILD_WITH_TM2=false
or remove -DIMPORT_DEPTH_CAM_FW=false
can help in your case

dorodnic, 2020-05-05:
... @malapatiravi, in addition to what @RealSenseSupport suggested, most likely you can find error message during CMake (before compilation) about not detecting internet connection. If it is possible I would suggest looking into that as well. Perhaps you need to configure proxy settings (depends on your network)

MartyG-RealSense, 2020-05-05:
@malapatiravi   I also notice that you are using **-DENFORCE_METADATA=true** ... you have listed Ubuntu as your OS, and the build settings information page says that the build will break if -DENFORCE_METADATA is set to True and the Windows SDK is not installed.

[image]

My understanding is that with Linux, metadata support would normally be added by patching support in when compiling from source or by building the SDK using the RUSB backend "bypass" method listed below:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

malapatiravi, 2020-05-05:
adding -DBUILD_WITH_TM2=false flag worked

closed 2020-05-05 by malapatiravi