realsenseai/librealsense #6347: Permanent "Device or Resource Busy" after start

kremerf, 2020-05-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.02.100 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Platform                                 | NUC  |
| Language                            |  C++                          | 

This is a follow up question to [this topic](https://github.com/IntelRealSense/librealsense/issues/6307).
The program itself works, but now and then upon start it cannot establish the `rs2::pipeline` and cannot thereafter. I have not found a way to exactly reproduce this failure, as all it takes sometimes is a restart of the package to work or do not work anymore.

Once again the core code:
```
rs2::pipeline p;
p.start();
std::this_thread::sleep_for(std::chrono::milliseconds(1000));
cout << "Setting Vars RGB" << endl;
device dev1 = p.get_active_profile().get_device();
auto sensor = dev1.query_sensors()[1];
std::this_thread::sleep_for(std::chrono::milliseconds(1000));
sensor.set_option(RS2_OPTION_EXPOSURE, 1);
cout << sensor.get_info(RS2_CAMERA_INFO_NAME) << endl;
cout << sensor.get_option(RS2_OPTION_EXPOSURE) << endl;
std::this_thread::sleep_for(std::chrono::milliseconds(1000));
cout << "Setting Vars Depth" << endl;
dev1.query_sensors()[0].set_option(rs2_option::RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
auto frameset = p.wait_for_frames();
p.stop();
```
The output (When not working):
```
[ INFO] [1588662810.456078015]: Device with name Intel RealSense D435 was found.
[ INFO] [1588662810.459209567]: Device with port number 2-3 was found.
[ INFO] [1588662810.459383333]: Resetting device...
Setting Camera!
[...] // Other not related startup lines
 05/05 09:13:36,938 ERROR [139771674334976] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
Could not set Vars, will try again.
RealSense error calling rs2_pipeline_start(pipe:0x55f0053bac20):
    xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
```
The code above is within a try-catch, that generates the "Could not set Vars, will try again.":
```
catch (const rs2::error & e) {
	cout << "Could not set Vars, will try again." << endl;
	repeat = true;
	std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
}
catch (const std::exception & e) {
	cout << "Could not set Vars, will try again." << endl;
	repeat = true;
    std::cerr << e.what() << std::endl;
}
```
Right after all this output there are a couple:

```
05/05 09:13:37,139 ERROR [139771674334976] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
05/05 09:13:37,398 ERROR [139771674334976] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
05/05 09:13:37,599 ERROR [139771674334976] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
05/05 09:13:38,503 ERROR [139771674334976] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
05/05 09:13:38,804 ERROR [139771674334976] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
05/05 09:13:39,005 ERROR [139771674334976] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
```
I am not sure whether these get_device-calls are related to my problem or not. They sometimes are generated randomply during the runtime, but usually not more than one at a time and not that often.

Now to the busy-errors, the only .cpp-file that includes and uses the realsense-sdk is the main.cpp. And even with adding a retry-routine, that tries to connect a pipeline every couple seconds the error is just being reproduced every single time the first connect does not work.

So, to actually produce a guiding question:
How can I establish a reliable pipeline, that connects 100% of the time (excluding actual system errors or hardware failures).

Additionally..
I found a topic some time ago, where someone had to wait for 30 seconds to get his connection going, but that didn't work for me:
```
Setting Camera!
Could not set Vars, will try again.
RealSense error calling rs2_pipeline_start(pipe:0x55868c45f3e0):
    xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
 05/05 09:29:44,918 ERROR [139891245577984] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
Setting Camera!
 05/05 09:30:11,941 ERROR [139891245577984] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
Could not set Vars, will try again.
RealSense error calling rs2_pipeline_start(pipe:0x55868c44ce70):
    xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
Setting Camera!
Could not set Vars, will try again.
RealSense error calling rs2_pipeline_start(pipe:0x55868c4379c0):
    xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
Setting Camera!
Could not set Vars, will try again.
RealSense error calling rs2_pipeline_start(pipe:0x55868c430200):
    xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
 05/05 09:31:35,021 ERROR [139891245577984] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
Setting Camera!
Could not set Vars, will try again.
RealSense error calling rs2_pipeline_start(pipe:0x55868c40b780):
    xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
```
One thing every single fail has in common is, that no output line is published becide "Setting Camera!", which means, that the problem is right at the establish-point of the pipeline.

MartyG-RealSense, 2020-05-05:
The link below has some C++ scripts provided by RealSense users for their different approaches to catching device-busy and pipeline start errors:

https://github.com/IntelRealSense/librealsense/issues/2240

kremerf, 2020-05-07:
I remember that topic, yes. Some of it I tried and I think I had some problems with some parts. Let's see again:

[](https://github.com/IntelRealSense/librealsense/issues/2240#issuecomment-413555585) is exactly my try-catch approach.

[](https://github.com/IntelRealSense/librealsense/issues/2240#issuecomment-415433682) this one is missing some crucial parts if I'm not mistaken.
The `logWriter()` is not declared and it accepts `rs2_error` as well as generic `Strings` therefor it won't be the `check_error(rs2_error* e)` mentioned [here.](https://github.com/IntelRealSense/librealsense/blob/9437bc24c0f4ce16193f1045ceea703a8151b96d/doc/rs400/rs400_advanced_mode.md)

I changed that, so that all those LogLines are printed via cout. `m_intel->print_device_info(dev)` is not declared either and results in just one google result, linking me to the comment itself ( :
```
void logWriter(rs2_error* e) {
	std::cerr << "RealSense error calling " << rs2_get_failed_function(e) << "(" << rs2_get_failed_args(e) << ")" << std::endl;
}

void check_errors()
{
	rs2_error* e = 0;

	// Create a context object. This object owns the handles to all connected realsense devices.
	// The returned object should be released with rs2_delete_context(...)
	rs2_context* ctx = rs2_create_context(RS2_API_VERSION, &e);
	if (e)
		logWriter(e);

	/* Get a list of all the connected devices. */
	// The returned object should be released with rs2_delete_device_list(...)
	rs2_device_list* device_list = rs2_query_devices(ctx, &e);
	if (e)
		logWriter(e);

	int dev_count = rs2_get_device_count(device_list, &e);
	if (e)
		logWriter(e);

	if (dev_count == 0)
	{
		cout << "There is no Intel RealSense camera conected" << endl;
		return;
	}
	else
	{
		cout << "There are " << std::to_string(dev_count);
		cout << " connected RealSense devices" << endl;
	}

	//TODO, hardcode for the first sensor, loop for every device detected
	/*create this rs2::device hardware for make the hardware_reset, cannot use the rs2_device* dev
          because rs2_device is a typedef and don't have the definition,  so we can only use a pointer for
          that type but not access the content*/
	rs2::context cxt;
	rs2::device_list devices = cxt.query_devices();
	rs2::device hardware = devices[0];

	// Get the first connected device
	// The returned object should be released with rs2_delete_device(...)
	rs2_device* dev = rs2_create_device(device_list, 0, &e);
	// m_intel->print_device_info(dev);
	if (e)
		logWriter(e);

	// Create a pipeline to configure, start and stop camera streaming
	// The returned object should be released with rs2_delete_pipeline(...)
	rs2_pipeline* pipeline = rs2_create_pipeline(ctx, &e);
	if (e)
		logWriter(e);

	// Create a config instance, used to specify hardware configuration
	// The retunred object should be released with rs2_delete_config(...)
	rs2_config* config = rs2_create_config(&e);
	if (e)
		logWriter(e);

	// Request a specific configuration
	// rs2_config_enable_stream(config, RS2_STREAM_DEPTH, -1, m_DepthWidth, m_DepthHeight, RS2_FORMAT_RGB8, 30, &e);
	if (e)
		logWriter(e);

	// Start the pipeline streaming
	// The retunred object should be released with rs2_delete_pipeline_profile(...)
	rs2_pipeline_profile* pipeline_profile = rs2_pipeline_start_with_config(pipeline, config, &e);
	if (e)
		logWriter(e);

	rs2_delete_device(dev);
	rs2_delete_device_list(device_list);
	rs2_delete_context(ctx);

	//Force a Hard reset because the app only will enter on this code if the pipeline.start() failed
	hardware.hardware_reset();
	return;
}

```

So, now to some outputs:
```
Setting Camera!
There is no Intel RealSense camera conected

///

Setting Camera!
There are 1 connected RealSense devices
RealSense error calling rs2_pipeline_start_with_config(pipe:0x558eeaad1dc0, config:0x558eeaad6070)
Setting Vars RGB
Could not set Vars, will try again.
RealSense error calling rs2_set_option(options:0x558eeb152920, option:Exposure, value:2):
    xioctl(VIDIOC_G_CTRL) failed Last Error: No such device
```

When including this check_errors() function there hasn't been a single successfull start. When not including said function (as of today) 9/10 starts were successfull.

kremerf, 2020-05-07:
Looking up the [documentation](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline__profile.html), I did not find any of my functions with suitable `rs2_error` options as Input, the ones used in `check_errors()` seem to be special in that case.

MartyG-RealSense, 2020-05-07:
If you keep getting a warning message in the log with a retry routine each time the camera tries to connect to the pipeline, I wonder if you could disable logging by setting the log severity level to **RS2_LOG_SEVERITY_NONE** 

I'd normally be hesitant to disable logging as warning / error messages are very useful.  If you are otherwise confident that the program is working fine and it is only generating the message because it is retrying the connection, it may be worth considering disabling log messaging and just let the program keep retrying silently.

The logging level could be set with **log_to_console**:

https://intelrealsense.github.io/librealsense/doxygen/namespacers2.html#a29c3e985cd3a515a9231ccedde4168ff

kremerf, 2020-05-07:
The program itself is working, but without a connection to the cameras, there is not much work to do. So in theory I could silence the log, but I let it try to reconnect for ten minutes (with a 30 second delay between each try) and it did not work. Which is what I still don't understand. Do you have any idea how to investigate who is actually making the device busy? As I am using `start()` and `stop()` I am at least quite certain, that is is not my code.

Can I see which process, user or whatever is connected to the camera?

I just tested to open the camera with `realsense_viewer` and started my program. But the `initial_reset` that I have in my launch File disconnected the device from said viewer and it started perfectly fine. So can the camera be busy "with itself" doing something that's heavy on its hardware?

kremerf, 2020-05-07:
Starting the program and then running the `realsense-viewer` also results in
```
Backend in rs2_open_multiple(sensor:0x7f11480107d0, profiles:0x7f114858e7a0, count:1):
xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy
```

MartyG-RealSense, 2020-05-07:
There was a case where someone was getting this error when accessing the camera through a Docker container.  Are you using Docker please?

https://github.com/IntelRealSense/realsense-ros/issues/642

kremerf, 2020-05-07:
No, I am not. 

I have made a ROS-Package that includes the camera (Depth + RGB), another thermal camera, a ROS-Listener for each camera, that saves the sent images, so that I dont have to request them when needed and one node, that does stuff with those images.

MartyG-RealSense, 2020-05-07:
If you use a pipeline stop command before using the pipeline start, that might ensure that any process that is already using the pipeline is shut off before the pipeline start call is made.

If that does not make a difference then the cause of the error may be something other than a process already using the camera.

kremerf, 2020-05-07:
You cannot call stop before start unfortunately. So we have to rely on the Device resetting for this one.

```
RealSense error calling rs2_pipeline_stop(pipe:0x560ba86ae8c0):
    stop() cannot be called before start()

```

kremerf, 2020-05-07:
> ...the pipeline doesn't own the device, thus, the pipeline selected device may change in subsequent activations.
So even if there were another instance of a pipeline it should not interfere.
[Source](http://docs.ros.org/kinetic/api/librealsense2/html/classrs2_1_1pipeline.html)

MartyG-RealSense, 2020-05-07:
You mentioned that you were using a ROS node.  I found a case where someone who was using a ROS node had the same busy error and fixed it by terminating the node.

https://github.com/IntelRealSense/librealsense/issues/1828#issuecomment-519962862

Subsequent comments on this discussion suggested applying a kernel patch to fix it.

kremerf, 2020-05-07:
/D435/realsense2_camera_manager/bond
/D435/realsense2_camera
/D435/realsense2_camera_manager

Are my ROS-Nodes within the D435 package. I am not sure which one to terminate, as they seem rather important. The camera_manager is connecting/managing all topics that I am subscribing to and has the camera_manager_bond as its own parent.

Applying a Kernel fix eludes my knowledge of ubuntu. 

I killed the /D435/realsense2_camera_manager while my program was stuck in the restart connect loop and it did actually finish loading. But it did not receive any image thereafter.

MartyG-RealSense, 2020-05-07:
Yes, they are apparently both vital nodes.  

I wonder if a delay could be put on the ROS launch so that the node starts after the program that starts the pipeline has launched, to see if that resolves the conflict?

kremerf, 2020-05-07:
The realsense2_camera Node is just connected to the camera_manager (and I am not entirely sure, which line launches it, I just found the name in the rs_camera.launch). But if there is no camera to manage, can it establish a connection?

`<arg name="manager"             default="realsense2_camera_manager"/>`

If I delay the camera_Manager node, there cannot be any connection to the camera. Hmm..

kremerf, 2020-05-07:
[image]

This is the camera package. (With hidden leaf topics and hidden tf tree)

MartyG-RealSense, 2020-05-07:
I have considered the problem very carefully but was unable to develop a solution.  It is one that would benefit from the advice of a RealSense team member with expert ROS knowledge, such as @doronhi or @radfordi ... I do apologise.

kremerf, 2020-05-07:
I just felt like testing success-to-failure rate.
At first it was 1 failure to about 4 successes, but in the end there have been 7 failures back to back. After a system reboot the first try was successful.
Followed by a 1 to 2 ratio.

After that I rebooted before every start six times, and every single one succeeded.

Of course, tests like these should be taken with caution, but the sudden increase in problems and the randomness makes me think, that maybe some processes aren't properly killed between two runs. As you already mentioned above. 

Though I still don't know which node that might be.

kremerf, 2020-05-13:
Still haven't found a solution yet.

MartyG-RealSense, 2020-05-13:
@kremerf    I'll remind the manager that I mentioned earlier in the case about looking at it.

doronhi, 2020-05-27:
Hi @kremerf ,
I try to understand the issue.
Do you run your own code that opens and starts a rs2::pipeline? If so, why is this a ROS question?
Or do you run a realsense2_camera node? In that case, where is your code coming in?

kremerf, 2020-05-27:
It got labeled as such during my conversation with @MartyG-RealSense, I run my own code, in a ROS environment. I mentioned this environment because sometimes it is a crucial info [6347-comment](https://github.com/IntelRealSense/librealsense/issues/6347#issuecomment-625186279).

I run a realsense2_camera node.

`<include file="$(find realsense2_camera)/launch/rs_camera.launch">`

The procedure goes as follows:
1. roslaunch "myProject"
2. startup
2.1 Main.cpp gets loaded and is processed
2.1.1 rs2::pipeline is being created to set exposure of the camera
2.1.2 Pipeline either finishes cleanly or doesnt
2.2 Main.cpp is finished, rest of code is inside a class that was created and is now proccessed in a loop

The errors are..
A: ..directly in 2.1.1 which is being executed right after the hardware_reset routine of the camera. In this case the pipeline cannot be created and I get an error message.
B: ..directly in 2.1.1 right after the hardware_reset. Sometimes the pipeline can be created, but the setting of the exposure value is not being correctly transmitted (standard value is not changed)
C: .. inside the loop, when the initial pipeline can be created and successfully changes the exposure time. In this case I try to open another pipeline during runtime which never succeeds.
D: .. like C. but the initial creation is also not successful.

kremerf, 2020-05-28:
```
[ INFO] [1590665862.604891687]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[ INFO] [1590665862.604975851]: Device with name Intel RealSense D435 was found.
[ INFO] [1590665862.610146983]: Device with port number 2-2 was found.
[ INFO] [1590665862.610443209]: Resetting device...
[ INFO] [1590665868.616165183]:
[ INFO] [1590665868.672444137]: Device with serial number 740112071197 was found.

[ INFO] [1590665868.672494195]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[ INFO] [1590665868.672515867]: Device with name Intel RealSense D435 was found.
[ INFO] [1590665868.673845741]: Device with port number 2-2 was found.
[ INFO] [1590665868.676855884]: getParameters...
[ INFO] [1590665868.749315027]: setupDevice...
[ INFO] [1590665868.749344187]: JSON file is not provided
[ INFO] [1590665868.749376042]: ROS Node Namespace: D435
[ INFO] [1590665868.749392060]: Device Name: Intel RealSense D435
[ INFO] [1590665868.749413952]: Device Serial No: 740112071197
[ INFO] [1590665868.749431689]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0
[ INFO] [1590665868.749445471]: Device FW version: 05.12.03.00
[ INFO] [1590665868.749460730]: Device Product ID: 0x0B07
[ INFO] [1590665868.749475020]: Enable PointCloud: On
[ INFO] [1590665868.749488834]: Align Depth: On
[ INFO] [1590665868.749509083]: Sync Mode: On
[ INFO] [1590665868.749769149]: Device Sensors:
[ INFO] [1590665868.749920463]: Stereo Module was found.
[ INFO] [1590665868.750001043]: RGB Camera was found.
[ INFO] [1590665868.750067452]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1590665868.750100974]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1590665868.750119461]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1590665868.750261146]: Add Filter: pointcloud
[ INFO] [1590665868.750875347]: num_filters: 1
[ INFO] [1590665868.750891669]: Setting Dynamic reconfig parameters.
[ INFO] [1590665868.879695448]: Done Setting Dynamic reconfig parameters.
[ INFO] [1590665868.883036458]: depth stream is enabled - width: 1280, height: 720, fps: 6, Format: Z16
[ INFO] [1590665868.883621992]: infra1 stream is enabled - width: 1280, height: 720, fps: 6, Format: Y8
[ INFO] [1590665868.888363527]: color stream is enabled - width: 1280, height: 720, fps: 6, Format: RGB8
[ INFO] [1590665868.892946601]: setupPublishers...
[ INFO] [1590665868.894911746]: Expected frequency for depth = 6.00000
[ INFO] [1590665868.913314223]: Expected frequency for infra1 = 6.00000
[ INFO] [1590665868.924796376]: Expected frequency for aligned_depth_to_infra1 = 6.00000
[ INFO] [1590665868.936640162]: Expected frequency for color = 6.00000
[ INFO] [1590665868.947496112]: Expected frequency for aligned_depth_to_color = 6.00000
[ INFO] [1590665868.959602297]: setupStreams...
[ INFO] [1590665868.960714310]: insert Depth to Stereo Module
[ INFO] [1590665868.960884257]: insert Color to RGB Camera
[ INFO] [1590665868.960919599]: insert Infrared to Stereo Module
[ INFO] [1590665868.967939996]: SELECTED BASE:Depth, 0
[ INFO] [1590665868.974404372]: RealSense Node Is Up!
 28/05 13:37:49,865 ERROR [140081474029312] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 28/05 13:37:50,082 ERROR [140081474029312] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 28/05 13:37:50,737 ERROR [140081474029312] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 28/05 13:37:51,240 ERROR [140081474029312] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 28/05 13:37:51,692 ERROR [140081474029312] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 28/05 13:37:52,194 ERROR [140081474029312] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 28/05 13:37:52,696 ERROR [140081474029312] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 28/05 13:37:53,198 ERROR [140081474029312] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
Running..
Setting Camera!
RGB Camera, Exposure: 2
Running..
 Image is too old     -> 0
 Image is too old     -> 1
 Image is too old     -> 2
Setting Camera!
RGB Camera, Exposure: 5
Running..
Setting Camera!
RGB Camera, Exposure: 8
Running..
Setting Camera!
[... 77 lines cut ...]
		if (repeatNow) {
			if (ic.doCamera()){
				try{
					cout << "Setting Camera!" << endl;
					repeatNow = false;
					ic.setExposure(exposure);
					exposure+=3;
				}
				catch (const rs2::error & e) {
					cout << "1Could not set Vars, will try again." << endl;
					repeat = true;
					std::cerr << "2RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
					return EXIT_FAILURE;
				}
				catch (const std::exception & e) {
					cout << "2Could not set Vars, will try again." << endl;
					repeat = true;
				    std::cerr << "3" << e.what() << std::endl;
				    return EXIT_FAILURE;
				}
```

What did the trick?:

I changed from using rs2::pipeline to rs2::device. This does in fact not answer my original question. But it is a workaround for now. I would still like to know, why the previous approach wasn't successful.

MartyG-RealSense, 2020-06-05:
@kremerf   Do you require further assistance on this case or can it be closed please?  Thanks!

kremerf, 2020-06-10:
No, it works now. But in my opinion the pipelines are not working as intended (as they are not working), so this should be fixed.

closed 2020-06-10 by kremerf