realsenseai/librealsense #6348: Runtime Error: No device connected

Carlito9, 2020-05-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Framos D435e  | 
| Firmware Version                   | 5.12.0.0 | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC  |
| SDK Version                            |   2.29.0                          | 
| Language                            |  python                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi,
I found a lot of issues with a similar name but I wasn't able to find a solution for my problem.
The problem is this: every example (for Python) I run gives me the runtime error: "No device connected" when I call pipeline.start(), but, If I use the tool "rs-record", the camera works correctly. The tool rs-enumerate-devices show me that the camera is recognised, and gives me the devices info but if I call query_devices() function it returns an empty list.
I can't use realsense-viewer because the camera is not plugged in my computer, but in a computer which I can use using a ssh connection, so I have only access to the shell.
Thanks for help

dorodnic, 2020-05-05:
D435e is a product of Framos GmbH
Please consider contacting [Eng-support@framos.com](Eng-support@framos.com)

Carlito9, 2020-05-05:
Thank you for the answer

closed 2020-05-05 by Carlito9