realsenseai/librealsense #6351: Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-exposure

LYKlyk, 2020-05-05:
| | |
|---|---|
|**librealsense**|2.34.0 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

ev-mp, 2020-05-06:
@LYKlyk , Please specify the camera type, FW version used and provide a screenshot or any other description of the problem.

LYKlyk, 2020-05-07:
My Os is Windows10, My device is Realsense SR300, I try to use "Intel RealSense SDK Sample Browser" (Hand Tracking for Windows* Release Notes (11.0.27.1404), https://software.intel.com/content/www/us/en/develop/articles/realsense-sdk-windows-eol.html) to get color/depth stream and hand skeleton. But the application is not work. When I use "Raw Stream" , I cannot get any frame. When I use "Hands Viewer", I always show "Init Failed".

ev-mp, 2020-07-27:
@LYKlyk , this warning mentioned in ticket's title is about the non-intel web or another UVC camera connected to PC. Can be ignored.
https://github.com/IntelRealSense/librealsense/issues/6784#issuecomment-662108300
duplicated in #6784

ev-mp, 2020-07-27:
The new wrapper functionalities are now part of the [v2.37.0 release](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#pre-release-2370)

ev-mp, 2020-07-30:
@LYKlyk hello, to clarify -
You mention an issue with Intel RealSense SDK ((11.0.27.1404), which is a discontinued product as stated on the link. 
While at the ticket description you specify LRS v2.34.0, and also the message that appears in the ticket title is fired by Librealsense SDK (that I posted earlier)

Do you need support with LRS v.2.34.0?

RealSenseSupport, 2020-08-03:
Hi @LYKlyk  Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

closed 2020-08-14 by RealSenseCustomerSupport