realsenseai/librealsense #6352: Camera not available on Ubuntu 18.04 with realsense-viewer

llwp, 2020-05-06:
| | |
|---|---|
|**librealsense**|2.34.0 RELEASE|
|**OS**|Ubuntu 18.04|
|**Model**|D435|

Various issues "detecting frames" when first make/installed.  Currently no camera shown in viewer, even though "camera detected" during app launch.  OS doesnt detect camera either...

Would like to use this camera, simply as a camera, before anything else and the experience has been .. sub-optimal.

Is the only path here to uninstall/re-install or is there a better way to just use the camera?

ev-mp, 2020-05-06:
@llwp , could you be more specific what are the `Various issues "detecting frames"` and `OS doesnt detect camera either` ? Are you building from sources or using Debians?
Attaching kernel and console logs will be helpful:
 - lsusb
- dmesg -wT
 - Librealsense log/console messages

llwp, 2020-05-07:
Tried both .deb and based on docs, seemed like I needed to still use `make`.

Right now, I'm sure its a mess.

I didn't bother with a screenshot of the frame errors, they were after the camera was detected and at times I even got an initial image, frozen or rgb/heatmap style.  Not currently tho.  They changed several times, but the consistent style was about frame rate or no frame detected.  Sorry I don't have the specific error right now.

I'll try cleaning out the existing install - and proceed with the steps here https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

After that, will try again.

RealSenseCustomerSupport, 2020-05-19:
----------------------------------------------

Hi,

Do you still require assistance with this?

Thanks

ev-mp, 2020-06-11:
@llwp , the device is enumerated properly, and so is the UVC v4l driver.
Please share the following info (requires LRS installed and `sudo apt install v4l-utils`):
```sh
ls /dev/video*
ls /dev/video* -1a | xargs -n1 v4l2-ctl --all -d
ls /dev/video* -1a | xargs -n1 v4l2-ctl --list-formats-ext -d
rs-enumerate-devices
sudo rs-enumerate-devices
```

llwp, 2020-06-12:
I wasn't seeing any device list under /dev/video* ... so I tooled with some packages, usb and rebooted until I had something showing up.

Good news is I didn't break anything.  Also, device went from "not detected" in cheese to just a black screen. Video is plain black / not detected otherwise.

Re-checked kernel log and it remains the same with one more entry (not sure if it is related, doubtful):
```
[delta+2s] perf: interrupt took too long (2512 > 2500), lowering kernel.perf_event_max_sample_rate to 79500
```
Not much system load frankly, so I thought it was interesting but probably inconsequential.

Under devices, there are 90 additional Stream Profiles listed under `sudo` .. and device information is now appearing (also a positive change from earlier since re-install).  Not sure what else might be useful here to be honest, so here everything is attached in the enclosed file.

Thank you!
[6352.txt](https://github.com/IntelRealSense/librealsense/files/4769162/6352.txt)

llwp, 2020-06-20:
@ev-mp .. any thoughts on this?

ev-mp, 2020-06-22:
@llwp, the provided log is perfectly correct - the camera is recognized and enumerated.

The other comments you post are often scattered between different domains so I find it hard to follow and combine into a clear picture of what is actually malfunctioning. 
I'd suggest to break it down into pieces and handle with increments - can you provide a snapshot or a log of the original issue ?

A side-note on `sudo` - if the camera is being recognized and can stream only with `sudo` then the `udev-rules` are either not applied or in incorrect/outdated version.

closed 2020-12-15 by RealSenseSupport