realsenseai/librealsense #6353: Visual studio property sheet location

BarRae, 2020-05-06:
Hello, I have a quick question.

Last time I made code for using my realsense d435i in visual studio 2019 I included some kind of property sheet that handled everything for me. I could start programming immediately.
Today I wan't to start on a new project, but I can't find that property sheet in the librealsense-master folder anymore.

Does anyone have a idea where this might/should be?

MartyG-RealSense, 2020-05-06:
On the Windows version of the SDK, the *intel.realsense.props** property sheet is in the root folder (default installation location **C: > Program Files (x86) > Intel RealSense SDK 2.0**

[image]

BarRae, 2020-05-06:
Found it, Thank you.

closed 2020-05-06 by BarRae