realsenseai/librealsense #6356: Camera intrinsics/extrinsics

timprepscius, 2020-05-06:
D435

I am running tests on my side.
I am not sure what I am seeing yet.  I am seeing something.
I need to do better measurements, which I'll start doing today or tomorrow.

I believe I am seeing some slight barrel distortion in the realsense full-spectrum-ir lenses.
The camera intrinsics for the the brown equations have 0 for all of the coeffs.
Is this correct?  Maybe I'm not setting some flag to get all of the coefficients.

Also, as to extrinsics.
I understand this is a bad question, but:
How confident are you in the factory provided extrinsics for the realsense?  (when I run the realsense-viewer-calibration-extrinsics-tried-all-options they don't change but-maybe-i-did-something-wrong).

MartyG-RealSense, 2020-05-06:
Hi @timprepscius  If you are doing your calibration in the RealSense Viewer with its 'On-Chip Self Calibration' function, this is a quick and lightweight calibration feature.  For a full advanced calibration of the camera, the **Dynamic Calibrator Tool** software should be used.

https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API

The Dynamic Calibrator at the above link is a free public version of the tool that only calibrates extrinsics.  There is also an OEM version of the software that calibrates both extrinsics and intrinsics, can calibrate multiple cameras and enables remote connection to cameras through a Linux server.  Unless you have an Intel Non Disclosure Agreement (NDA) account, the easiest way to obtain that software is to purchase the $1500 'OEM Calibration Target' board from the official online RealSense Store.  For 99% of RealSense users, the free public version of the software should suffice though, as it is calibration of extrinsics that has the most effect on image quality.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

In the majority of RealSense cameras, the factory calibration is of a high quality when the camera is taken out of the box after purchase.

timprepscius, 2020-05-06:
Ok, so the information I get from the linked threads and this thread are:

The coeffs are not 0, but I have to pay $1500 to have them not be zero.
Unless I do it myself.
Is this true?

I guess I will write an intrinsics calibrator.

Is the pattern for your dots published?  I guess I can take a picture at different distances.
I really wish I could change your dots pattern.  It would help me out a lot and make it so I don't need to attach external hardware to your cameras.

MartyG-RealSense, 2020-05-06:
My understanding is that you have the option to write your own calibration algorithm.  Reading downward from the comment linked below may provide useful insights into this process.

https://github.com/IntelRealSense/librealsense/issues/2704#issuecomment-446098909

If the coefficients are intrinsics and they are meant to all be '0' by default but they are not on your particular camera ... if you do not have the ability to calibrate the intrinsics then you could either write your own software, or do a 'gold reset' to reset the calibration to its original factory settings.

https://github.com/IntelRealSense/librealsense/issues/1588

Does the image below answer your question about the dot pattern please?

[image]

closed 2020-05-24 by MartyG-RealSense