realsenseai/librealsense #6358: unexpected behavior when reading bag file; can't read composite frames into memory

cpatdowling, 2020-05-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Ubuntu 18.04.4  | 
| Kernel Version (Linux Only)    |  5.3.0-46-generic                                      | 
| Platform                                 | PC Intel Core i7-7700  |
| SDK Version                            | liberealsense 2.34  | 
| Language                            | python 3.6.9 and 3.7.6     | 
| Segment			|  other   | 

### Issue Description

I have a ~20 second test video recorded using the Realsense-Viewer utility. No issues reading the bag file with the utility. I need to do some post-hoc computational analysis in python requiring the frames are aligned. I've found the easiest way to perform the alignment with pyrealsense2 is to read a sequence of composite frame objects into a list and performing the [alignment after reading the frames.](https://github.com/IntelRealSense/librealsense/issues/6330). I find that I can't read the composite frames into a list.

Using boilerplate code provided by the pyrealsense2 PyPI page, I would expect the follow code to satisfactorily read a sequence of composite frames into a list object:

```
# Setup:
pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
cfg.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)
cfg.enable_device_from_file("my_file.bag", repeat_playback=False)
profile = pipe.start(cfg)

i = 0
frames = []
try:
    while True:
        frameset = pipe.wait_for_frames()
        frames.append(frameset)
        i+= 1         
except Exception as err:
    print(err)
    pass

Cleanup:
pipe.stop()
print(i, " frames seen")
print(len(frames), " composite frame objects in memory")
return()
```
And rather than the expected 672 frames, the output I receive:
```
Frame didn't arrive within 5000
15  frames seen
15  composite frame objects in memory
```
I've seen related issues where users need to pause the playback in order to allow time for the process to acquire each frame. This does seem to work in some cases , however, I see conflicting behavior. I've written a sequence of test functions on this file and get disparate results on composite frames, whether not I use a pause. Then, when reading data, pauses are required. I've attached these test functions and their results are below, with identical results in python 3.6 and 3.7. What I don't understand is that I can run wait_for_frames(), and the process "sees" each frame without pausing as in the first example, but operating on the frame or reading it into memory causes difficulty.

```
def loop_through_frames():  #this one seems to work as intended
    # Setup:
    pipe = rs.pipeline()
    cfg = rs.config()
    cfg.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
    cfg.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)
    cfg.enable_device_from_file("my_file.bag", repeat_playback=False) #approx 20 second video
[... 329 lines cut ...]
            playback.resume()
            
    except Exception as err:
        print(err)
        pass

    # Cleanup:
    pipe.stop()
    print(i, " frames read")
    print("aligned color frame shape: ", color_frames[0].shape)
    print("alighed depth frame shape: ", depth_frames[0].shape)
    print(len(color_frames), " aligned color frames in memory")
    print(len(depth_frames), " aligned depth frames in memory")
    return()

==========
Reading depth and color data with alignment process into list with playback pausing
Error occured during execution of the processing block! See the log for more info
16  frames read
aligned color frame shape:  (480, 640, 3)
alighed depth frame shape:  (480, 640)
16  aligned color frames in memory
16  aligned depth frames in memory
==========
```

MartyG-RealSense, 2020-07-05:
Hi @cpatdowling   I'm very sorry that you did not receive a response on this case.  Were you able to find a solution please, and if not do you still require assistance with this problem?  Thanks!

cpatdowling, 2020-07-13:
@MartyG-RealSense Haven't found a solution yet, no. Thanks for following up; I'm having to step back and use the C libraries, but offline data processing seems like a Python-appropriate task.

MartyG-RealSense, 2020-07-14:
@cpatdowling  If you are doing short recording sessions of 20 to 30 seconds, an approach that may suit you is an SDK function called **Keep()**.  It allows you to store frames in memory instead of writing them to disk.  Once the pipeline is closed, you can then perform post-processing and align on the memory-stored frames and save them to file.

Because Keep() stores the frames in memory, it can use up memory capacity quickly (hence the limited recording time) and so is more suited to desktop / laptop computers than low-power computing devices.  I see that you have an Intel Core i7 computer though.  If it is a relatively modern computer, it should therefore have enough memory for a 20 second sequence.

Some scripts for using Keep() with Python are in the link below.

https://github.com/IntelRealSense/librealsense/issues/6164#issuecomment-607714808

closed 2020-08-08 by MartyG-RealSense