realsenseai/librealsense #6359: Intel® RealSense™ Viewer can not find device

LYKlyk, 2020-05-07:
My device is Realsense SR300 and OS is window10. I install SDK 2.34. I try to use Intel® RealSense™ Viewer to show image, but it always show can not find device. How can i do? Thank you.

RealSenseSupport, 2020-05-07:
Hi @LYKlyk 
Could you check if the SR300 is enumerated under wiin10 device manager?
you should see Intel realsense camera SR300 Depth and Intel realsense camera SR300 RGB
enumerated under Device manager->Cameras

LYKlyk, 2020-05-07:
Thank you. I can see "Intel realsense camera SR300 Depth" and "Intel realsense camera SR300 RGB" under Device manager->Cameras.  When I use "intel_rs_sdk_mw_hand_hybrid_11.0.27.1404.exe"(https://software.intel.com/content/www/us/en/develop/articles/realsense-sdk-windows-eol.html), it does not work. When I use "Hand Viewer", it show "Init failed."  When I use "Raw Stream", it does not show rgb/depth image.

jakemin, 2020-05-27:
SR300 Virtual Driver should be recognized in the imaging device.
If there is no SR300 virtual driver, the device does not operate normally.
[image]

RealSenseSupport, 2020-05-29:
Hi LYKlyk,
as Jakemin said to run the old version of SDK  intel_rs_sdk_mw_hand_hybrid_11.0.27.1404.exe
you will need to install the SR300 DCM and see the virtual driver is installed
https://downloadcenter.intel.com/product/92329/Intel-RealSense-Camera-SR300

for new SDK that published on github, there is no need to install DCM
basically, the viewer should able to streaming the SR300

closed 2020-07-25 by RealSenseSupport