realsenseai/librealsense #6360: 安装程序无法检测到此系统中的英特尔® 实感™ 三维摄像头。安装已终止。

LYKlyk, 2020-05-07:
My device is Realsense SR300, my os is window10. When I try to use "intel_rs_dcm_sr300_3.3.27.5718.exe" to install driver, it show "安装程序无法检测到此系统中的英特尔® 实感™ 三维摄像头。安装已终止。(Setup cannot detect Intel realsense  in this system Gamma 3D camera. Installation terminated.)"

RealSenseSupport, 2020-05-07:
Hi @LYKlyk 
if you use Librealsense SDK, you don't need to install the SR300 DCM
and please check if the SR300 is enumerated under device manager

MartyG-RealSense, 2020-05-07:
If you have an internet connection and Windows then the DCM program should not be required, as the drivers should have automatically been downloaded and installed through Windows Update when the camera is inserted into the USB port. 

 Intel introduced this auto-install feature for the SR300 in 2017, a year after the camera launched.  So the DCM was no longer needed, and the program can also have compatibility problems with modern versions of Windows 10.

LYKlyk, 2020-05-07:
@MartyG-RealSense @RealSenseSupport Thank you. I can see "Intel realsense camera SR300 Depth" and "Intel realsense camera SR300 RGB" under Device manager->Cameras. When I use "intel_rs_sdk_mw_hand_hybrid_11.0.27.1404.exe"(https://software.intel.com/content/www/us/en/develop/articles/realsense-sdk-windows-eol.html), it does not work. When I use "Hand Viewer", it show "Init failed." When I use "Raw Stream", it does not show rgb/depth image.  Could you provide some suggestion?

MartyG-RealSense, 2020-05-07:
Are you using the 2016 R2 SDK (the 1.8 gb download one) or 2016 R3 (the modular installation one) please?

LYKlyk, 2020-05-07:
I am using "Downloads"->"2016 R3 Individual Components"->"Full Hand Tracking and Gesture Recognition". (https://software.intel.com/content/www/us/en/develop/articles/realsense-sdk-windows-eol.html)

LYKlyk, 2020-05-07:
@MartyG-RealSense I am using "Downloads"->"2016 R3 Individual Components"->"Full Hand Tracking and Gesture Recognition". (https://software.intel.com/content/www/us/en/develop/articles/realsense-sdk-windows-eol.html)
When I use "Hand Viewer", it show "Init failed." When I use "Raw Stream", it does not show rgb/depth image.   Thank you.

MartyG-RealSense, 2020-05-07:
With the 2016 R3 SDK, you have to download and install the 'Essentials'module first before installing the other optional modules such as Hand.  Did you install the Essentials module first please?

To be honest, the all-in-one R2 SDK really is the best way to use the 2016 SDK as it makes things far simpler.

LYKlyk, 2020-05-07:
@MartyG-RealSense I have install 'Essentials' module. 2016 R2 SDK can provide "Hand Viewer"? I want to get the skeleton of hand.

MartyG-RealSense, 2020-05-07:
2016 R2 installs the entire SDK including hand tracking.

Does your PC have an Intel Core processor please?  Unlike the modern RealSense SDK 2.0, the 2016 SDK is very strict in the processors that it will work with.  Other processor types may work but there is a much greater chance of having problems.  Init Failed can occur if the processor is incompatible.

LYKlyk, 2020-05-07:
@MartyG-RealSense Thank you. I am use USB 3 port. When I use 2016 R2 SDK, do I need to install DCM indenpently?

MartyG-RealSense, 2020-05-07:
You should not need to use the DCM, and it seems that you have the drivers correctly installed.

Another cause of failure of camera detection of getting the Init Failed error can be if the camera disconnects because it is not getting enough power from the USB port.  An indicator of this is that when you start a depth stream, the green light on the front of the camera will turn on for a second and then go out again.  The green light should be on all the time if depth streaming is working correctly.

LYKlyk, 2020-05-07:
@RealSenseSupport @MartyG-RealSense When I use 2016 R2 sdk, it is still does not work. "Intel RealSense SDK Sample Browser"->"Tool"->"Camera Eplorer" show "No device connected."

MartyG-RealSense, 2020-05-07:
My main suspicion would be that there is an incompatibility with the CPU processor that your computer uses.  The 2016 SDK (of which the Camera Explorer tool is a part) prefers Intel Core processors, and for the SR300 an additional condition is added that a 6th Generation Intel Core processor or newer is preferred.

The best non-Core processor that has a chance of working with the 2016 SDK is Intel Xeon, and even then there is only a 50% chance at best of it working.  For other processor types, the percentage odds of the camera working with them are much lower.

LYKlyk, 2020-05-07:
@MartyG-RealSense My device is Intel(R) Core(TM) i7-6500 CPU,  Whether the CPU processor used by the computer is compatible？

MartyG-RealSense, 2020-05-07:
Your CPU should be compatible with SR300 then as it is 6th Generation (indicated by the '6' at the start of "6500".  Thanks for the information!  

This increases the chances that the problem is related to the camera receiving insufficient power on the USB port.  In these circumstances, the drivers may appear in the Device Manager but the camera cannot be detected or disconnects as soon as a program tries to activate the camera streams.  Using a mains electricity powered USB hub with the camera instead of plugging the camera directly into the computer or an "unpowered" hub (a hub that draws its power from the computer) can help provide a stable USB connection in this situation.

MartyG-RealSense, 2020-05-24:
Do you require any further assistance with this case or can it be closed, please?   Thanks!

closed 2020-05-30 by MartyG-RealSense