realsenseai/librealsense #6361: RGB format convertion on Jetson Nano

7oud, 2020-05-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 0.5.12.03.00 | 
| Operating System & Version |   UBuntu 18.04 for Tegra  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                        | 
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            |  2.23.1                         | 
| Language                            |  C++/opencv                          | 

### Issue Description

- When using Intel RealSense Viewer, the color of YUYV stream more accurate than RGB stream. the images of RGB stream look redder. how the Intel RealSense Viewer convert YUYV stream

- After getting YUYV stream using C++, call `yuy_decoder.process()`, it also gives a redder image and the convertion cost about 15ms, but if  calling `cv::cvtColor` of opencv, it gives a accurate color image, just like using RealSense Viewer with YUYV stream, and the convertion costs 5 to 8ms. The convertion to RGB with `yuy_decoder` should using CUDA, but it is slower than convertion with opencv, which only use CPU. Maybe the memory transfer between host and device leads to this?

MartyG-RealSense, 2020-05-07:
The documentation for **yuy_decoder.process()** says that it automatically uses CUDA "where available".  I wonder if this is dependent on Librealsense having been built with CUDA support.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1yuy__decoder.html#a2754499ae0f8fa1892ef33fe6bdc4313

If you are installing Librealsense using the Debian packages, CUDA support should already be included.  Otherwise, Librealsense can be built with CMake to have CUDA support by including the **BUILD_WITH_CUDA** flag in the CMake build instruction.  

https://github.com/IntelRealSense/librealsense/issues/4816#issuecomment-529195322

dorodnic, 2020-05-07:
Hi @7oud 
Thank you for the feedback. Interesting... 
Could you please share more info about item 1. You mean YUY render in the Viewer vs RGB in the Viewer? Same exact hardware config? Perhaps there is inconsistency between CUDA and GLSL implementations. Please share what you see.
Regarding second item, GPU is not guarantied to be always faster than CPU. The reason we use CUDA on Jetson, is mainly because our CPU implementation on Arm is not optimized. OpenCV clearly is. We decided not to add opencv as a dependency to simplify installation.
The team did some work on this problem, and our plan is to use [libyuv](https://chromium.googlesource.com/libyuv/libyuv/) to get good CPU performance on Arm.

7oud, 2020-05-07:
@dorodnic 
yes, I mean [YUY] render in the viewer vs RGB.
[image]
[image]
The two images(cropped) are captured with viewer,  both resolution 1920x1080, 30FPS, without Auto Exposure, without Auto White Balance, without Post Processing, the only difference is stream, one is YUYV, the other is RGB8. 
If save the YUYV stream `.bag`, and convert to RGB with `yuy_decoder` or `opencv`, it gives similar result(RGB8 stream renders redder).
Some HW info, Jetson [Nano](https://developer.nvidia.com/embedded/jetson-nano), anything else?

MartyG-RealSense, 2020-05-24:
Do you require any further assistance with this case or can it be closed, please?   Thanks!

7oud, 2020-05-25:
@MartyG-RealSense You can close this issue, it is not serious problem and may be fixed in future update.

MartyG-RealSense, 2020-05-25:
Okay, thanks very much!

closed 2020-05-25 by MartyG-RealSense