realsenseai/librealsense #6362: T265 stops sending pose data if run together with D435

dschnabel, 2020-05-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 & D434 | 
| Firmware Version                   | T265: `0.2.0.926`, D435: `05.12.03.00` | 
| Operating System & Version |  Raspbian 10.3 | 
| Kernel Version (Linux Only)    |  4.19.97-v7l+ armv7l | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |  2.33.1                          | 
| Language                            |  C++/ROS                          | 
| Segment			|  Robot                         | 

### Issue Description

After upgrading from **SDK 2.31.0** to **2.33.1** my T265 stops sending pose data when running for some time (1-5 min). This only happens if the T265 is run together with the D435. If I run the T265 alone, I don't see this problem.

This wasn't a problem in **2.31.0** but I was experiencing crashes in **2.31.0** so I decided to upgrade to the latest stable version (2.33.1).

In **2.33.1** I see no more crashes but I seem to be experiencing a very similar problem as described in https://github.com/IntelRealSense/librealsense/issues/5509#issuecomment-587422436.

I saw some [discussion](https://github.com/IntelRealSense/librealsense/issues/3434#issuecomment-476290706) on missing serial numbers but in my case both cameras seem to get detected properly (see my ROS startup logs for [T265](https://gist.github.com/dschnabel/875cba6b89ca7dcb03a817b956eea90d) and [D435](https://gist.github.com/dschnabel/57ebf221d1acc8ecbe53ff073a37616d)).

When I monitor the depth topic from the D435 and the odom topic from the T265, after a while I don't get any more pose (odom) data:

```
$ rostopic hz /d435/aligned_depth_to_color/image_raw /odom

                topic                     rate   min_delta   max_delta   std_dev    window
==========================================================================================
/d435/aligned_depth_to_color/image_raw   14.9    0.03358     0.1134      0.004032   3271  
/odom                                    200.4   2.384e-05   0.08126     0.0148     3271  

                topic                     rate   min_delta   max_delta   std_dev    window
==========================================================================================
/d435/aligned_depth_to_color/image_raw   14.9    0.03358     0.1134      0.004024   3286  
/odom                                    200.4   2.384e-05   0.08126     0.0148     3286  

                topic                     rate   min_delta   max_delta   std_dev    window
==========================================================================================
/d435/aligned_depth_to_color/image_raw   14.9    0.03358     0.1134      0.004016   3301  
/odom                                    200.3   2.384e-05   0.08126     0.0148     3301  

<... no odom data hereafter ...>

                topic                    rate   min_delta   max_delta   std_dev    window
=========================================================================================
/d435/aligned_depth_to_color/image_raw   14.9   0.03358     0.1134      0.004049   3317  

                topic                    rate   min_delta   max_delta   std_dev   window
========================================================================================
/d435/aligned_depth_to_color/image_raw   14.9   0.03358     0.1134      0.00404   3332  

                topic                    rate   min_delta   max_delta   std_dev    window
=========================================================================================
/d435/aligned_depth_to_color/image_raw   14.9   0.03358     0.1134      0.004031   3347  
```

For completeness here are my two ROS launch files:
* [rovy_d400_and_t265.launch ](https://gist.github.com/dschnabel/1aa9a588766a85ffed138f485a14bb19)
* [rovy_d435.launch](https://gist.github.com/dschnabel/b4d70ab7673acb09ef2557d363ab10e1)

Is there any fix for this?

dschnabel, 2020-05-11:
@RealSenseCustomerSupport @RealSense-Customer-Engineering @MartyG-RealSense 

Can you help us? Is this a known issue and if so, is there a workaround or proposed fix? Any additional information we can collect for you too look at?

This problem renders the T265 unusable for our project.

dschnabel, 2020-05-19:
We did some more investigation and it looks like a firmware or a libusb issue.

This is how the pose/gyro/accel is being received from the T265 in librealsense:

1. A new initial [request](https://github.com/IntelRealSense/librealsense/blob/master/src/tm2/tm-device.cpp#L1321-L1322) is generated.
2. This request gets [forwarded to libusb](https://github.com/IntelRealSense/librealsense/blob/master/src/libusb/messenger-libusb.cpp#L87).
3. When a response arrives the [interrupt_callback](https://github.com/IntelRealSense/librealsense/blob/master/src/tm2/tm-device.cpp#L1260) gets called.
4. At the end of the callback function a new [request](https://github.com/IntelRealSense/librealsense/blob/master/src/tm2/tm-device.cpp#L1318) is created.

Steps 2-4 are repeated. As long as the callback function `_interrupt_callback` gets called we receive data. But for some reason, occasionally the callback doesn't get called anymore, even if the request has been dispatched to libusb correctly without any errors. This is when the T265 stops sending pose/gyro/accel data.

We implemented a watchdog thread which monitors the callback and if the callback is not called anymore after some time we call [stop_interrupt()](https://github.com/IntelRealSense/librealsense/blob/master/src/tm2/tm-device.cpp#L1326) followed by a [start_interrupt()](https://github.com/IntelRealSense/librealsense/blob/master/src/tm2/tm-device.cpp#L1254). We were hoping this would reinitialize the callback and cause the data to show up again. But this didn't help. Only a relaunch of the ROS node helped.

Unfortunately we cannot go deeper since we don't have the firmware of the T265. But our guess is that the T265 just wouldn't respond anymore.

@RealSenseCustomerSupport @RealSense-Customer-Engineering @MartyG-RealSense could you please see what could cause the callback function not getting called anymore? Are there logs I can enable in the T265 firmware?

dschnabel, 2020-05-25:
We ran the ROS node in debug mode. These are the last few lines before T265 stops sending data:
```
[DEBUG] [1590375454.268119176]: Publish Pose stream
[DEBUG] [1590375454.272547826]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1590375454.272984024]: Frame arrived: stream: Pose ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1590375454.273135615]: Publish Pose stream
[DEBUG] [1590375454.273732108]: Frame arrived: stream: Accel ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1590375454.278416105]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1590375454.278897526]: Frame arrived: stream: Pose ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1590375454.279104097]: Publish Pose stream
[DEBUG] [1590375454.286298009]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1590375454.286856576]: Frame arrived: stream: Pose ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1590375454.287164961]: Publish Pose stream
[DEBUG] [1590375454.287998580]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1590375454.288404798]: Frame arrived: stream: Pose ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1590375454.288710257]: Publish Pose stream
[DEBUG] [1590375454.289378008]: Frame arrived: stream: Accel ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1590375454.292816706]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
```
After the last line there's no more output.

csr-kick, 2020-05-31:
I have also been seeing this exact issue, with the same setup. RPI4 with same OS, SDK version, with T265 and D435i. Happens with both ROS and simple C++ test scripts. 

Data comes in from the T265 (using `poll_for_frames`, `try_wait_for_frames`, or `wait_for_frames`) until suddenly the frames just stop. Relaunching whichever program will get data from the device(s) just fine for another short period of time (but I do call `hardware_reset()` on the T265 as part of my code). The only data stream I am getting from the T265 is pose via: `enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);`

Other Observations:
- Rpi temperature does not correlate with duration before data stops
- Neither using different cables nor using a powered USB hub helps
- With both devices plugged in and streaming data, the T265 does seem to fail sooner than just the T265 alone, but that's not always the case. When T265 fails, the D435i continues to function normally.
- Oddly, fails faster if you just let the sensor sit in one place, but takes longer if you move the device around. The longest 'uptime' I have seen with the sensor was from moving it a little every 60 or so seconds.
- USB 2.0 or 3.x makes no difference

Personally have not had any success with `pipe.stop()` and `pipe.start()`. I tried to see if there was a usb disconnection or such that would account for this issue, but the log file attached does not show any events to support that. It really feels like there is something that crashes or hangs on the T265.

This problem is currently a showstopper for me, which is unfortunate since during the windows when it works the sensor is pretty great. I might try rolling back to 2.31, but based on what I read here and the release notes that may just be trading one set of problems for another. Please help.

[udev log.txt](https://github.com/IntelRealSense/librealsense/files/4706934/udev.log.txt)

csr-kick, 2020-05-31:
I switched to the development branch and it seems like the T265 is working correctly there. The downside now is that the D435i is not recognized. It shows up properly as a usb device: 
```
Bus 002 Device 004: ID 8086:0b3a Intel Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 012: ID 8087:0b37 Intel Corp.
Bus 001 Device 004: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
but the SDK refuses to detect it. The rs-enumerate-devices example does not see it, nor do tools like the fw-logger/fw-update. 

I switched to the `rc` branch in case the pull request [#6480](https://github.com/IntelRealSense/librealsense/pull/6480) that was merged in was the problem, but nothing. Maybe somewhere in the merge history there is a commit that will allow both to function.

As a side note, both sensors have always worked when plugged into a windows computer.

csr-kick, 2020-06-02:
Using latest development branch at commit [2a79d31](https://github.com/IntelRealSense/librealsense/commit/2a79d31599910ee2799e2bfe7ea1459e483d8c5a) merged by @ev-mp 

Compiled and ran the `rs-pose` example with the line `rs2::log_to_console(RS2_LOG_SEVERITY_INFO);`

Attached is the log, failure after about 60 seconds of running.  The message received after failure is:

`02/06 01:04:05,676 INFO [2949632528] (tm-device.cpp:1408) T265 FW message: 13042424847758289392: [0x/2:721] Host not reading - stopping`

[Pose_Info_log_output.txt](https://github.com/IntelRealSense/librealsense/files/4714005/Pose_Info_log_output.txt)

Anyone have any ideas?

dschnabel, 2020-06-02:
Glad to hear I'm not the only one with this problem, @csr-kick. When you tried with the latest dev branch and you got the `Host not reading - stopping` error, did you have the D435 run in parallel or only the T265?

And is it still true that a `pipe.stop()`/`pipe.start()` won't fix it once you get into that state?

csr-kick, 2020-06-02:
I can replicate the failure without the D435i plugged in, but it does seem to make it happen faster when it is running. The data I am collecting now is as bare as possible, running the rs-pose example with only the T265 plugged in at all to the Rpi4. It has been a while since I tried the pipe start/stop, let me try again and I will add in the results.

csr-kick, 2020-06-02:
@MartyG-RealSense @dorodnic Any ideas/help on this issue? 'Host not responding' error makes me think the SDK stops processing messages or maybe requesting the message in the first place, so the sensor stops as well.

dschnabel, 2020-06-02:
@csr-kick I was just thinking, what if you create a new pipe after stopping the old one? This should release any resource that the old pipe holds:
```
pipe->stop();
pipe = std::make_shared<rs2::pipeline>();
pipe->start();
```

I'm not home right now so I can't test this myself.

csr-kick, 2020-06-03:
Using your code snippet does recover, but I want to try it a few times with the D435i also running. However, this isnt a viable solution beyond basic prototyping.

1. It takes 15 seconds for that error to be thrown when using `pipe.wait_for_frames`. Maybe one of the other frame wait methods will be able to fail faster, but any real delay will be a problem for a mobile robot.
2. When it resets, your back at 0,0,0 meaning you would maybe have to track you actual last pose and then offset after each reset?
3. How will exporting and importing the localization maps play with this? If you are far from the origin will they actually be able to re-localize? Do you then drop your manual offset after?

It really seems like somewhere in the SDK it stops processing the pose data coming from the sensor. The sensor sees the messages backing up and it stops sending to avoid any issues. The bulk data transfers and time syncing continue to work, which tells me that it is specific to whatever protocol the pose message uses. We also know that none of this happens if you plug into a windows computer and run the realsense viewer app, so its probably low in the code stack.

dschnabel, 2020-06-03:
I agree that this is not a fix and ultimately we need something better than restarting the pipe. I think the workaround can work fairly well though if configured properly but we need to do some more testing.

To your questions:

Re 1: `pipe.wait_for_frames()` takes a timeout in milliseconds as an optional parameter (see [here](https://github.com/IntelRealSense/librealsense/blob/master/src/pipeline/pipeline.cpp#L222)). So you can override the default 15s timeout with whatever makes sense to you.

Re 2: I might be wrong on this but the way I understand is when you reset the pipe, you are only reconnecting to the T265, you're not hard resetting the camera. In order to hard reset the T265 you'd have to call [hardware_reset()](https://github.com/IntelRealSense/librealsense/blob/master/src/tm2/tm-device.cpp#L2016). That means your camera will keep the last pose even after you've reset the pipe.

Re 3: Since the T265 didn't hard reset you shouldn't need to re-import the localization map.

Would be nice to have someone from Realsense chime in on this to make sure I'm not mistaken.

csr-kick, 2020-06-03:
Thanks, forgot about the timeout. Make sure that you set it initially to a high value to allow the sensor boot up or whatever before dropping the timeout down for normal operations. Then in the catch statement put it back up. My current testing code:

```
rs2::log_to_console(RS2_LOG_SEVERITY_DEBUG); //verbose, can make this _INFO for less

rs2::config cfg;
cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
auto pipe = std::make_shared<rs2::pipeline>();
pipe->start(cfg);
unsigned int to = 20000;
while (true)
{
    try {
            
        auto frames = pipe->wait_for_frames(to);
        to = 40;
        auto f = frames.first_or_default(RS2_STREAM_POSE);
        // Cast the frame to pose_frame and get its data
        auto pose_data = f.as<rs2::pose_frame>().get_pose_data();
    
        // Print the x, y, z values of the translation, relative to initial position
        std::cout << "\r" << "Device Position: " << std::setprecision(3) << std::fixed << pose_data.translation.x << " " <<
            pose_data.translation.y << " " << pose_data.translation.z << " (meters)" << std::endl;
    }
    catch (const rs2::error& e) {
        to = 20000;
        std::cout << "T265 crashed. Restarting ..." << std::endl;
        pipe->stop();
        pipe = std::make_shared<rs2::pipeline>();
        pipe->start(cfg);
        std::cout << "... restarted." << std::endl;
    }
}
```
My testing shows that (without loading any maps right now) localization does go back to 0,0,0. 

@RealSenseCustomerSupport can we get any sort of statement about this problem?

csr-kick, 2020-06-05:
Still working to see if there is some way to actually make the T265 work with the RPI.

I have tried setting up ubuntu 18 on the rpi, and same problem, T265 will eventually stop responding, and it happens in a matter of seconds with the D435 sending depth at 640x360x30fps. 

@MartyG-RealSense please can we get some sort of response from intel on this issue.

MoBaT, 2020-06-09:
Just wanted to put out there that I'm having the exact same problem and am also restarting the pipeline on failure. Would love a fix to this @MartyG-RealSense

m4xr1sk, 2020-06-15:
Same problem here with the SDK v2.35.2
I also would like to have a fix because otherwise this product is not usable at all.

alanypf, 2020-07-10:
Hi all, 
I have the same problem.
I am using Ubuntu 18.04 server. Librealsense is built from the source with the LTS kernel patch.
The data link broke after a minute or so. The problem only exists on raspberry pi4 for me since if I plug the same SD card into a raspberry pi 3b+, the data links remain healthy. And on raspberry pi 4,  neither USB2 nor USB3 ports work.
I hope this helps the developer team.

Thanks.

RealSenseSupport, 2020-09-23:
Thanks for highlighting this. At this time we have moved our focus to our next generation of products and consequentially will not be addressing this T265 issue.

dschnabel, 2020-09-23:
@RealSenseSupport are you dropping support for the T265? I don't see any mention on https://www.intelrealsense.com/tracking-camera-t265

ArkadiuszNiemiec, 2021-01-04:
Same issue here, I did not find an information about T265 not being supported.

Twentystudios, 2021-03-28:
If T265 won’t be supported anymore (just bought one), what is the direct replacement for it?

csr-kick, 2021-03-31:
> But I may find out the solution. The root cause is "cout" or "printf" function. I cannot explain, but it works. You could remove all of your "cout", "printf" to the console. And this problem is gone. It solves my problem. And it may solve yours too.
> 
> My problem is due to the console window of Windows 10. Its default mode is Quick_Edit. the console pauses the program if you select any texts on the console screen.
> 
> Have fun,

Appreciate the thought, but its a bit different for different hardware. From my experience the root issue with the T265 is down to timing. There is a mismatch between the computer's USB messaging and the device's where the computer polls from the device, but the device doesn't notice. Then, the computer sits around waiting for a message that will never arrive, until it times out. Further supporting evidence is that bulk messages will still pass between the device and computer while it waits for the interrupt to return. I would imaging that using cout or printf for you is actually changing those timings, but I can confirm the problem persists on the rpi regardless. There are a few things you can do knowing this to improve results, but I have not been able to eliminate its occurrence, since we only have access to one side of the code. 

Sadly, this looks to be one of those problems that wouldn't show up in the limited testing they would have put the T265 through before putting it on sale, and if we had access to the firmware could probably be solved by someone knowledgeable in the USB protocol. Lets hope this bug doesn't carry forward into future products.

RealSenseSupport, 2021-04-09:
Hi @dschnabel 

Curious if this pre-release build resolves your issue. 

https://github.com/IntelRealSense/librealsense/releases/tag/v2.44.0

Particularly because it integrates this PR https://github.com/IntelRealSense/librealsense/pull/8561 that addresses a T265 specific race condition.

Thanks

dschnabel, 2021-04-09:
Thanks @RealSenseSupport

As far as I can tell PR #8561 addresses cases where the pipeline stop procedure gets stuck. But the problem I have is that the T265 becomes unresponsive during normal operation even without `pipeline.stop()` being called.

So the PR seems unrelated to this issue. Or am I missing something?

Based on my investigation in https://github.com/IntelRealSense/librealsense/issues/6362#issuecomment-630577088 the problem seems to be deeper than the librealsense library, possibly a firmware or a libusb issue.

Twentystudios, 2021-04-09:
@RealSenseSupport I would really appreciate some clarification regarding Intel dropping support for the T265. Is this really the case? It was released just two years ago and is still being sold. Will there be a direct replacement for it?

JoseDinis1, 2021-04-10:
https://github.com/IntelRealSense/librealsense/issues/8779

nixinator, 2021-04-12:
I'm using a pi4 with the T265 and having the exact same problem. I'm reading from are not doing fixes or support the T265. I think this is a 'we cannot fix it'. If it's not a problems with the software/hardware stack on the pi4, or the actual driver in the real-sense SDK, then it will require a firmware fix from intel.

We've got the T265 working fine on the Nvidia Xavior board, so i'm going to see what software differences there are between to the two systems. It may turn out this is something more subtle with the PI4's USB?

But, come on intel, the PI4 for robot development, don't leave us high and dry for 'newer expensive products'.

silverjoda, 2021-07-04:
Hello, I had the same problem with my Pi4 and T265. What fixed it for me is going back to a december 2019 github commit, somewhere around version SDK 2.19 perhaps. Using this older SDK version It doesn't cause the T265 to stop after a few minutes.
I found out about this after digging for hours and seeing a mention of one of the devs to roll back.

nixinator, 2021-07-04:
intel + closed!

Business as usual.

sigh.

Construkted-Reality, 2021-07-20:
@silverjoda what alternatives to the T265 do you know of.
We don't have much choice which is why we are still trying to figure this out.

silverjoda, 2021-07-21:
> @silverjoda what alternatives to the T265 do you know of.
> We don't have much choice which is why we are still trying to figure this out.

To be honest I don't know. I want to try Orb-slam or some other monocular variant when I get the time.

zhouzhiwen2000, 2021-08-02:
> > @silverjoda what alternatives to the T265 do you know of.
> > We don't have much choice which is why we are still trying to figure this out.
> 
> To be honest I don't know. I want to try Orb-slam or some other monocular variant when I get the time.

That requires quite a bit of CPU power.

siddharthcb, 2021-08-06:
@dschnabel same issue here.. Sad to see this issue still floating even after an year. @RealSenseSupport There is no point selling the products if you cannot take the responsibility of fixing issues. Atleast putting an effort to solve is deeply appreciated.

nixinator, 2021-08-06:
yep, open source can be a bit of finger point exercise, your firmware is incorrect, No! the rpi usb drivers are wrong! NO!, the how debian package the usb subsystem is in error! No!, a kernel regression caused the problem!  So, it's not as simple as that, i think if it was easy to fix, 'they' would. However, contacting 'they' , i.e. not a department but a 'person' who is responsible for is probably impossible. i'm not sure if that's by design or on purpose.

I could probably get the bottom of this, but i'm not going to do it for free, and if you factor in the cost of reverse engineer and getting into the weeds of all the subsystem involved, then it's probably just cheaper to upgrade to the latest and greatest piece of hardware, and take the T265 as a release hobbyist prototype... 

we've decided to use other ways to localize the robot for now. If someone wants to offer a bounty to get this working on the rpi, then i'm open to that and so will others here.

silverjoda, 2021-08-07:
So what other way did you decide to localize the robot?:)  On our big robots we use Ouster lidars with ALOAM and other mappers which works fine, but for small prototype robots I can't use any of those because there is no space or budget to put a full system.

yanboli, 2021-08-18:
> @RealSenseCustomerSupport @RealSense-Customer-Engineering @MartyG-RealSense
> 
> Can you help us? Is this a known issue and if so, is there a workaround or proposed fix? Any additional information we can collect for you too look at?
> 
> This problem renders the T265 unusable for our project.

Intel Says It’s Shuttering RealSense Camera Business. I guess there is no hope for T265 any more. Find another way ASAP.

https://www.crn.com/news/components-peripherals/intel-says-it-s-shuttering-realsense-camera-business

Construkted-Reality, 2021-08-18:
What are the alternatives to affordable computer vision and depth camera systems like the ones from Intel?

MartyG-RealSense, 2021-09-02:
I am now able to provide information about the future of RealSense, as an 'End of Life' notice has been released by Intel.  Although some RealSense models are being discontinued, the 400 Series stereo depth cameras are continuing.  The official PDF of the End of Life notice explaining the changes is attached below.

[PCN118463-00.pdf](https://github.com/IntelRealSense/librealsense/files/7100557/PCN118463-00.pdf)

dschnabel, 2021-09-02:
Thank you for this update @MartyG-RealSense.

As the T265 will be de-supported, can at least the source code of the firmware be made public? Even if Intel isn't interested in fixing this bug, many other in the community will be eager and capable of tackling this problem if we just had access to the source code.

MartyG-RealSense, 2021-09-02:
Hi @dschnabel   Whilst the RealSense SDK is open-source and developers are free to 'fork' it to create their own custom versions that they host on their own GitHub, Intel do not open-source RealSense's firmwares or camera algorithms.

nixinator, 2021-09-02:
i think you just highlighted the problem... maybe i can ask 'do not', to 'to the reason why'?

MartyG-RealSense, 2021-09-02:
@nixinator  A reason given in the past has been that such resources are Intel's intellectual property.

nixinator, 2021-09-03:
root cause analysis...complete!

zhouzhiwen2000, 2021-09-03:
> @nixinator  A reason given in the past has been that such resources are Intel's intellectual property.

If you can't release the source code, can you solve this issue before dropping support? Perhaps you could release part of the source code?

MartyG-RealSense, 2021-09-03:
@zhouzhiwen2000  Development work for the T265 has ended so no further software or firmware updates can be provided for it.

csr-kick, 2021-09-04:
> @MartyG-RealSense Development work for the T265 has ended so no further software or firmware updates can be provided for it.

This is why no one will ever trust the depth sensors that are still around. Companies cannot base their products on a device where the creator has no 'anchor' product or vested interest in it's success and support. If you have an issue and wake up the next day to find that the supplier has decided to simply walk away rather than do the right thing for their customers, your existence is at risk while theirs is not.

HanDaSeul, 2021-09-13:
I think I solved the problem, which the T265 suddenly stops with RPi using ROS where the RPi is 8GB.
The problem is neither the T265 nor the software(SDK, ROS whatever). It's the lack of power output for USB3.0 port on the RPi.
I guess most of you guys plug at least two items to the RPi usb ports, especially wireless usb wifi adapter and theoretically, there will be no problem in case of power consumption, but it has problem. I think because of power consumption is too high so one of them disconnects. Or you just using RPi's wifi chip, you need to consider it.  
Here's how did I fixed: You need 1 powered usb hub. Some cheap powered usb hubs still have problems, but in my case i used 'ugreen usb 3.0 hub Model name: cr113' with external power. Plug all of USB devices to that usb hub, except t265. Only the T265 is connected to the RPi directly. I guess you can also plug to that usb hub but there's another problem 'when boot up you have to unplug and plug for working'. My T265 is directly connected to RPi using 'uhubctl' so I don't need to unplug for detection without unplug. 
So, My RPi's usb port have t265 and usb hub. And usb hub have wireless usb wifi adapter and RPLidar A1.  My usb hub takes 1A current with 5V from external power. RPi's OS is gnome-session but not fully installed ubuntu-desktop. I tested how long can RPi handle without losing connection on both ROS and ROS2.  And the result is both are fine! On the ROS, still keeps connected after 50 minutes and ROS2, I just disconnected after 30 minutes but in ROS2 there's little bit of lag but I don't think that's related to the power. 
Hope it helps you guys!

HanDaSeul, 2021-09-13:
And in the case of the problem unplug then plug after first boot up, why don't T265 try turn off then on when launch the node? For example, I saw RPLidar a1 keeps rotate and when I launch the node RPLidar a1 on ROS, it stops rotate for 1~2 seconds and rotate again. Even that Lidar connected to the usb hub, still works.

silverjoda, 2021-09-13:
After plugging in and out  about 200 times on the Rpi, I found out that you can use Uhubctl to automatically power cycle all the USB ports after boot, so that the t265 is recognized all the time.

nixinator, 2021-09-13:
nice work and power fluctuation debugging, nice. can i ask what power meter you are using in your screenshot, probably a essential item for robot builders.

HanDaSeul, 2021-09-13:
> nice work and power fluctuation debugging, nice. can i ask what power meter you are using in your screenshot, probably a essential item for robot builders.

My bad, it's a photo not a screenshot😂
It's j7-c. When I plug T265 through that usb tester, placing between RPi usb port and T265, t265 is always recognized as being connected to usb 2.0, not 3.0.

csr-kick, 2021-09-14:
@HanDaSeul When I did my testing the RPI4 (4GB) was sitting on my desk with just the T265 plugged in. It happens more frequently when you stress the CPU, and if you stress the USB by also taking in the video streams. When I used a powered USB hub with the D453i it also crashed more frequently, but as you pointed out not all usb hubs can provide the right power. I'll do a quick check with a custom USB hub I have somewhere around here that has its own 5v/5A regulator on it. Can you tell me what USB power analyzer you have? Looks pretty nice. 

As we talked earlier in this thread, I wrapped the driver in code that tracks the last pose, restarts the T265 pose stream, then applies the old pose as an offset to the now zero'd poses. Some other trickery with matching depths or lidar scans will relocalize any drift pretty well.

HanDaSeul, 2021-09-14:
> @HanDaSeul When I did my testing the RPI4 (4GB) was sitting on my desk with just the T265 plugged in. It happens more frequently when you stress the CPU, and if you stress the USB by also taking in [...]

If i'm right that USB power analyzer = power meter, it's 'j7-c'.

As you point out about stress, I tried overclocking little bit and making sure keep cool(didn't exceed 50 celsius degree), but nothing changed. Through the whole test myself, conclusion is 'as long as you are using RPi, don't try streaming  in ROS'. But just realsense-viewer, you can try streaming and there's no problem with connection. I still can't understand why. Only I can make it clear is difference of using resources like cpu, power whatever. 
I don't have D453i. Before I changed the usb hub(CR113), i tried other powered usb hub. Some bad powered usb hubs are not stable and have back-power issue.

Construkted-Reality, 2021-09-14:
I don't think it's a power delivery issue. At least not from my tests:
I'm running only the T265 attached to the Pi 4 (8gb). No other USB devices. I'm running a simple python script that outputs to the screen the T265 pose values to the screen. There is minimal CPU load. 

The Pi is sucking back around 0.8-0.9A with the T265 running. The USB power bank I'm using is capable of delivering 2.5A, so there's no bottleneck there.
The T265 still stops at random intervals...

[image]

HanDaSeul, 2021-09-14:
> I don't think it's a power delivery issue. At least not from my tests:
> I'm running only the T265 attached to the Pi 4 (8gb). No other USB devices. I'm running a simple python script that outputs to [...]

@Construkted-Reality  I tried several times after I applied my solution, there's no losing connection. Sometimes, having without usb tester is helpful. In my experience, which is not exactly clear tho, powerbank is less stable than power adapter. RPi 4 8GB is recommended 3A. for input. I tested with 3A adapter for RPi, 2A for powered usb hub.  

I'm using wireless usb adapter for sending pose data and it takes 0.45A, 5V, which means I turned off RPi's wifi chip.

Construkted-Reality, 2021-09-14:
@HanDaSeul 
I just tried two different 3A power adapters (one was from an LG phone, and another from a Pixel 3 phone) and have experienced the same loss of connection.
I have disabled the camera stream in the python script.

HanDaSeul, 2021-09-14:
@Construkted-Reality Then, probably only I can say is use wireless usb wifi adapter with powered usb hub. Don't let other usb devices take power from RPi. Let RPi take less power as you can like setting RPi's wifi chip off. And in my case as you see my screenshot, RPi gathers pose data and sends. My laptop receives that pose data, so I can see the t265's movement almost without lag through rviz.

Construkted-Reality, 2021-09-14:
I don't think cooing the T265 has anything to do with it since I've had the T265 work perfectly on a Pi3, and it never disconnects (without any cooling).
I'll do some tests with the wifi module disabled to see if that makes any difference.

zhouzhiwen2000, 2021-09-15:
> @Construkted-Reality Then, probably only I can say is use wireless usb wifi adapter with powered usb hub. Don't let other usb devices take power from RPi. Let RPi take less power as you can like setting RPi's wifi chip off. And just in case keep T265 cool.
> In my case as you see my screenshot, RPi gathers pose data and sends. My laptop receives that pose data, so I can see the t265's movement almost without lag through rviz.
> 
> Wait...it looks like the problem is not only the power, but also RPi's wifi chip performance...I didn't test using RPi's wifi chip.

Try putting more pressure on the RPI USB port while testing, eg. copying a large file from an SSD. It seems that the problem occurs more often when the bus is busy. The SSD can be powered externally to avoid extra power consumption.

zhouzhiwen2000, 2021-09-15:
> Try putting more pressure on the RPI USB port while testing, eg. copying a large file from an SSD. It seems that the problem occurs more often when the bus is busy. The SSD can be powered externally [...]

I tested using T265 and a USB camera, with the USB cam externally powered, and the problem is still around.

[5 more replies]

closed 2026-10-23 by MartyG-RealSense