realsenseai/librealsense #6365: feature selection

gwkyle, 2020-05-07:
Issue Description
Some applications (such as Skype ver 8.56) do not allow the user to select between Depth, Virtual and RGB modes for the RealSense F200. The default is Depth which provides green contour image. In order to display the F200 needs to be in Virtual mode. Similarly, Zoom initially displays the Depth mode as well. Zoom provide the capability to switch RealSense and it turns out the RGB mode provides appropriate video display. Provide a utility that allows users to change the default video mode that an application would find when it turns on the F200.

MartyG-RealSense, 2020-05-07:
Thanks so much for your feature request!  Thanks so much for signing up.  It is not likely to be a feature that the RealSense developers would create for the F200 however, as it is a retired camera model and they don't create updates for the F200 any more (the oldest model that still receives updates is SR300 because of its compatibility with the current RealSense SDK 2.0).  

My research indicates that once you have selected a camera driver (such as the F200 Virtual Driver) from the drop-down menu on the 'Audio & Video' section of Skype's 'Settings' interface, it should become the default setting for each subsequent time that you open video communications on Skype.  I hope that this helps you.

MartyG-RealSense, 2020-06-14:
I reviewed this case again for the possibility of logging it onto Intel's official feature request system.  I concluded that the best course of action would be to close the case unfortunately, as development updates for the F200 ceased several years ago and so the feature request would not be taken forward.

closed 2020-06-14 by MartyG-RealSense