realsenseai/librealsense #6366: Segmentation fault on a Raspberry Pi 4 when recieving a frame inside a ROS service

f4b14n1, 2020-05-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Raspbian Buster | 
| Kernel Version (Linux Only)    | 4.19.97-v7l+ | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            | 2.3.4 (latest commit) | 
| Language                            | C++ | 
| Segment			| Robot | 

### Issue Description

For a project in ROS I want to implement a service that returns a set number of depth frames from a D415 on a Raspberry Pi 4. Due to the fact that the camera should not be running all the time I planned on directly using the SDK inside the service call to turn the camera on and off. However I recieve a segfault whenever I try to get a frame from the SDK. The code works without problems on a normal computer. This is the relevant bit inside the service function of a minimal working example: 

`int currentBufferSize_{0}, maxBufferSize_{30};`
`// Create a Pipeline - this serves as a top-level API for streaming and processing frames`
`rs2::pipeline p;`
`// Configure and start the pipeline`
`rs2::pipeline_profile selection = p.start();`
`ROS_INFO_STREAM("Pipeline started.");`
`while (currentBufferSize_ <= maxBufferSize_)`
`{`
`// Block program until frames arrive`
`rs2::frameset frames = p.wait_for_frames(); // <-- segfault occurs here`
`// Try to get a frame of a depth image`
`depthImages_.push_back(frames.get_depth_frame());`
`currentBufferSize_++;`
`}`

I built the SDK on the Pi from source and the examples like rs-callback or rs-hello-world (whose relevant bits are almost the same as the ones here) are working without any problems. According to gdb the segfault occurs somewhere in rs_frame, you can find the output of gdb below. Thank you in advance for your assistance.

        roslaunch depth_image_processing_node_debug.launch
        ... logging to /home/pi/.ros/log/58594abc-90a2-11ea-b42d-dca63240524b/roslaunch-raspberrypi-11662.log
        Checking log directory for disk usage. This may take a while.
        Press Ctrl-C to interrupt
        Done checking log file disk usage. Usage is <1GB.

        started roslaunch server http://192.168.2.109:42369/

        SUMMARY
        ========

        PARAMETERS
        * /pi/depth_image_processing/buffer_size: 100
        * /pi/depth_image_processing/filtering_iterations: 2
        * /pi/depth_image_processing/gaussian_sigma: 3.0
        * /rosdistro: melodic
        * /rosversion: 1.14.4

        NODES
        /pi/
        depth_image_processing (pi_depth_image_processing_ros_tool/depth_image_processing)

        ROS_MASTER_URI=http://localhost:11311

        process[pi/depth_image_processing-1]: started with pid [11683]
        GNU gdb (Raspbian 8.2.1-2) 8.2.1
        Copyright (C) 2018 Free Software Foundation, Inc.
        License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
        This is free software: you are free to change and redistribute it.
        There is NO WARRANTY, to the extent permitted by law.
        Type "show copying" and "show warranty" for details.
        This GDB was configured as "arm-linux-gnueabihf".
        Type "show configuration" for configuration details.
        For bug reporting instructions, please see:
        <http://www.gnu.org/software/gdb/bugs/>.
        Find the GDB manual and other documentation resources online at:
        <http://www.gnu.org/software/gdb/documentation/>.

[... 122 lines cut ...]
            at /home/pi/librealsense/src/core/streaming.h:146
            #17 0xb5b4bbe8 in std::_Function_handler<void(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>), librealsense::uvc_sensor::open(const stream_profiles&)::<lambda(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void()>)> >::_M_invoke(const std::_Any_data &, librealsense::platform::stream_profile &&, librealsense::platform::frame_object &&, std::function<void()> &&) (__functor=..., __args#0=..., __args#1=..., __args#2=...)
            at /usr/include/c++/8/bits/move.h:182
            #18 0xb5a87f80 in std::function<void (librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>)>::operator()(librealsense::platform::stream_profile, librealsense::platform::frame_object, std::function<void ()>) const (__args#2=..., __args#1=..., __args#0=...,
            this=<optimized out>) at /usr/include/c++/8/bits/std_function.h:682
              #19 librealsense::platform::uvc_streamer::<lambda(dispatcher::cancellable_timer)>::operator() (
              __closure=<optimized out>, cancellable_timer=...)
                at /home/pi/librealsense/src/uvc/uvc-streamer.cpp:95
                #20 std::_Function_handler<void(dispatcher::cancellable_timer), librealsense::platform::uvc_streamer::in--Type <RET> for more, q to quit, c to continue without paging--
                  it()::<lambda(dispatcher::cancellable_timer)> >::_M_invoke(const std::_Any_data &, dispatcher::cancellable_timer &&) (__functor=..., __args#0=...) at /usr/include/c++/8/bits/std_function.h:297
                  #21 0xb5a7abb0 in std::function<void (dispatcher::cancellable_timer)>::operator()(dispatcher::cancellable_timer) const (__args#0=..., this=<optimized out>) at /usr/include/c++/8/bits/std_function.h:682
                    #22 active_object<std::function<void (dispatcher::cancellable_timer)> >::do_loop()::{lambda(dispatcher::cancellable_timer)#1}::operator()(dispatcher::cancellable_timer) const (this=<optimized out>, ct=...)
                      at /home/pi/librealsense/src/hid/../concurrency.h:395
                      #23 std::_Function_handler<void (dispatcher::cancellable_timer), active_object<std::function<void (dispatcher::cancellable_timer)> >::do_loop()::{lambda(dispatcher::cancellable_timer)#1}>::_M_invoke(std::_Any_data const&, dispatcher::cancellable_timer&&) (__functor=..., __args#0=...)
                      at /usr/include/c++/8/bits/std_function.h:297
                      #24 0xb5927010 in std::function<void (dispatcher::cancellable_timer)>::operator()(dispatcher::cancellable_timer) const (__args#0=..., this=0xa765ad80) at /usr/include/c++/8/bits/std_function.h:682
                      #25 dispatcher::dispatcher(unsigned int)::{lambda()#1}::operator()() const (__closure=0x2f0f44)
                      at /home/pi/librealsense/src/concurrency.h:231
                      #26 std::__invoke_impl<void, dispatcher::dispatcher(unsigned int)::{lambda()#1}>(std::__invoke_other, dispatcher::dispatcher(unsigned int)::{lambda()#1}&&) (__f=...) at /usr/include/c++/8/bits/invoke.h:60
                      #27 std::__invoke<dispatcher::dispatcher(unsigned int)::{lambda()#1}>(std::__invoke_result&&, (dispatcher::dispatcher(unsigned int)::{lambda()#1}&&)...) (__fn=...) at /usr/include/c++/8/bits/invoke.h:95
                      #28 std::thread::_Invoker<std::tuple<dispatcher::dispatcher(unsigned int)::{lambda()#1}> >::_M_invoke<0u>(std::_Index_tuple<0u>) (this=0x2f0f44) at /usr/include/c++/8/thread:244
                      #29 std::thread::_Invoker<std::tuple<dispatcher::dispatcher(unsigned int)::{lambda()#1}> >::operator()() (this=0x2f0f44) at /usr/include/c++/8/thread:253
                      #30 std::thread::_State_impl<std::thread::_Invoker<std::tuple<dispatcher::dispatcher(unsigned int)::{lambda()#1}> > >::_M_run() (this=0x2f0f40) at /usr/include/c++/8/thread:196
                      #31 0xb551c9b0 in ?? () from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
                      #32 0xb6698494 in start_thread (arg=0xa765b3d0) at pthread_create.c:486

MartyG-RealSense, 2020-05-08:
Hi @f4b14n1   The first question I would ask in a case involving a segmentation fault is whether you have configured a Swap File on your Pi.

https://github.com/IntelRealSense/librealsense/issues/3341

f4b14n1, 2020-05-08:
Hello @MartyG-RealSense Yes, the Pi has 4 GB of RAM and I have a swap file of 2 GB configured. 
[image]
When running the program the memory usage is around 300 MB.

MartyG-RealSense, 2020-05-08:
Do you have an up to date firmware driver installed inside your camera please?  One Pi user with a segmentation fault was able to fix it just by updating the firmware.  This can be done through the RealSense Viewer program by going to the 'More' option at the top of the options side-panel and selecting the option from its menu to update to recommended firmware.

f4b14n1, 2020-05-08:
Thank you for your quick response. I updated the camera firmware to version 05.11.06.250, sadly the error still seems to persist.

MartyG-RealSense, 2020-05-08:
It may not make a difference to this particular problem but it is a good idea to use a relatively recent firmware.  There has not been a Production firmware for a while, so if you are concerned about having production-level stability, a good rule is to use "one version back" for SDK and firmware.  This gives time for any problems in a new version to come to light.

For firmware, the second newest version is currently **5.12.2.100**.

MartyG-RealSense, 2020-05-24:
Do you require any further assistance with this case or can it be closed, please?   Thanks!

f4b14n1, 2020-05-24:
I updated the firmware version to 5.12.2.100 as mentioned, but the error is still there. I rewrote the problematic bit to use the ros wrapper now, so this issue can be closed. Thanks for your assistance.

closed 2020-05-24 by f4b14n1

MartyG-RealSense, 2020-05-24:
I'm pleased that you found a solution that worked for you.   Thanks for the update!