realsenseai/librealsense #6367: Integration with ROS

ShrutheeshIR, 2020-05-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Apologies if the question has been asked. However, I was not able to find any answer. Using a gazebo ROS plugin, I am receiving my images on the rostopics /camera/color/image_raw and similarly depth. How do I interface with this module (python scripts in particular). Should it be a subscriber?
Does someone have a concrete example code of what I am looking for? (i.e. interface librealsense with rostopics)
I have gone through the realsense-ros package, however, it gives only launch files and not any python scripts. I'm new to this. Any help would be appreciated. Thanks!

MartyG-RealSense, 2020-05-08:
**realsense2_camera** has a test script called **rs2_test.py**

https://github.com/IntelRealSense/realsense-ros/tree/development/realsense2_camera/scripts

A guide by the website **idorobotics** demonstrates how to use this script to "run RealSense ROS nodes and visualize them in RViz, inspect the topics being published and play around withu different paramater settings".  This information can be found in the section of the guide headed **Running the camera in ROS**

https://idorobotics.com/2018/11/02/integrating-the-intel-realsense-d435-with-ros

ShrutheeshIR, 2020-05-09:
Thank you for the resources! These are extremely helpful. 
Is there any way to operate directly with the data, i.e. without using a rosbag and working with the images real time?
Thanks

MartyG-RealSense, 2020-05-09:
I did a very extensive search but found very little that was useful regarding accessing live RealSense ROS data with Python.  The best resource available was this:

https://github.com/IntelRealSense/realsense-ros/issues/1086

ShrutheeshIR, 2020-05-09:
Thanks for the help. I'll close the issue. Please drop a comment if you'd like this issue to be open.

closed 2020-05-09 by ShrutheeshIR