realsenseai/librealsense #6368: Linux 5.4 support?

slitcch, 2020-05-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Ubuntu 20.04}  | 
| Kernel Version (Linux Only)    |  (e.g. 5.4.0)                                         | 
| Platform                                 | PC.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  { C }                          | 
| Segment			|  {/VR/AR }                          | 

### Issue Description
Can we get kernel modules for Linux 5.4? Currently, it looks like y'all only support 4.x.

MartyG-RealSense, 2020-05-08:
Officially Librealsense already supports Kernel 5.0 and 5.3, though a RealSense user tried 5.4 and said that it worked for them.

https://github.com/IntelRealSense/librealsense/issues/5723#issuecomment-607107643

thorstink, 2020-05-30:
I installed from source on ubuntu 20.04 / kernel 5.4, but the frame metadata doesn't seem work... still figuring out if there's something I can do.

MartyG-RealSense, 2020-05-30:
When installing from source on Linux, metadata support has to be patched in.  

Have you tried the patching instruction that works with 18.04 LTS to see if it works with 20.04, please?

**./scripts/patch-realsense-ubuntu-lts.sh**

thorstink, 2020-05-30:
The script doesn't allow it:
`Unsupported distribution focal kernel version 5.4.0-29-generic . The patches are maintained for Ubuntu16/18 (Xenial/Bionic) with LTS kernels 4-[4,8,10,13,15,18]
`

I don't know if how safe it is to disable the check and just see?

MartyG-RealSense, 2020-05-30:
If patching is a problem when installing using the usual installation methods, there is the option of doing a "backend" installation that requires an internet connection but is not reliant on Linux versions, kernel versions or patches.  The instructions for this process are below.

1.  Go to the Librealsense root directory and create a Build folder, and then change to that created folder using the command below (if you have not created the Build folder already):

**mkdir build && cd build**

Whilst in the Build folder, run the CMake command below:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

thorstink, 2020-05-30:
make'ing as we speak. I'll update asap.

thorstink, 2020-05-30:
Seems to have done the trick. Thanks.

I didn't look at performance yet, but from some quick scanning, it seems there's (minimal) performance consequences of using RSUSB_BACKEND?

MartyG-RealSense, 2020-05-30:
Great news - thanks for the update!

The link below describes the advantages and disadvantages of the backend installation method compared to the usual methods such as compiling from source.  Please scroll down the linked-to comment to the section headed **'What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

thorstink, 2020-06-08:
Tiny update though:

I haven't tested too much with it, but it does seem that video stream is less stable compared to the kernel patches. None the less, good workaround for now until 20.04 gets support.

MartyG-RealSense, 2020-06-14:
I reviewed this case again and concluded that since support for new kernels gets added by the Librealsense developer team over time, there was no need to make a special request to do so for kernel 5.4, as new kernels will already be in the developers' minds.   I will therefore close this case.   Thanks!

closed 2020-06-14 by MartyG-RealSense

malapatiravi, 2020-06-15:
> Tiny update though:
> 
> I haven't tested too much with it, but it does seem that video stream is less stable compared to the kernel patches. None the less, good workaround for now until 20.04 gets support.

Are you able to use pyrealsense2 using python ?

MartyG-RealSense, 2020-11-18:
I just wanted to let commenters on this discussion know that support for Kernel 5.4 and Ubuntu 20 have been added in the new SDK **2.40.0** release.