realsenseai/librealsense #6369: D410 firmware streaming fails on Linux 4.4

imsherlock, 2020-05-08:
We are having issues streaming with the D410 using certain firmware versions.
We are currently using the upstream version of Rockchip's RK3399 Android Kernel (4.4) connected to the D410 over USB 3.1 using version 2.21 of the librealsense API.

The kernel has been patched using the provided kernel patches from Intel RealSense meta layer.

With certain firmware versions, we experience the following error when we attempt to start streaming:
`RealSense error calling rs2_pipeline_start_with_config_and_callback_cpp (pipe:0x55904e23c0, config:0x55)
        set_xu(ctrl=1) failed! Last Error: Input/output error`

Unplugging the USB from the D410 and plugging it back in does not always work. We've also tried unbinding and binding the USB device in Linux as well, but that has also produced inconsistent results.

The following firmware versions have been tested:
5.11.6-250 (Production) : WORKING
5.11.11-100 (Development) : WORKING
5.11.15-0 (Development) : FAILED
5.12.0 (Development) : FAILED
5.12.1 (Development) : FAILED
5.12.2-100 (Development) : FAILED
5.12.3.0-1 (Development) : FAILED

I'll post a more minimal example later, but we've tested with both Infrared and Depth only and together, with different formats and resolutions (Z16, RGB8, Y8, 1280 x 720, 640 x 480).
This issue has also been seen on the D430 we have, but thorough testing has not been done yet.

MartyG-RealSense, 2020-05-09:
Hi @imsherlock  SDK 2.21 is over a year old, having been released in April 2019, and there have been many SDK updates released since then.  Is there a reason why your project needs to use SDK 2.21, rather than an up to date SDK version and a recent firmware, please?  

Also, are you using a rooted or unrooted Android device?  The Android wrapper originally required a rooted device but later switched to being able to use unrooted devices with a version that requires 4.4 or newer.  Can you confirm which wrapper (old or current) you are using please?

**Rooted (old)**
https://github.com/IntelRealSense/librealsense/tree/v2.18.1/doc/android

**Unrooted (current)**
https://github.com/IntelRealSense/librealsense/blob/master/doc/android.md

imsherlock, 2020-05-11:
Hi @MartyG-RealSense , we're actually not running Android OS, just the kernel maintained by Rockchip for Android. Ideally we'd move to upstream, but there are a few things we need that force us to stay with that particular kernel.

We are using the _warrior_ branch provided in Intel's Yocto meta repository for librealsense. I believe this is currently set to version 2.21.
Is this a known issue fixed by the SDK? Or have there been USB issues reported in more recent firmwares?

MartyG-RealSense, 2020-05-11:
It has been a while since I experienced cases involving Librealsense for Yocto - not since the Intel Aero flight drones!

May I confirm please that this is the Yocto Layer for Librealsense version that you are using:

https://github.com/IntelRealSense/meta-intel-realsense

In general, I have never had a case where so many firmware versions have failed to work correctly for a customer.  The Yocto process for installing Librealsense support seems to be so different from the standard Librealsense installation method that I will refer this case to a support manager for advice.  Thanks for your patience!

imsherlock, 2020-05-12:
Hi @MartyG-RealSense , yes that's the repository we are using.
In addition, we've also tested version 2.29 of the SDK on our system and had the same results with all firmware versions.

MartyG-RealSense, 2020-05-12:
Thanks very much for the additional information.  I referred your case to a support manager yesterday and got confirmation of receival of the request, so hopefully they will be able to help you find a solution.

RealSenseSupport, 2020-05-12:
@imsherlock , hi. Thank you so much for sharing the testing results of different firmware versions over kernel 4.4 on RK3399. Could you please confirm if the fw/sdk build configuration mentioned in #5931 work on your board?

imsherlock, 2020-05-14:
Hi @RealSenseSupport , I recompiled the librealsense2 library under Yocto and attempted to re-run the same tests with firmware version 5.12.3.0-1.
I ran into the same issue I originally experienced and then afterward received a segault followed by issues of "Unable to find a device".
Are there any logs or information I can collect to help triage this issue?

RealSenseSupport, 2020-05-14:
@imsherlock , thank you for checking. 
Please clarify:
> Unplugging the USB from the D410 and plugging it back in does not always work
Does that mean that the newer firmware, ie., 5.12.3.0 would sometimes work after re-plugging and this issue does not happen every time you start the pipeline? Please also double check if the D410 is being enumerated as USB2 or USB3 mode. If you also happen to have our peripheral device (D415/D435) on hand, please give it a try on RK3399 and x86 PC/laptop. 

Collecting log is also helpful for further investigation, please use one of the example app provided in the SDK such as rs-capture and [capture ](https://github.com/IntelRealSense/librealsense/blob/master/doc/troubleshooting.md#troubleshooting-tips-and-tricks) librealsense log, dmesg log, kernel info, and lsusb. 

Thank you.

imsherlock, 2020-05-14:
@RealSenseSupport , yes, with newer versions of the firmware if I unplug and plug in the USB the device sometimes starts streaming. Sometimes re-binding the USB device driver in Linux fixes the issue as well, but there is no consistent pattern in doing either of these.

The D410 is being enumerated as a USB3 device. I'll try the rs-capture tools for both sets of firmware on the RK3399 and get back to you with my findings.

RealSenseSupport, 2020-05-14:
@imsherlock, thank you. Can you please also provide the serial number of the D410 and/or picture of the actual unit?

imsherlock, 2020-05-15:
@RealSenseSupport , the serial I see listed is 813612020225. We have a second unit with serial number 802612020234.

imsherlock, 2020-05-20:
hi. @RealSenseSupport , I wanted to follow up and see if there was any further insight you could give to this problem?

RealSenseSupport, 2020-05-20:
@imsherlock, thank you for providing the device serial number. 
When the stream start failed, could you please check, or provide relevant log, if the device still show up under `lsusb `and `rs-enumerate-devices -s` plus any indication of camera disconnection from the USB bus in the dmesg log? 
Thank you.

imsherlock, 2020-05-27:
Hi @RealSenseSupport , I attempted to capture using one of the devices above and got the same failure I reported. The logs requested are attached below

[lsusb.txt](https://github.com/IntelRealSense/librealsense/files/4689914/lsusb.txt)
[dmesg.txt](https://githu
[rs-enumerate.txt](https://github.com/IntelRealSense/librealsense/files/4689915/rs-enumerate.txt)
b.com/IntelRealSense/librealsense/files/4689913/dmesg.txt)

RealSenseSupport, 2020-05-27:
@imsherlock , thank you for providing the log. 
So the problem occurs even when D410 shows up correctly under lsusb/rs-enumerates-devices but failed when attempting to stream from it. The device seems to be attached directly to the USB root hub of the host. Could you try to perform a [hardware_reset()](https://github.com/IntelRealSense/librealsense/blob/842ee1e1e5c4bb96d63582a7fde061dbc1bebf69/include/librealsense2/hpp/rs_device.hpp#L90) before configuring and starting the stream? 

By the way, I just want to be sure, is this issue only reproducible on the RK3399? 
Thank you.

imsherlock, 2020-05-27:
Hi @RealSenseSupport , I forget to mention in my original post that we already perform a [hardware_reset()](https://github.com/IntelRealSense/librealsense/blob/842ee1e1e5c4bb96d63582a7fde061dbc1bebf69/include/librealsense2/hpp/rs_device.hpp#L90) before configuring the stream.
As of right now, it only seems to happen on the RK3399 platform.

imsherlock, 2020-05-29:
@RealSenseSupport , following up to see if any additional information is necessary, or if you have reproduced this on your end?

RealSenseSupport, 2020-06-05:
@imsherlock , I have not yet encountered this issue here. The kernel and hw information used are attached here: [lshw.txt](https://github.com/IntelRealSense/librealsense/files/4737527/lshw.txt) and [uname_a.txt](https://github.com/IntelRealSense/librealsense/files/4737534/uname_a.txt)

100 start-stop cycle result based on data-collect from SDK [v.2.35.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.0) release package
[result.txt](https://github.com/IntelRealSense/librealsense/files/4737531/result.txt)

Please note that the ticket will be closed in 7 days if not hearing back from you. Thank you.

imsherlock, 2020-06-05:
@RealSenseSupport thank you for sharing your results. Could you share the data_collect.cfg file you used? I'll try and reproduce this with the firmware I reported on, as well as reproducing with the newer version of the RealSense firmware and SDK package.

RealSenseSupport, 2020-06-05:
@imsherlock , of course and please find the attached data_collect file. 
[data_collect.cfg.txt](https://github.com/IntelRealSense/librealsense/files/4738037/data_collect.cfg.txt)
Thank you.

imsherlock, 2020-06-15:
@RealSenseSupport following up to see if anything else was discovered

RealSenseSupport, 2020-06-19:
@imsherlock , just want to be sure, the SDK was built with the RSUSB backend, correct?
`cmake ../ -DBUILD_EXAMPLES=true -DFORCE_RSUSB=true -DCMAKE_BUILD_TYPE=Release`

The RSUSB backend was introduced in 2.30 and above and replaced the old libuvc and without patching the kernel module. This is also the backend I used to test D410 with RK3399.

RealSenseCustomerSupport, 2020-07-01:
----------------------------------------------

Hi, just curious if any further help is needed for this case. Thank you.

imsherlock, 2020-07-07:
Hi @RealSenseCustomerSupport , yes, I'm attempting to confirm the alternative FORCE_RSUSB=true is set. I'll have some feedback and logs provided shortly. Thanks!

imsherlock, 2020-07-15:
Hi @RealSenseCustomerSupport , quick update. I was able to test the FORCE_RSUSB=true on version 2.29 of the librealsense library in our Yocto environment.
I was not able to reproduce the issue with the provided test using the rs-data-collect tool with the 05.12.03.00 firmware. I did, however, experience a kernel oops when trying to update the RealSense device using the rs-fw-update tool and enumerate the device. Not sure what actually caused this and was quite unexpected.
I'm going to perform one additional test using our original setup and will report back with logs before closing this issue.
Thanks!

imsherlock, 2020-07-31:
Hi @RealSenseCustomerSupport , as it turns out the FORCE_RSUSB_BACKEND was not introduced until version 2.30.0 of the librealsense API, so any data I had collected for 2.29.0 was invalid.

I was able to compile the the 2.31.0 version of the librealsense API with FORCE_RSUSB_BACKEND and retest.
The good news is that this version of the library allows us to use newer version of the RealSense firmware, but the bad news is that we experience very high CPU usage (33-56% of the available CPU).

The high CPU usage seems to be present in newer versions of the API as well when FORCE_RSUSB_BACKEND is enabled.
Having high CPU usage is not acceptable for our use case so I do not believe we can move forward using FORCE_RSUSB_BACKEND.

Is there another alternative? Can you help diagnose and solve the original problem?

RealSenseSupport, 2020-08-06:
@imsherlock , thank you for the update. The high CPU usage with RSUSB_BACKEND should already be addressed in build v2.34.0 and beyond. See PR [#6083](https://github.com/IntelRealSense/librealsense/pull/6083) and release note here. 
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2340

RealSenseSupport, 2020-08-07:
Since the original issue was resolved using RSUSB_BACKEND with libRealSense 2.30.0+ and the high CPU usage should also be addressed by 2.34.0+, I will close this ticket for now. 
Please feel free to open another ticket if high CPU usage still persists or if there is any other problem. 
Thank you.

closed 2020-08-07 by RealSenseSupport