realsenseai/librealsense #6371: rs-kinfu crashes on launch

Inffzy99, 2020-05-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version | Windows 10, macOS Catalina 10.15.4 | 
| Kernel Version (Linux Only)    |  | 
| Platform                                 | PC, Mac |
| SDK Version                            | 2.0 | 
| Language                            | C++ | 
| Segment			| AR | 

### Issue Description

rs-kinfu crashes at launch. On Windows, when launching the executable, 2 command prompts show up, then disappear after about 1 second. On mac, the following frame appears, freezes, then crashes along with a crash report. 
[image]
[image]

This is not the only example executable that does not run correctly. Others include rs-ar-basic, rs-ar-advanced, rs-convert, rs-data-collect, rs-dnn, rs-motion, rs-pose, rs-pose-and-image, rs-pose-predict, rs-tracking-and-depth, rs-trajectory.

This is the first time for me submitting an issue. If there are standards that I am not meeting, please let me know. Thank you very much.

MartyG-RealSense, 2020-05-09:
Hi @Inffzy99   Some of the example programs that you listed are not for the D435.  

*T265**
rs-ar-basic, rs-ar-advanced, rs-pose, rs-pose-and-image, rs-pose-predict, rs-tracking-and-depth, rs-trajectory.

**D435i with IMU**
rs-motion

The ones that should work with the IMU-less D435 are::
rs-convert, rs-data-collect, rs-dnn (which needs the RealSense OpenCV wrapper though).

If you are launching the rs-convert and rs-data-collect from the **Intel RealSense SDK 2.0 > Tools** folder of Windows and it starts to launch and then disappears, please try running them from the Windows Command Prompt text interface window instead.  You can find the Command Prompt by typing 'command' into the text box at the bottom of your Windows screen, next to the button in the corner that you use to shut the PC down.

Regarding rs-kinfu: the program is designed to switch to a camera 'preset' configuration called High Density.  I wonder if this is the cause of the problem, as changing the preset has been known to cause crashes and freezes in Librealsense in the past.  Commenting out or removing the Kinfu code line below may therefore help:

**dpt.set_option(RS2_OPTION_VISUAL_PRESET, RS2_RS400_VISUAL_PRESET_HIGH_DENSITY);**

Inffzy99, 2020-05-19:
Thank you very much for the response. Everything is good now regarding this issue. I'm just replying for the record. 

Although commenting out the given line did not directly solve the problem, it revealed the real problem which in turn showed up on the command line, which is: OpenCL error CL_OUT_OF_RESOURCES. This is then solved by setting MAX_WORK_GROUP_SIZE to 512 in ocl.cpp in OpenCV's source file.

closed 2020-05-19 by Inffzy99

MartyG-RealSense, 2020-05-19:
Awesome news - thanks for the update and for sharing your solution with the RealSense community!   :)