realsenseai/librealsense #6372: Can the Intel T265 camera give me a map of an envionment to do the route planning?

Andrew5c, 2020-05-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.140                                         | 
| Platform                                 | Jetson Nano |
| SDK Version                            |  2.33.1                         | 
| Language                            |  Python                         | 
| Segment			|  Quadrotor                        | 

### Issue Description
Hi, I want to get the map of an unknow envionment, Can the T265 give me a map of an envionment so that I can do the route planning?
Thanks very much!

neilyoung, 2020-05-09:
What do you mean by route planning? You could draw the track in a 2D or 3D diagram. If you are looking for WGS84 coordinates, then it is "the other way around": You would have to provide a geographical context to the poses. See also neilyoung.serveblog.net. Then from knowing the geographical position or the X/Y/Z position of the cam it is surely possible to control a device, which carries this camera, to go here and there.

Andrew5c, 2020-05-09:
Thanks @neilyoung for your reply. 
The **route planning** what I said is that to plan a path in an envionment for the mobile robot which equiped with T265. In other words, what I want to ask is that, can the T265 camera give us a *3D model* (i.e. the map) of the envionment it located.

neilyoung, 2020-05-09:
Well, yes, but you maybe will not be able to use it, since the format of the map produced is proprietary and undocumented. According to @radfordi on a conference lately it would not survive a reverse-engineering attack, but I didn't try so far.

But if you go the other way around (having a true to scale floor plan, map the site, geo-reference it), then you later on could reuse that map on your robot and give the robot geo-spatial awareness. I would say (w/o having checked it) it should be sufficient for the robot to come from A to B. Then the obstacle detection needs to be solved in order to find paths around them. Maybe a bundle of D435 and T265 would help you.

Andrew5c, 2020-05-09:
OK， got it . 
We can coupling of localization and depth data for mapping using Intel RealSense T265 and D435i cameras. Burt it is a little diffcult for me....
thank you very much!

neilyoung, 2020-05-09:
Then the other way should work for you. It’s easy

closed 2021-03-17 by RealSenseSupport