realsenseai/librealsense #6376: rs2::net_device multiple device

merttoka, 2020-05-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC/Raspberry Pi  |
| SDK Version                            |  2.34                          | 
| Language                            |  C                          | 
| Segment			|  Network                          | 

### Issue Description
I updated and configured rs2-net by following [this article](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras). @dorodnic et al, thank you for the articulate explanation. 

I have a small question about using network devices in the multi-camera setting. I have been reading multiple cameras with a `rs2::config` instance and passing it to the `rs2::pipeline_profile`. With the networked version, I understand that I am expected to avoid using `rs2::config`, is that right? 

My local method of using multiple cameras was as follows:
```c++
ctx.set_devices_changed_callback([&](rs2::event_information& info) {
		RemoveDevices(info);
		for (auto&& dev : info.get_new_devices()) {
			EnableDevices(dev);
		}
	});
// Initial population of the device list
for (auto&& dev : ctx.query_devices()) // Query the list of connected RealSense devices
{
	EnableDevices(dev);
}

void EnableDevices(rs2::device dev) {
        ...

        rs2::pipeline p;
	rs2::config c;
	c.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        
        ...
	
        // Start the pipeline with the configuration
	rs2::pipeline_profile profile = p.start(c);	

        ...
}
```

But now, if I add the networked device in this pipeline, application throws an exception:
```
RealSense error calling rs2_pipeline_start_with_config(pipe:00000192D7A486E0, config:00000192D7A48320):
    No device connected
``` 

Instead, I modified my code like so:
```c++
ctx.set_devices_changed_callback([&](rs2::event_information& info) {
		RemoveDevices(info);
		for (auto&& dev : info.get_new_devices()) {
			EnableDevices(dev);
		}
	});

// ---- NET DEVICE --- 
rs2::net_device dev(ip);
dev.add_to(ctx);

// Initial population of the device list
for (auto&& dev : ctx.query_devices()) // Query the list of connected RealSense devices
{
	EnableDevices(dev);
}

void EnableDevices(rs2::device dev) {
        ...

        rs2::pipeline p(ctx);
	rs2::pipeline_profile profile = p.start();	

        ...
}
```
With this change, the network device works.  However, I can only see the streams of the last camera that has been enabled (possibly caused by passing the `context` to `pipeline` constructor?). Unfortunately, I have only one Pi4 at this point, so I cannot try multiple networked cameras. However, when I try to run one local and one networked camera, the local one seems to be overriding the streams that I can see. 

So my question is that can I use net devices without needing to add `rs2::context` to the `rs2::pipeline`? Or, do I misunderstand something?

dorodnic, 2020-05-12:
Hi @merttoka 
Let me check. I think network_device is exposing serial number info like any other camera and hence config enable device with s/n should work.

RealSenseSupport, 2020-08-05:
@dorodnic Do you have a chance to check this?

RealSenseSupport, 2021-01-04:
We will implement this in next SDK release.

bobluoluo, 2021-02-22:
I have the problem of using multiple cameras under net_device/Raspberry PI too. Here is the minimum code to create the issue.
 
[image]

Then the console output:
[image]

RealSenseSupport, 2021-05-28:
@merttoka @bobluoluo multi Ethernet cameras is not currently supported. It is in the backlog, but there is no schedule yet.

closed 2021-08-20 by RealSenseSupport