realsenseai/librealsense #6377: No "common/fw/target.h" file?

dongxuanlb, 2020-05-11:
**Version:**
HEAD detached at v2.34.1

**Build command:** 
dongxuan@dongxuan-2 build % cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false

**Output:**
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.34.1
-- Setting Unix configurations
-- using RS2_USE_LIBUVC_BACKEND
-- GLFW 3.3 not found; using internal version
-- Using Cocoa for window creation
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB)
-- Unable to find apriltag library, skipping pose-apriltag example
-- Building with TM2
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   realsense2-gl

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    WITH_TRACKING

-- Build files have been written to: /Users/dongxuan/workspace/librealsense/build

**Problem:**
/Users/dongxuan/workspace/librealsense/src/tm2/tm-info.cpp:13:10: fatal error: 'common/fw/target.h' file not found
#include "common/fw/target.h"

MartyG-RealSense, 2020-05-11:
Could you try building librealsense in Release mode by including the flag **-DCMAKE_BUILD_TYPE=Release** in your CMake build statement, please?

dongxuanlb, 2020-05-12:
Hi
I already know where the problem is, because this file was downloaded and decompressed by “http://realsense-hw-public.s3.amazonaws.com/Releases/TM2/FW/target/0.2.0.951/target-0.2.0.951.mvcmd”

The problem is that even if I add -DWith_build_TM2=fale to the cmake command, it will still include this file.

MartyG-RealSense, 2020-05-12:
May I confirm please that you expressed the BUILD_WITH_TMS instruction like this:

 **-DBUILD_WITH_TM2=false**

MartyG-RealSense, 2020-05-30:
Case closed due to no further comments received.

closed 2020-05-30 by MartyG-RealSense

whubaichuan, 2020-09-10:
@dongxuanlb Hi, how do you solve this problem?