realsenseai/librealsense #6379: How to use 10 D435i cameras simultaneously in Nvidia Xavier

malith1992, 2020-05-11:
| Camera Model                       | { D435i } | 
| Operating System & Version |   {Ubuntu 18.4}  | 
| Platform                                 | NVIDIA Xavier  |
| Language                            |  {realsense-viewer }                          | 
| Segment			|  {Robot}                          | 

### Issue Description

I want to view the Stereo Module, RGB Camera and Motion Module of 10 Relasense D435i cameras through realsense viewer in NVIDIA Xavier. I have connected cameras via several usb hubs to the Xavier and try to open them in realsense viewer. But the realsense viewer crashed. Is there any other step I should follow to overcome this issue ?

MartyG-RealSense, 2020-05-11:
The RealSense SDK has a C++ sample program called **rs-multicam** that opens stream display panels for as many cameras as you have attached to the same computer.  You could use this to test whether the problem is with the RealSense Viewer specifically or whether the Xavier is struggling to handle 10 cameras attached to the same board with several USB hubs.

https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

The number of hubs should not be a problem, as the USB standard permits the chaining together of up to 5 hubs (in a '5 levels deep' arrangement).  

Although the Xavier is a powerful board, there have been recent cases where problems occur if more than 2 cameras are attached to the board at the same time.  Are the hubs that you are using all mains electricity powered hubs connected to a wall power socket (the ideal situation for using multiple cameras with single-board computers) or are they "unpowered" hubs (not plugged into the mains socket) that draw their power from the USB port of the Xavier?

malith1992, 2020-05-12:
> The RealSense SDK has a C++ sample program called **rs-multicam** that opens stream display panels for as many cameras as you have attached to the same computer. You could use this to test whether the problem is with the RealSense Viewer specifically or whether the Xavier is struggling to handle 10 cameras attached to the same board with several USB hubs.
> 
> https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam
> 
> If all 10 are all streaming simultaneously on the same board then that would be very demanding on computing resources. A full PC with an Intel i7 processor is recommended even for 4 cameras being used simultaneously on the same computer.
> 
> It is therefore better to activate the cameras in batches if possible (activate 4, take a snapshot and turn them off, and activate the next set of 4 and take a snapshot with those, until you get back to the first camera set and start the cycle again), If all 10 need to be streaming at the same time, it would be better to divide the cameras up between multiple computers and send the data over a network to a computer dedicated to combining all the camera feeds together and doing processing on them.
> 
> The number of hubs should not be a problem, as the USB standard permits the chaining together on the same computer of up to 5 hubs (in a '5 levels deep' arrangement).
> 
> Although the Xavier is a powerful board, there have been recent cases where problems occur if more than 1 camera is attached to the board at the same time. Are the hubs that you are using all mains electricity powered hubs connected to a wall power socket (the ideal situation for using multiple cameras with single-board computers) or are they "unpowered" hubs (not plugged into the mains socket) that draw their power from the USB port of the Xavier?

I tried the rs-multicam with just 2 cameras and it crashed. I use unpowered hubs. May be the issue is with the hubs.

MartyG-RealSense, 2020-05-12:
The most recent person who had this multiple camera problem with Xavier experienced it with both unpowered and powered hubs.  They concluded that the problem may be related to how they were calling the individual cameras by their serial number, as commenting out the serial number code stopped the program from crashing.

 https://github.com/IntelRealSense/librealsense/issues/5828#issuecomment-619517603

Another user on the same discussion recently posted their script workaround.

https://github.com/IntelRealSense/librealsense/issues/5828#issuecomment-622993107

MartyG-RealSense, 2020-05-27:
Hi @malith1992 

The RealSense team have been working on a "pull" to address problems with multi-camera setups that use RSUSB and asked me to link relevant cases to the pull's page:

https://github.com/IntelRealSense/librealsense/pull/6467

MartyG-RealSense, 2020-06-14:
Hi again @malith1992  The new multicam improvements were introduced in the recently released librealsense SDK **2.35.2**.  Have you had the opportunity to try this new SDK and see if it resolves your mlticam problems, please?   Thanks!

closed 2020-07-05 by MartyG-RealSense