realsenseai/librealsense #6380: get_xu (id = 4) failed! Last Error: No error

xiaoniu001, 2020-05-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)   | 
| SDK Version                            | pyrealsense2: 2.33.1.1388  | 
| Language                            | python     | 

### Issue Description
I tried to set advanced mode for D435i, but i got this error:  
RuntimeError: get_xu(id=4) failed! Last Error: No error

MartyG-RealSense, 2020-05-11:
I was unable to find any previous references to this error.

Were you using the SDK's Advanced Mode sample program to set the mode, please?

 https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py

xiaoniu001, 2020-05-12:
> I was unable to find any previous references to this error.
> 
> Were you using the SDK's Advanced Mode sample program from the Python wrapper to set the mode, please?
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py

import pyrealsense2 as rs
    import numpy as np
    import cv2
    import json
    import time

    jsonObj = json.load(open("config/setting435High.json"))
    json_string = str(jsonObj).replace("'", '\"')

    # Configure depth and color streams
    pipeline = rs.pipeline()
    config = rs.config()

    freq = int(jsonObj['stream-fps'])
    print("W: ", int(jsonObj['stream-width']))
    print("H: ", int(jsonObj['stream-height']))
    print("FPS: ", int(jsonObj['stream-fps']))
    config.enable_stream(rs.stream.depth, int(jsonObj['stream-width']), int(jsonObj['stream-height']), rs.format.z16,
                         int(jsonObj['stream-fps']))
    config.enable_stream(rs.stream.color, int(jsonObj['stream-width']), int(jsonObj['stream-height']), rs.format.bgr8,
                         int(jsonObj['stream-fps']))
    cfg = pipeline.start(config)
    dev = cfg.get_device()
    advnc_mode = rs.rs400_advanced_mode(dev)
    advnc_mode.load_json(json_string)

    try:
        while True:

            # Wait for a coherent pair of frames: depth and color
            frames = pipeline.wait_for_frames()
            depth_frame = frames.get_depth_frame()
            color_frame = frames.get_color_frame()
            if not depth_frame or not color_frame:
                continue

            # Convert images to numpy arrays
            depth_image = np.asanyarray(depth_frame.get_data())
            color_image = np.asanyarray(color_frame.get_data())

            # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

            # Stack both images horizontally
            images = np.hstack((color_image, depth_colormap))

            # Show images
            cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
            cv2.imshow('RealSense', images)
            key = cv2.waitKey(1)
            # Press esc or 'q' to close the image window
            if key & 0xFF == ord('q') or key == 27:
                cv2.destroyAllWindows()
                break

    finally:

        # Stop streaming
        pipeline.stop()

this is my sample program

MartyG-RealSense, 2020-05-12:
I went through your script very carefully.  Currently you have the json setup lines positioned before the pipeline start.  I would recommend trying to move them into the pipeline, just above the load statement, so it looks somethig like this:

advnc_mode = rs.rs400_advanced_mode(dev)
**jsonObj = json.load(open("config/setting435High.json"))
json_string = str(jsonObj).replace("'", '\"')**
advnc_mode.load_json(json_string)

xiaoniu001, 2020-05-13:
> I went through your script very carefully. Currently you have the json setup lines positioned before the pipeline start. I would recommend trying to move them into the pipeline, just above the load statement, so it looks something like this:
> 
> dev = cfg.get_device()
> advnc_mode = rs.rs400_advanced_mode(dev)
> **jsonObj = json.load(open("config/setting435High.json")) json_string = str(jsonObj).replace("'", '"')**
> advnc_mode.load_json(json_string)
> 
> try:
> while True:

Thank you , 
I have found the cause of the error, and I found that I am connected to the USB hub before reporting the error

MartyG-RealSense, 2020-05-13:
Thanks very much for the update.  Were you able to correct the error after finding its cause, please?

closed 2020-05-30 by MartyG-RealSense