realsenseai/librealsense #6381: In unity,How to get the instrinsics of the RealSense?

coldcoda, 2020-05-11:
Hello, when I use Unity to build a project, how to get the instrinsics of the Camera?(I can't find the parameters)
And, How to transform a pixel coordinate in RGB image that I needed to the corresponding 3D coordinate?
Thanks a lot!!!

MartyG-RealSense, 2020-05-11:
The link below provides guidance about getting the intrinsics in Unity.

https://github.com/IntelRealSense/librealsense/issues/1570

If you are aiming to convert 2D information to 3D information to generate a point cloud, I wonder if the Unity wrapper's point cloud example program might help you to achieve the same goal if you do not need to retrieve the values of the 3D coordinate.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity#point-cloud

If you have not installed the RealSense Unity package already, go to the 'Assets List' at the base of the SDK release page details for the SDK version that you are currently using (e.g 2.34.0) and download and run the **Intel.RealSense.unitypackage** file.

https://github.com/IntelRealSense/librealsense/releases/

[image]

In Unity, go to **Assets > Scenes > Start Here** to access a RealSense home-screen interface where the sample programs can be launched.

[image]

coldcoda, 2020-05-11:
Thanks! I just want to  convert some 2D information that I need to 3D information,not all of the 2D information. I don't know how to get it.
And, I have follow the the link you provided above https://github.com/IntelRealSense/librealsense/issues/1570,
but I am so sorry, I still don't understand how to get the instrinsics.

MartyG-RealSense, 2020-05-11:
I apologise for the delay in responding further.  If you need the intrinsics of the camera but do not need to read them from within your program, there are other ways to get the intrinsics details.

The RealSense SDK has a tool called **rs-enumerate-devices** that can be used to get a list of all the intrinsics and extrinsics of your camera.  If you are using Windows, there is a pre-built version of the tool that you can run.

https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-615365409

The tool can also be compiled from source code using the files here:

https://github.com/IntelRealSense/librealsense/tree/master/tools/enumerate-devices

wodndb, 2020-05-19:
You can get intrinsics using RealSense SDK for unity in [VideoStreamProfile.cs](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Intel.RealSense/StreamProfiles/VideoStreamProfile.cs)

Check the function "OnStartStreaming()" in RsPointCloudRenderer.cs. the type of variable 'depth' is VideoStreamProfile which has member function GetIntrinsics().

And You can align color frame coordinate and depth coordinate using Rs processing pipe. Check the example "BackgroundSegmentation" in RealSense SDK for unity. RsProcessingPipe object has the component "Rs ProcessingPipe" and which has the profile "BackgroundSegmentation". It has a option to align between depth and color frame (RsAlign). If you want to know more information about  processing pipe, read this Intel whitepaper ([link](https://dev.intelrealsense.com/docs/depth-post-processing)).

MartyG-RealSense, 2020-05-19:
@wodndb   Thanks for your detailed advice about getting RealSense intrinsics in Unity!

MartyG-RealSense, 2020-05-30:
@coldcoda   Do you require further assistance please, or can this case be closed?   Thanks!

MartyG-RealSense, 2020-06-05:
Case closed due to no further comments.

closed 2020-06-05 by MartyG-RealSense