realsenseai/librealsense #6382: Bug in rs2_project_point_to_pixel() (rsutil.h)?

rafaelspring, 2020-05-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| SDK Version                            |  2.34.0 | 

### Issue Description
The same code path is used for "modified Brown Conrady" and "inverse Brown Conrady" models:

    static void rs2_project_point_to_pixel(...)
    {
      float x = point[0] / point[2], y = point[1] / point[2];

      if ((intrin->model == RS2_DISTORTION_MODIFIED_BROWN_CONRADY) ||
          (intrin->model == RS2_DISTORTION_INVERSE_BROWN_CONRADY))
      {

          float r2  = x*x + y*y;
          float f = 1 + intrin->coeffs[0]*r2 + intrin->coeffs[1]*r2*r2 + intrin->coeffs[4]*r2*r2*r2;
          x *= f;
          y *= f;
          float dx = x + 2*intrin->coeffs[2]*x*y + intrin->coeffs[3]*(r2 + 2*x*x);
          float dy = y + 2*intrin->coeffs[3]*x*y + intrin->coeffs[2]*(r2 + 2*y*y);
          x = dx;
          y = dy;
      }

At the same time the description for the "inverse Brown Conrady" model is:
_Equivalent to Brown-Conrady distortion, except undistorts image instead of distorting it_

So in the case of RS2_DISTORTION_INVERSE_BROWN_CONRADY the code seems to be using an undistorion model when converting to pixel space, instead of a distortion model, which doesn't make sense.

So is this a bug or are RS2_DISTORTION_INVERSE_BROWN_CONRADY and RS2_DISTORTION_MODIFIED_BROWN_CONRADY just assumed equivalent for sensors that are projected into (e.g. RGB)?

ev-mp, 2020-05-11:
@rafaelspring hello, the enums names sound confusing, partly as they are not meant to describe whether their usage is exclusively distort or un-distort, but to define the model selected for producing the distortion coefficients.
In both cases 3D->2D projection **distorts** the texture coordinates, i.e. it assumes that the 3D input is correct but the target 2D frame data is intrinsically distorted, as in the RGB sensor case. Note however, that when the 3D->2D projection is applied to undistorted data, as with Aligning Color to (undistorted) Depth/IR, then those coefficients are zeroed no distortion occurs.

In 2D->3D de-projection a reverse distortion is applied to un-distort the pixel coordinates accordingly.

rafaelspring, 2020-05-11:
@ev-mp Ok, so can I assume that rs2_project_point_to_pixel() is doing the right thing and that if I get `RS2_DISTORTION_INVERSE_BROWN_CONRADY` as the model for a sensor that I project into, then I can just use the coefficients and plug it into a normal distortion model?

I guess the failure case is if you have a sensor that can be used both to project "out of" as well as to project "into". Because as far as I see it, there is no consistent model that you could assign to such a sensor that would handle both projection as well as de-projection correctly.

ev-mp, 2020-05-11:
@rafaelspring, 
> ..then I can just use the coefficients and plug it into a normal distortion model

In case by 'normal distortion model' you refer to Brown-Conrady model equations - then probably yes. 

>there is no consistent model that .. would handle both projection as well as de-projection correctly

The SDK provides clear evidence to the contrary - each calibrated 2D sensor with a well-defined intrinsic (and extrinsic) matrix can and will be used to be "project onto" or "project from".
For instance, when registering Depth to Color then the Color is "projected onto", When you align in the opposite direction the the Color frame is used to "project out of". So the distortion/reverse distortion methods are invoked consistently in a symmetric manner.

rafaelspring, 2020-05-11:
@ev-mp 
Uhh.. Nope. The math just doesn't work out.
Have a look in the code and try `deproject(project(X))` for a nonzero point `X` and a nonzero Inverse-BC coefficient vector.  Are you getting x back?

For the simplest of cases, try `[a 0 0 0 0]` as dist. coefficients with a nonzero a.

rafaelspring, 2020-05-26:
@ev-mp Just found [this page](https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#point-coordinates) where things are explained reasonably well:

_**Inverse Brown-Conrady Distortion**
An image is distorted, and has been calibrated according to the inverse of the Brown-Conrady Distortion model. This model provides a closed-form formula to map from distorted points to undistored points, while mapping in the other direction requires iteration or lookup tables. Therefore, images with Inverse Brown-Conrady Distortion are being undistorted when calling rs2_deproject_pixel_to_point(...). This model is used by the RealSense SR300's depth and infrared image streams.
Although it is inconvenient that projection and deprojection cannot always be applied to an image, the inconvenience is minimized by the fact that RealSense devices always support deprojection from depth images, and always support projection to color images. Therefore, it is always possible to map a depth image into a set of 3D points (a point cloud), and it is always possible to discover where a 3D object would appear on the color image._

closed 2020-05-26 by rafaelspring