realsenseai/librealsense #6384: What is the unit of rs.option.exposure (python API)?

jurc192, 2020-05-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 (D435i) | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version | Arch Linux | 
| Kernel Version (Linux Only)    |  5.6.7-arch1-1 | 
| Platform                                 | PC  |
| SDK Version                            |  2.3.3.1 | 
| Language                            |  python                        | 
| Segment			|  others | 

### Question
What is the unit of exposure setting when using python API?
[Documentation](https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet) says that exposure range is 1 -> 166 milliseconds.

Using `sensor.get_option_range(rs.option.depth_units)` gives this answer:
```
1-165000/1 [8500]
```
What does this mean? Is it microseconds?

ev-mp, 2020-05-11:
That's correct - the exposure range is within 166 msec configured with microsec. granularity.
[image]

jurc192, 2020-05-11:
Great, thanks! :)

closed 2020-05-11 by jurc192