realsenseai/librealsense #6385: IMU stream not woking in D435i

malith1992, 2020-05-12:
| Camera Model | { D435i } |
| Operating System & Version | {Ubuntu 18.4} |
| Platform | NVIDIA Xavier |
| Language | {python } |
| Segment | {Robot} |

### Issue Description

I want to get the angle of the camera. So I run the following code.

```
import pyrealsense2 as rs
import numpy as np

#get camera angle

anglepip = rs.pipeline()
angleconfig = rs.config()
angleconfig.enable_stream(rs.stream.accel)
anglepip.start(angleconfig)

camera_angle = 0

try:
    while True:

        

        f1 = anglepip.wait_for_frames()
        accel = f1[0].as_motion_frame().get_motion_data()

        if not accel:
            print("no frame at cam ")
            continue

        accel_angle_x = math.degrees(math.atan2(accel.y, accel.z))
        accel_angle_x = int(accel_angle_x)
        camera_angle= accel_angle_x
        break

finally:
    anglepip.stop()

print("camera angle = " + str(camera_angle))
```

But the pipline is not giving any frames. I got this error

> Traceback (most recent call last):
>   File "camera_test.py", line 18, in <module>
>     f1 = anglepip.wait_for_frames()
> RuntimeError: Frame didn't arrived within 5000

It works fine on windows PC. But it doesn't work on NVIDIA Xavier.

ev-mp, 2020-05-12:
@malith1992 , IMU drivers are not included in NVIDIA kernels by default. Support for IMU streaming is provided by an alternative implementation in SDK and requires building CMake with -`DFORCE_RSUSB_BACKEND=true`.
https://github.com/IntelRealSense/librealsense/issues/6112#issuecomment-603970215

malith1992, 2020-05-13:
> @malith1992 , IMU drivers are not included in NVIDIA kernels by default. Support for IMU streaming is provided by an alternative implementation in SDK and requires building CMake with -`DFORCE_RSUSB_BACKEND=true`.
> [#6112 (comment)](https://github.com/IntelRealSense/librealsense/issues/6112#issuecomment-603970215)

I have build the CMake with the mentioned flag. Now it gives the angle only once after I connect the camera. If I run the code again, it gives the same error. If I unplug and replug the camera, it gives the angle again for the fresh start.

ev-mp, 2020-05-13:
@malith1992 hello, we'll look and try to reproduce the issue you describe. What SDK version you're using?

The code that you posted will not calculate the camera angles (I suppose you mention the camera inertial orientation or Euler angles):
 - The accelerometer measures linear forces , i.e. forces applied along the body axes. So the angle  you extract represents the vector of combined forces (gravity + inertial acceleration) applied to the camera's body. It doesn't reflect the body orientation.   
The accelerometer can be used to calculate the camera translation vector w.r.t to an arbitrary start point by using  X=Vt+(at^2)/2.

To track the camera's rotation matrix you actually need to measure and integrate the gyro sensor readings.
Combining those two in theory is sufficient to calculate the objects 6-DOF. In practice the inertial measurements are intrinsically noisy and prone to things such as hysteresis. Therefore filtering and additional sensors are often offered to address this, like a dedicated T265 device by Intel.

To get the orientation matrix take a look at the following Librealsense demo -https://github.com/IntelRealSense/librealsense/tree/master/examples/motion

malith1992, 2020-05-14:
> @malith1992 hello, we'll look and try to reproduce the issue you describe. What SDK version you're using?
>
> The code that you posted will not calculate the camera angles (I suppose you mention the [...]

I used Jetpack 4.3 for the Xavier.

It's my mistake to mention it as camera angle. To be honest, I need the Euler angle as written in the code. Thank you for look in to it.

RealSenseSupport, 2020-06-19:
Just curious about the status because the test script seems to work fine here:
```
nvidia@xavier2:~/work/librealsense$ git log
commit 8e04cdd59276f60dc625cf8ef974daf833bc8d18 (HEAD)
Author: ev-mp <ev_mp@yahoo.com>
Date:   Sun Jun 14 09:47:12 2020 +0300

    Fix CUDA compilation. Fix gcc-pedantic remarks. Addresses #6573

commit 8594d09f092347a8b3d832d14e4fb631140620c5 (tag: v2.35.2)
Merge: a9b2615a8 91cc6c73d
Author: ev-mp <evgeni.raikhel@intel.com>
Date:   Tue Jun 9 21:05:47 2020 +0300

    Merge pull request #6557 from IntelRealSense/rc3
    
    v2.35.2 - Merge to development

nvidia@xavier2:~/github-6385$ uname -ra
Linux xavier2 4.9.140-tegra #1 SMP PREEMPT Mon Dec 9 22:52:02 PST 2019 aarch64 aarch64 aarch64 GNU/Linux
nvidia@xavier2:~/github-6385$ cat /etc/nv_tegra_release 
# R32 (release), REVISION: 3.1, GCID: 18186506, BOARD: t186ref, EABI: aarch64, DATE: Tue Dec 10 07:03:07 UTC 2019
nvidia@xavier2:~/github-6385$ rs-enumerate-devices -s
Device Name                   Serial Number       Firmware Version    
Intel RealSense D435I         843112073792        05.12.05.00         
nvidia@xavier2:~/github-6385$ python3 ./github-6385.py 
camera angle = 1
nvidia@xavier2:~/github-6385$ python3 ./github-6385.py 
camera angle = -86
nvidia@xavier2:~/github-6385$ python3 ./github-6385.py 
camera angle = -179
nvidia@xavier2:~/github-6385$ python3 ./github-6385.py 
camera angle = 0
```
The ticket will be closed in 7 days if no further help is needed. Thank you.

closed 2020-06-26 by RealSenseSupport