realsenseai/librealsense #6386: New version realsense canot support low usb power supply

FrankCreen, 2020-05-12:
I bought a new realsense d435i and use it on the jetson xaiver.If I pulgging the realsense on the jetson it is working successfully. But if i use a  USB hub,the jetson cannot recogniezed it,but the mouse and keyboard working successfully.

What should I do?

MartyG-RealSense, 2020-05-12:
There have been a few recent cases of problems with USB hubs on Xavier boards.  Usually though, at least one camera works on the hub and problems only emerge if more than one RealSense camera is attached to the hub.  

If you are only using one camera, would it be possible to have the camera directly connected to one of the Xavier AGX's two USB 3 ports and connect the USB hub to the other USB 3 port?  Then you could attach all the other USB devices such as keyboard and mouse to the hub.

MartyG-RealSense, 2020-06-10:
Do you still require assistance with this case or can it be closed, please?

If you are only using a single camera, the SDK's Jetson documentation strongly suggests enabling the power **barrel jack** connector to help with power requirements.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#getting-started

FrankCreen, 2020-06-11:
> Do you still require assistance with this case or can it be closed, please?

No Problem。Thanks！

MartyG-RealSense, 2020-06-11:
Thanks very much!

closed 2020-06-11 by MartyG-RealSense