realsenseai/librealsense #6390: D435 Errors after powering off then on back on with a 8 minute delay.

577nitro, 2020-05-13:
| | |
|---|---|
|**librealsense**|2.32.1 RELEASE|
|**OS**|Mac OS|
|**Name**|Intel RealSense D435|
|**Serial Number**|944122073552|
|**Firmware Version**|05.12.03.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|944523023778|
|**Firmware Update Id**|944523023778|

D435 outputs error code 999 for distance after being power cycled with an ~8 minute delay.  Camera strays in error mode until powered off and on again.  This is experienced with at least 5 different cameras, and is repeatable 60% of the time.  Camera is used on s drone system and is used for closest object detection.  It is calibrated and tested before being installed.  Seen in both bright and dim sunlight.   Hardware/Camera is very flakey and unreliable. 

Please provide a description of the problem

MartyG-RealSense, 2020-05-13:
When you say "power cycled", do you mean a 'soft reboot' of resetting the computer (not shutting it down and then using the power button to turn it on again)?  

And when you say "powered off and on". do you mean shutting the computer down and then restarting it with the power button?

This sounds like a known issue where the camera can work fine the first time that the computer is booted up and then the camera may not be detected after a reboot of the computer is performed to start it up a second time.  Shutting the PC down and then starting it up with the power button corrects the problem.

The link below has a long discussion of a recent example of this phenomenon from February / March 2020.

https://github.com/IntelRealSense/librealsense/issues/5913

A couple of options offered for a solution in that discussion are to either set an option called **rtcwake** in the computer's BIOS or to use a mains electricity powered USB 3 hub with the camera.  For your drone application, it would be ideal if **rtcwake** could correct the issue, since you would not be able to connect a mains electricity powered hub to a drone.

https://github.com/IntelRealSense/librealsense/issues/5913#issuecomment-595579190

577nitro, 2020-05-20:
Thanks for the info...will try to implement.  By power off..I mean, power off to the camera, it hooked up to a C410 dragon board, and the board is powered off.  What we do see is that when it comes backup, the video works but the 3d distance camera doesn't. We get no distance information and it won't until it is hard reset.

MartyG-RealSense, 2020-05-20:
When a RealSense camera has a problem with its drivers, the RealSense functions such as depth sensing may become unavailable but video is still accessible because the computer is treating the camera as an ordinary USB video webcam.  Hence why RGB video may still work.

I look forward to hearing about the results of your next tests.  Good luck!

MartyG-RealSense, 2020-05-30:
Do you require further assistance please, or can this case be closed?   Thanks!

577nitro, 2020-05-30:
I have more information.   THe problem has been drilled down to the
3d/Distance Camera returning objects to be closer than .1 meter when there
are no objects at all that close.  It will work correctly for about 15 to
30 minutes then start having this issue continuously...for hours.
Sometimes itself heals after three or four hours...sometimes it never
recovers.  If it starts the problem can be temporarily fixed by power it on
and off.   This is the really bad part...every camera we've tried does
this.  We have tried 10 camera's all of them fail this way.  They all have
the latest firm installed as well.   These are all running dragon board
C410...however, I see problems testing on a MacPro as well.   We have tried
reducing the frame rate down to 6 the lowest and it helps some...but we
need it to be at least 15 for safety reasons.

Thanks
Jason

MartyG-RealSense, 2020-05-30:
I would recommend using 15 FPS as a minimum anyway, as errors can occur at 6 FPS that disappear when running at 15 FPS.

Because your camera will not work correctly unless the computer undergoes a hard reboot (full power-down and restart with the power button), it suggests that using a script with a **hardware_reset()** instruction to automatically reset a camera with faulty conditions in its image would not be of use to you.

If it works correctly for the first 15 to 30 minutes, it is probably not related to the IR Emitter saturating the IR Imager with light in dim or bright lighting conditions and disrupting it.

If you are only using one camera on the drone then it would not be related to the build-up of electrostatic discharge that can occur in a multicam system with sync cables.

I once handled a case where a D435 was being used on a flying drone where there were random artifacts appearing on the camera stream that resembled insects.  The case has a short video.  Do you have anything resembling this on your drone image?

https://github.com/IntelRealSense/librealsense/issues/2875#issuecomment-446486850

MartyG-RealSense, 2020-06-05:
@577nitro   Do you require further assistance with this case please, or can it be closed?   Thanks!

closed 2020-06-14 by MartyG-RealSense