realsenseai/librealsense #6393: D400 On-chip self-calibration crashes realsense viewer and works only on 30 fps on rs-depth-quality

ArviSyrjanen, 2020-05-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.3 |  
| Operating System & Version |   Linux Ubuntu 18  | 
| Kernel Version (Linux Only)    |  4.15 Bionic | 
| Platform                                 | PC |
| SDK Version                            |  2.34.0 }                          | 

Hello!

Im trying to use the new On-chip self-calibration feature, but when using it through realsense-viewer GUI, the program hangs and never recovers.
[image]

I did manage to use the self-calibartion feature through the rs-depth-quality application.
[image]

But even then I can only use the rate of 30 fps, if any other frame rate is chosen the calibration stops at 0% and never finishes.
[image]

This prevents me from using the 256x144 resolution for self-calibration recommended in the [documentation](https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras), which is available only in 90 fps.
[image]

MartyG-RealSense, 2020-05-14:
There is an ongoing issue with SDK version 2.34.0 dating back to this SDK version's release at the end of March 2020, where the error message "**get_device_time_ms() took too long (more then 2 mSecs)**" can be generated continuously in the log window at the bottom of the Viewer window and On-Chip Calibration freezes the Viewer when used.  I have experienced both of these problems myself in Windows 10 and still do.

A fix was developed in a Pull Request but it has not passed merging into the RealSense SDK yet.

https://github.com/IntelRealSense/librealsense/pull/6150

In the meantime, if you are affected by these errors then I recommend using the earlier SDK version **2.33.1** if possible until the **get_device_time_ms()** situation is rectified.

RealSenseCustomerSupport, 2020-05-15:
----------------------------------------------

Hi @ArviSyrjanen
As to the issue about on-chip calibration hangs with Ubuntu 18.04
Thanks for your input. we will looking into this issue
by the way, could you confirm the kernel version you are using?
$uname -r
eg. 5.3.x ?

ArviSyrjanen, 2020-05-15:
The kernel version is 4.15.0

RealSenseSupport, 2020-05-15:
@ArviSyrjanen 
Yes, also able to duplicate the issue with  my Ubuntu PC (18.04) with different camera sku (D415, D435).. We will looking into this issue definitely.

Hi @agrunnet ,  is there anything we need to configure or limitation that running on-chip calibration on Ubuntu 18.04

agrunnet, 2020-05-15:
From a programming perspective you need to make sure you use the poling mode and not the interrupt mode. (This is one of the flags in the call).

RealSenseSupport, 2020-05-19:
Hi @ArviSyrjanen

before have newer version of viewer with the fix

you can either running 2.34 viewer on win10 PC

or running 2.33.1 viewer on your ubuntu PC   #6231

RealSenseCustomerSupport, 2020-05-20:
----------------------------------------------

Hi @ArviSyrjanen,
There  a fix that already commit to the latest development branch
you can check the detail from this pull request
https://github.com/IntelRealSense/librealsense/pull/6427
best

closed 2020-05-26 by RealSenseCustomerSupport

RealSenseSupport, 2020-06-01:
The Realsense SDK (build 2.35) is just released which has the fix