realsenseai/librealsense #6394: Use Qt5 to create a simple app with realsense camera

duongbaoquy, 2020-05-15:
Hello,
I am using realsense D415 camera. Now I'm exploring and will use realsense in conjunction with Qt Creator to do a basic app (for example using a button to turn the camera on and off) but I can't find an example in using Qt5 to Create applications using realsense
I have researched and created some applications that use cameras from USB - but it seems that it is different from realsense cameras - so can someone share and share some documents or some basic examples?
Sorry if this was a silly question.
Thanks for reading!

MartyG-RealSense, 2020-05-15:
Hi @duongbaoquy   The website **JetsonHacks** (a site supporting Nvidia Jetson computing boards) has a tutorial video on YouTube for using s Jetson with QTCreator and Librealsense2.  The process may not be exactly the same if you are not using a Jetson board, but the example in the video should hopefully provide some direction about setting up a Librealsense-enabled QTCreator on your own computer.

https://www.youtube.com/watch?v=Yed4HIepykw

duongbaoquy, 2020-05-15:
Thanks for the quick reply.
There seems to be a mistake here, I mean, I want to create my own app - like the image below - The video you gave me was watching, but that's not the point of my question. . Maybe the following picture will help you figure out what I want to do.

[image]

MartyG-RealSense, 2020-05-15:
I am sorry if I misunderstood, as your original message suggested that you wanted to find out how to get QTCreator to recognise a RealSense camera instead of an ordinary USB camera.

Do you get any error messages when trying to access the camera?

duongbaoquy, 2020-05-15:
Not really - I think that realsense camera and ordinary USB camera have a few differences and I don't know how to replace those points. For example, with an ordinary USB camera, you do not need to use the get frame function, but with realsense, so how to use the get_frame function in Qt Creator ?

MartyG-RealSense, 2020-05-15:
Apologies for the delay in responding further.  If the RealSense camera will not display, I wonder if it is similar to a problem experienced with using the camera for video streaming with some programs such as Skype.  What I mean is, when using the camera with such a program, you may get a blank image because the program expects you to select a certain RealSense driver to use for the video streaming instead of the image appearing automatically.  This subject was discussed recently at the link below:

https://github.com/IntelRealSense/librealsense/issues/6209#issuecomment-615091643

Do you have a way in QTCreator to select a camera driver in video settings, please?

duongbaoquy, 2020-05-16:
Oh thanks, I think I'm figuring out some of the ways to do it. I can handle it myself from now on.
Thanks

closed 2020-05-16 by duongbaoquy

MartyG-RealSense, 2020-05-16:
@duongbaoquy    Thanks and good luck!