realsenseai/librealsense #6396: Intel RealSense Viewer Depth Sensor or image frozen

arossbach10, 2020-05-15:
| | |
|---|---|
|**librealsense**|2.33.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense SR300|
|**Serial Number**|635201003298|
|**Firmware Version**|3.27.3.0|
|**Camera Locked**|YES|
|**Product Line**|SR300|
|**Asic Serial Number**|635201003298|
|**Firmware Update Id**|635201003298|

It works for some time and then the depth image or the image is frozen.
In 2D view, I see "no frames received". It happens for both channels.
The other one still works.

MartyG-RealSense, 2020-05-15:
Hi @arossbach10   When you say "the other one" works, do you mean the infrared stream please?   Or the '3D' point cloud mode.   Thanks!

arossbach10, 2020-05-15:
I mean:
depth is frozen and RGB is working fine.
And the other way around.

Now I tested again.
When depth was working the IR channel was working too. RGB was frozen.

André.

MartyG-RealSense, 2020-05-15:
Thank you.   And how long does it work for before one of the streams gets frozen please?

arossbach10, 2020-05-15:
Different. Sometimes a few seconds and other times several minutes. But it happens regularly.

MartyG-RealSense, 2020-05-15:
When the 'No frames received' message appears, are there any red error messages on-screen?  If there are, what do they say please?

[image]

arossbach10, 2020-05-15:
No. Just something like this:
[image: image.png]

MartyG-RealSense, 2020-05-15:
Sorry, the image did not post so I cannot see it. =

arossbach10, 2020-05-15:
> Sorry, the image did not post so I cannot see it. =
Now the link should work.

MartyG-RealSense, 2020-05-15:
Thanks for the image, it was very helpful.  I found a past case that sounded similar to what you are experiencing.

https://github.com/IntelRealSense/librealsense/issues/840#issuecomment-355961686

arossbach10, 2020-05-15:
Thank you. 
What I noticed. I actually bought an SR305, but the camera is recognized as SR300.

MartyG-RealSense, 2020-05-15:
You could try updating the firmware driver in the RealSense Viewer to the recommended version.  I had an SR300 that I updated in this way and it is now recognised in the Viewer as an SR305, even though it was not one originally.

https://support.intelrealsense.com/hc/en-us/community/posts/360034808474/comments/360008729034

closed 2020-06-05 by MartyG-RealSense