realsenseai/librealsense #6398: [T256] SDK Hangs forever on exit

Yanis-F, 2020-05-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T256 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Windows 10  | 
| Platform                                 | PC |
| SDK Version                            |   2.34.0 / 2.34.1                           | 
| Language                            | C++                          | 

### Issue Description
The SDK hangs up forever about 50% of the time when destroying the `rs2::pipeline` or `rs2::config` (not sure which one) using a T256 sensor.

### How to reproduce

With a realsense T256 tracking camera plugged in in a USB 2 or 3 port :
1. clone the repo 
`git clone https://github.com/IntelRealSense/librealsense`

2. checkout to v2.34.0 (not: same result with v2.34.1)
`cd librealsense`
`git checkout v2.34.0`

3. Generate visual studio 2019 solution
`mkdir build`
`cd build`
`cmake ..`

4. Open the generated solution
`.\librealsense2.sln`

5. Change rs-pose.cpp to exit at some point
In Examples/rs-pose/Sources File/rs-pose.cpp, change line 19 from
`while (true)`
to
`for (int i = 0; i < 10; ++i)`

6. Set rs-pose as Startup project
Right click on rs-pose project > Set as Startup Project

7. Run the program (maybe several time, it's random) and see what seems to be a dead-lock on program exit

-------------------------------------------------------------------------

Here is the call stack of the main thread of where it get stuck :
[image]

Yanis-F, 2020-05-22:
up

Yanis-F, 2020-06-04:
I would be very happy if anyone could come up with a better solution.

What I did is that I launched everything in a separate thread, that I detached before destroying the objects. So the destruction may not complete but at least my entire program do not get stuck.

I'm still very worried that this will cause issues with the sensor because I have no guarantee that every resource needed is correctly released

RealSenseSupport, 2021-03-17:
This PR may help your situation.  Have a look and see if it fixes your issue.

https://github.com/IntelRealSense/librealsense/pull/8561

closed 2021-04-06 by RealSenseSupport