realsenseai/librealsense #6400: T265 - Pose stream + Fisheye stream

0y8w1zx, 2020-05-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.926 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.34.0                         | 
| Language                            |  C                         | 
| Segment			|  VR                          | 

### Issue Description
I try to communicate with the T265 over USB. I use the C programming language and the libusb library. So far I can get the pose data without problems and also the image streams from both fisheye cameras. But when I try to enable both of those streams (pose and image) it doesn't work. Is there something to consider ? What I found is, when I configure the stream to output nothing to the host it works fine (no image stream). And secondly when I enable the image stream (setting the output for the fisheye profiles to 1), it doesn't work unless I don't enable the 6DOF thing.

Is there something obvious I am missing ?

Alnilam1305, 2020-05-27:
Can you share some code snippet, how you setup the streams? This would be easier to follow and reproduce, what could be missing.

0y8w1x, 2020-06-15:
I solved it by enabling both and read either one with a thread. Thanks

closed 2020-06-15 by 0y8w1x