realsenseai/librealsense #6401: SR300 updated firmware detected as SR305 but Windows Hello no longer working

zanechua, 2020-05-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |  Win 10  |                                   | 
| Platform                                 | PC  |
| SDK Version                            |  2.34.0                   | 

### Issue Description
SR300 was updated to the latest 3.27.3.0 firmware and I noticed that the virtual driver is no longer installed and Windows Hello functionality no longer works. I am also unable to revert to the previous 3.26.3.0 firmware as the installation always fail with realsense viewer reporting that the recovery device failed to connect in time or no device was detected in fw update mode.

Already saw this and tried it: https://support.intelrealsense.com/hc/en-us/community/posts/360037842634-SR300-firmware-updated-and-is-now-detected-as-SR305

Trying to downgrade via the `rs-fw-update` tool doesn't work either

```
PS C:\Program Files (x86)\Intel RealSense SDK 2.0\tools> ./rs-fw-update -f SR3XX_FW_Image-3.26.1.0.bin
 16/05 10:03:16,269 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,285 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,296 ERROR [21320] (win-helpers.cpp:550) Could not find camera (vid 8086 pid b48 uid 1fc4ee5c) in windows device tree
 16/05 10:03:16,313 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,326 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,336 ERROR [21320] (win-helpers.cpp:550) Could not find camera (vid 8086 pid b48 uid 1fc4ee5c) in windows device tree
 16/05 10:03:16,374 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,387 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,398 ERROR [21320] (win-helpers.cpp:550) Could not find camera (vid 8086 pid b48 uid 1fc4ee5c) in windows device tree
 16/05 10:03:16,419 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,433 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,442 ERROR [21320] (win-helpers.cpp:550) Could not find camera (vid 8086 pid b48 uid 1fc4ee5c) in windows device tree
 16/05 10:03:16,463 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,477 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,485 ERROR [21320] (win-helpers.cpp:550) Could not find camera (vid 8086 pid b48 uid 1fc4ee5c) in windows device tree
 16/05 10:03:16,506 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,520 ERROR [21320] (win-helpers.cpp:688) CreateFile failed
 16/05 10:03:16,528 ERROR [21320] (win-helpers.cpp:550) Could not find camera (vid 8086 pid b48 uid 1fc4ee5c) in windows device tree

updating device:
Name: Intel RealSense SR305, serial number: 617205002936, update serial number: 617205002936, firmware version: 3.27.3.0, USB type: unknown
 16/05 10:03:16,765 ERROR [21320] (messenger-winusb.cpp:102) bulk_transfer failed, error: 23

failed to locate a device in FW update mode
```

MartyG-RealSense, 2020-05-16:
There was a RealSense user who posted a method for restoring an SR300 and its Virtual Driver and got Windows Hello working again.  It is not an officially recommended procedure so it is at your own risk to try it.

https://support.intelrealsense.com/hc/en-us/community/posts/360038275074/

You can reduce the risk of the process by using Intel's official SR300 offline driver installation package if possible instead of the guide creator's own driver download.  This is available by visiting the link below, finding the section headed '**Intel RealSense Camera SR300 Firmware**' and clicking on the '**Drivers**' link beneath it to launch a download of the SR300 Depth Camera Manager (DCM) driver installation program.

https://software.intel.com/en-us/realsense-sdk-windows-eol

zanechua, 2020-05-16:
Hmm. My virtual driver was working prior to this update, it was erractic but it did register and was present. However with the upgrade to the new 3.27 firmware, it's totally gone.

I'm not sure that my SR300 is even being recognised as an SR300 anymore. It's recognised as an SR305 which shouldn't be the case. Wouldn't it be a better idea to not recommend the SR305 driver for SR300 owners? Removing functionality via firmware update without a way to revert back seems like a misstep.

MartyG-RealSense, 2020-05-16:
Yesterday I did an update of my old SR300 to the recommended firmware in the RealSense Viewer and it is now detected as SR305 even though it was an SR300 originally (the SR305 still has an SR300 camera circuit board inside it).  So the update to a detected status of SR305 that you experienced seems to be normal.

The original purpose of the Virtual Driver was to allow the original generation of RealSense devices to be able to be accessed by non-RealSense applications such as video chat if the camera was currently also being accessed by a RealSense camera application.  

Intel did not continue with Windows Hello support for RealSense cameras after the SR300.  For a lot of people the Windows Hello functionality did not work or had problems.  

If Windows Hello support and ensured compatibility with old software that supports SR300 is important to you then the DCM program, or the linked-to guide if that does not fix the problem on its own, may be your best chance of restoring the camera to the status of an original SR300.  I apologise for the inconvenience.

zanechua, 2020-05-16:
I see. Understood.

I'll test out the guide and see if that works. Will update this.

MartyG-RealSense, 2020-05-16:
Thanks, I look forward to your update.   Good luck!

MartyG-RealSense, 2020-05-30:
Do you require further assistance please, or can this case be closed?  Thanks!

MartyG-RealSense, 2020-06-05:
Case closed due to no further comments.

closed 2020-06-05 by MartyG-RealSense

zanechua, 2021-01-11:
I'm revisiting this issue again and unfortunately I was unable to recover the camera into a working condition. After the update, the camera now only detects as an Audio Device and does not even seem to be able to see an Imaging Device at all. Guess the camera is bricked.

MartyG-RealSense, 2021-01-11:
Can you check whether you have a **Cameras** category in your Device Manager and if you do, whether it contains the SR300 drivers (Depth, RGB, Virtual).  Thanks very much.

zanechua, 2021-01-11:
It doesn't at all. Can't get any of the imaging devices to appear in Device Manager. All I see is the BlasterX Senz3D Audio Device

MartyG-RealSense, 2021-01-11:
Have you tried the DCM program recently to reinstall the firmware and drivers?

https://github.com/IntelRealSense/librealsense/issues/6401#issuecomment-629594160

zanechua, 2021-01-11:
Yup. Tried the latest DCM program available and still nothing. Was intending to try Linux but at this point I don't even see the imaging device appear in USBView. Here's the relevant debug information:

```
[Port2]  :  USB Composite Device

Is Port User Connectable:         no
Is Port Debug Capable:            no
Companion Port Number:            0
Companion Hub Symbolic Link Name: 
Protocols Supported:
 USB 1.1:                         yes
 USB 2.0:                         yes
 USB 3.0:                         no

Device Power State:               PowerDeviceD0

       ---===>Device Information<===---
English product name: "BlasterX Senz3D VF0810"

ConnectionStatus:                  
Current Config Value:              0x01  -> Device Bus Speed: Full (is not SuperSpeed or higher capable)
Device Address:                    0x28
Open Pipes:                           1

          ===>Device Descriptor<===
bLength:                           0x12
bDescriptorType:                   0x01
bcdUSB:                          0x0200
bDeviceClass:                      0x00  -> This is an Interface Class Defined Device
bDeviceSubClass:                   0x00
bDeviceProtocol:                   0x00
bMaxPacketSize0:                   0x40 = (64) Bytes
idVendor:                        0x041E = Creative Labs
idProduct:                       0x409B
bcdDevice:                       0x0102
iManufacturer:                     0x01
     English (United States)  "Creative Technology Ltd"
iProduct:                          0x02
     English (United States)  "BlasterX Senz3D VF0810"
iSerialNumber:                     0x03
     English (United States)  "170427183848"
bNumConfigurations:                0x01

          ---===>Open Pipes<===---

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x84  -> Direction: IN - EndpointID: 4
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0025 = 0x25 bytes
bInterval:                         0x01

       ---===>Full Configuration Descriptor<===---

          ===>Configuration Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x02
wTotalLength:                    0x0096  -> Validated
bNumInterfaces:                    0x03
bConfigurationValue:               0x01
iConfiguration:                    0x00
bmAttributes:                      0xC0  -> Self Powered
MaxPower:                          0x00 =   0 mA

          ===>Interface Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x04
bInterfaceNumber:                  0x00
bAlternateSetting:                 0x00
[... 121 lines cut ...]
bInterfaceNumber:                  0x02
bAlternateSetting:                 0x00
bNumEndpoints:                     0x01
bInterfaceClass:                   0x03  -> HID Interface Class
bInterfaceSubClass:                0x00
bInterfaceProtocol:                0x00
iInterface:                        0x00

          ===>HID Descriptor<===
bLength:                           0x09
bDescriptorType:                   0x21
bcdHID:                          0x0111
bCountryCode:                      0x00
bNumDescriptors:                   0x01
bDescriptorType:                   0x22 (Report Descriptor)
wDescriptorLength:               0x007C

          ===>Endpoint Descriptor<===
bLength:                           0x07
bDescriptorType:                   0x05
bEndpointAddress:                  0x84  -> Direction: IN - EndpointID: 4
bmAttributes:                      0x03  -> Interrupt Transfer Type
wMaxPacketSize:                  0x0025 = 0x25 bytes
bInterval:                         0x01
```

MartyG-RealSense, 2021-01-11:
And you are plugging it into a USB 3 port?  The SR300 does not work with USB 2.0.

zanechua, 2022-12-20:
Reporting back after almost 2 years from my last response.

Decided to pull the camera out again for one last try before I threw it out and what do you know, on my current system I managed to downgrade the firmware via the steps listed:

https://github.com/IntelRealSense/librealsense/issues/4435#issuecomment-525617614

I was using USB 3 before but perhaps it was some kind of controller issue with my old system or something.

So this issue is truly resolved.

MartyG-RealSense, 2022-12-20:
That's excellent news, @zanechua - thanks so much for the update  :)