realsenseai/librealsense #6402: Install python wrapper on MacOS

ysiegel29, 2020-05-16:
Hello,
I am a beginner and would like to test the D435 on my mac using python wrapper. 
I have literally spent 3 full hours on trying I think all possible options, following previous post suggestions etc.
I need help if possible...

Where I am is that pyrealsense2 cannot install with pip install pyrealsense2
I get this error:
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
I am using python 3.7

Can someone help me?
Thank you!

MartyG-RealSense, 2020-05-16:
@dorodnic the RealSense SDK Manager said in April 2020 that installing pyrealsense2 on MacOS with **pip install pyrealsense2** should now be possible with Python 3.7.

https://github.com/IntelRealSense/librealsense/issues/6246#issuecomment-614695052

Did you obtain the PyPi package for Pyrealsense2 before running the pip install instruction to install the package, please?

https://pypi.org/project/pyrealsense2/

It is also possible to build the wrapper from source, which I believe Python 3.8 users should do for now.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

ysiegel29, 2020-05-16:
thanks a lot for your answer.
Actually I thought a build existed reading some post but there are no file for Mac in https://pypi.org/project/pyrealsense2/ unless I am mistaken.

fyi since then I have it seems been able to build the wrapper with make following those steps https://github.com/IntelRealSense/librealsense/issues/5275

but I still can't import pyrealsense2, even if the .so files are in the project folder... not sure why
Any idea?
Thanks,

MartyG-RealSense, 2020-05-16:
I admit that it can be a slightly awkward process for someone who is new to PyPi.  I have never used Python installation myself yet or done a Pyrealsense2 installation (this is something I still need to learn), and as far as I knew, using the instruction **pip install pyrealsense2** should have been sufficient to launch the installation process.  

So as a Pyrealsense2 installation newbie myself, the instructions preceding **pip install pyrealsense2** can be a bit confusing, as it implies that there are preceding commands to input and files to manually download before you enter the **pip** instruction.

[image]

The PyPi package installation instructions at the official Python website also give the impression that the finding and downloading of the files is automated once a pip instruction for a recognised program package is received.

If a suitable wheel file **(.whl**) cannot be found then it automatically builds one locally (on the user's computer?) and caches it for future use.  So that is perhaps how Mac wheels are generated for Pyrealsense2 if one is not present on the Pyrealsense2 PyPi file download page (this is just my speculation though rather than certain knowledge).

Once Pyrealsense2 is installed by some means (whether PyPi or compilation from source) then some kind of simple test script can be run to test that Pyrealsense2 can be imported into an application.

When Pyrealsense2 was first launched, a simple test script was provided:

[image]

ysiegel29, 2020-05-16:
thanks!

but what does "We provide a PyPI distribution which is created from this folder by running python setup.py bdist_wheel." mean? I need to clone the repository to do that? or can I run it from terminal?

Thanks again!

MartyG-RealSense, 2020-05-16:
This is again speculation, but at the base of the file list on the front page of the Python wrapper there is a file called **setup.py**.  It sounds to me that it is just a technical explanation of how the wheel package files on the Pyrealsense2 PyPi download page are generated, rather than a step that the user has to carry out themselves.

[image]

ysiegel29, 2020-05-16:
Thanks again. The setup.py (I managed to execute it) import the pyrealsense2... so it failed. Must be something else. Really painful to loose so much time with these steps...

MartyG-RealSense, 2020-05-16:
The references that I checked suggested that simply inputting **import pyrealsense2 as rs** should be enough to check if the installation is okay, and that if there are not errors then assume that the installation is fine.

ysiegel29, 2020-05-16:
I think there something wrong with where the built library are located

MartyG-RealSense, 2020-05-16:
You used this guide to install with CMake, yes?

https://github.com/IntelRealSense/librealsense/issues/5275#issuecomment-555830996

ysiegel29, 2020-05-16:
Yes

MartyG-RealSense, 2020-05-16:
1.  After having done **sudo make install** in the build directory, did you change directory:

**cd wrapper & cd python**

2.  Once you had changed directory to the **python** folder, do you see some **.so** files?

3.  If you do, copy these files and paste them into the root directory of your project where you run the import instruction

4.  Run the import instruction.

**import pyrealsense2**

ysiegel29, 2020-05-17:
yes did that, my project folder has the following files:
[image]

also tried " alternative to steps 6 and 7, `export PYTHONPATH=$PYTHONPATH:/usr/local/lib"`

May be my build is not working so I tried again using the steps in https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python, ie `python setup.py bdist_wheel`

but this setup script requires pyrealsense2... so I obviously get "ImportError: No module named pyrealsense2". 

fyi I can access the viewer from terminal.

Can someone who managed to make it work help?

ysiegel29, 2020-05-17:
fyi also tried to redo everything using this steps:
https://github.com/GitwellAnyohub/Real_Sensible_Moseq/blob/master/bash_and_install/install_realsense_sdk_mac_os.sh

but get permission issue when doing `make install` msg is:
```
CMake Error at cmake_install.cmake:64 (file):
  file INSTALL cannot set permissions on "/usr/local/include/librealsense2":
  Operation not permitted.
```

MartyG-RealSense, 2020-05-17:
I went through the entire process again from the beginning on my Windows PC for using **pip install pyrealsense2** to try to get Pyrealsense2 installed for the first time by that method and learn some lessons that I could pass onto you.  I managed to fill in the gaps in the instructions through experimentation and write my own little manual for the installation process.  

The first lesson was that the **pip** program has to be manually installed on your computer in order to be able to use pip install with Python.  The most popular way to do so is to download to your computer a script file called **get-pip.py**.  There are a range of sources for this file, such as the link below:

https://pip.pypa.io/en/stable/installing/#installing-with-get-pip-py

Find a **get-pip.py** link on the browser page, right-click on it and select the Save Target As, or Save Link As, or whatever menu option there is for saving a link.  Save it to a location on your computer.

[image]

An alternative way to get the file on MacOS is apparently to use this command in the Terminal:

**curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py**

At this point, the installation instructions for Windows and Mac may differ, but I will try to provide guidance.

On MacOS in the terminal, run this command to install pip:

**python get-pip.py**

For Windows the process is a bit more complicated, so if you successfully get pip installed as easily as that on MacOS then I am pleased for you   :)

At this point you could try entering **pip install pyrealsense2** into the Terminal to see whether it responds and installs pyrealsense.  

On Windows I found that for the command to work once pip was successfully installed, I had to actually change directory to the folder where pip was installed.  On my Python 3.7 setup, this was **Python > Python37 > Scripts**.  

I changed to this folder in my Windows Command Prompt (the equivalent of MacOS' Terminal, I guess) and input **pip** first to test that pip was working by causing it to generate a long list of help instructions.  then I input **pip install pyrealsense2** whilst in the **Scripts** folder and pyrealsense2 installed.

I hope some of the advice above is helpful to you.

ysiegel29, 2020-05-17:
Thanks! unfortunately the issue is not pip for me, I have it and use it regularly however the pip install pyrealsense2 does not work even when launch from the directory mentioned in the guide.

MartyG-RealSense, 2020-05-17:
If you are still experiencing the error "**Could not find a version that satisfies the requirement pyrealsense2**", this suggests that pip cannot find a Python version compatible with the Pyrealsense2 package on your MacOS computer when you launch the **pip install pyrealsense2** from the same directory that **pip** is installed in.   

It installs successfully with pip on Python 3.7 with Windows.  I'll check with Dorodnic about the earlier statement about MacOS pip compatibility.  

@dorodnic  When you mentioned in https://github.com/IntelRealSense/librealsense/issues/6246#issuecomment-614695052 that pip install pyrealsense2 should work on Python 3.7 in the discussion about pip install on Mac, can you confirm whether this includes MacOS please?

MartyG-RealSense, 2020-05-30:
Dorodnic provided further details that if a computer has an Arm type processor then a suitable pip package is not available and pyrealsense2 should be compiled from source code instead.

MartyG-RealSense, 2020-06-05:
@ysiegel29   Do you require further assistance with this case please, or can it be closed?   Thanks!

cxbrooks, 2020-06-12:
Here's what worked for me under macOS 10.14.4 (Mojave) with /usr/bin/python 2.7.10: 

```
git clone https://github.com/IntelRealSense/librealsense
brew install cmake libusb pkg-config
cd librealsense
mkdir build
cd build
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_PYTHON_BINDINGS=true -DPYTHON_EXECUTABLE=/System/Library/Frameworks/Python.framework/Versions/2.7/bin/python -DCMAKE_MACOSX_RPATH=ON -DCMAKE_INSTALL_RPATH="/usr/local/lib" -G "Unix Makefiles"
rm -f *.dylib wrappers/python/*.so
make install
cd ../wrappers/python/
cp /usr/local/lib/*realsense* pyrealsense2/
python find_librs_version.py /usr/local/ pyrealsense2
sudo python setup.py install 
```

I have some notes at https://wiki.eecs.berkeley.edu/ptexternal/Main/IntelRealSense#Mac_Python_2.x

For Python3.x, I used Homebrew to install Python 3.8 and then did:

```
export PATH=/usr/local/opt/python@3.8/bin:$PATH
git clone https://github.com/IntelRealSense/librealsense
brew install cmake libusb pkg-config
cd librealsense
mkdir build
cd build
mkdir -p /usr/local/realsense/python/3.8.3
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_PYTHON_BINDINGS=true -DPYTHON_EXECUTABLE=/usr/local/Cellar/python@3.8/3.8.3/Frameworks/Python.framework/Versions/3.8/bin/python3 -DCMAKE_MACOSX_RPATH=ON -DCMAKE_INSTALL_RPATH="/usr/local/realsense/python/3.8.3" -G "Unix Makefiles"
make install
cd ../wrappers/python/
cp /usr/local/lib/*realsense* pyrealsense2/
python find_librs_version.py /usr/local/ pyrealsense2
sudo /usr/local/opt/python@3.8/bin/python setup.py install
export PYTHONPATH=$PYTHONPATH:/usr/local/lib
```

Note that for Python3.x I had to set PYTHONPATH, which I did not need to do for Python2.x

I'm a little unsure if it is necessary to set CMAKE_INSTALL_RPATH in the cmake command.

MartyG-RealSense, 2020-06-12:
@cxbrooks    Thanks very much for sharing your method!

@ysiegel29   Does the method provided by @cxbrooks solve your problem, please?   Thanks!

ysiegel29, 2020-06-14:
Sorry for not replying earlier, I have managed to make it work few days ago but to be honest I can't really tell what made it work... I have tried so many things.
I should have documented this better... Thanks a lot for your help.

MartyG-RealSense, 2020-06-14:
@ysiegel29   Thanks so much for the update - I'm pleased you got it working!

MartyG-RealSense, 2020-07-05:
Case closed due to no further comments received and an official feature request for pyrealsense2 on Python 3.8 having been filed.

whubaichuan, 2020-09-10:
@MartyG-RealSense Hi, I got the error information: 'common/fw/target.h' file not found when make. And I encounter the "No internet connection, disabling IMPORT_DEPTH_CAM_FW" when cmake. How to solve it?  Thanks.

MartyG-RealSense, 2020-09-10:
Hi @whubaichuan   Another RealSense user who experienced the **target.h** error when building the Python wrapper found that the solution that worked for them was to add the CMake flag **-DBUILD_WITH_TM2=false**

Dorodnic the RealSense SDK Manager also advises in the same case to check network proxy settings if you get the internet connection error. 

https://github.com/IntelRealSense/librealsense/issues/6346

whubaichuan, 2020-09-11:
@MartyG-RealSense Install it in MacOS is not the best choice. I use windows and make it using pip install.

MartyG-RealSense, 2020-09-11:
@whubaichuan   Installation via PC is indeed the ideal situation due to the ease of pip and the lack of pip packages for non-x86 / x64 processors.

MartyG-RealSense, 2020-12-04:
Adding a note to keep this case open for a further time period.

MartyG-RealSense, 2021-03-28:
Hi everyone,

pyrealsense2 wheel files for the **pip install pyrealsense2** installation method have now been added for Python 3,8 and 3.9.  As this Feature Request has been completed, this case will now be closed.  Thanks so much for your patience!

https://pypi.org/project/pyrealsense2/#files

closed 2021-03-28 by MartyG-RealSense