realsenseai/librealsense #6403: Segmentation fault (core dumped)

fsc1023, 2020-05-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 16.04LTS) }  | 
| Kernel Version (Linux Only)    |  (4.15.0-99-generic) | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.0}                          | 
| Language                            |  {C++/Opencv/cmake }                          | 
| Segment			|  {Robot }                          | 

### Issue Description 
--------------------------------------------------------
Fistly, I have to say sorry for my English.
The problem :
When I run the executable file: rs-dnn,  the "display image" just flash a moment, then error: Segmentation fault (core dumped) . 

[ INFO:0] Initialize OpenCL runtime...
Segmentation fault (core dumped)

---------------------------------------------------
But,there is no error appear on another PC which I had copied to.

The (gdb) bt : 
#0  0x00007f7ac6076de3 in ?? () from /usr/lib/x86_64-linux-gnu/libLLVM-3.7.so.1
#1  0x00007f7b2d5be6ca in call_init (l=<optimized out>, argc=argc@entry=1, 
    argv=argv@entry=0x7fff8e25d088, env=env@entry=0x7fff8e25d098)
    at dl-init.c:72
#2  0x00007f7b2d5be7db in call_init (env=0x7fff8e25d098, argv=0x7fff8e25d088, 
    argc=1, l=<optimized out>) at dl-init.c:30
#3  _dl_init (main_map=main_map@entry=0x1dfced0, argc=1, argv=0x7fff8e25d088, 
    env=0x7fff8e25d098) at dl-init.c:120
#4  0x00007f7b2d5c38f2 in dl_open_worker (a=a@entry=0x7fff8e25b0f0)
    at dl-open.c:575
#5  0x00007f7b2d5be574 in _dl_catch_error (
    objname=objname@entry=0x7fff8e25b0e0, 
    errstring=errstring@entry=0x7fff8e25b0e8, 
    mallocedp=mallocedp@entry=0x7fff8e25b0df, 
    operate=operate@entry=0x7f7b2d5c34e0 <dl_open_worker>, 
    args=args@entry=0x7fff8e25b0f0) at dl-error.c:187
#6  0x00007f7b2d5c2db9 in _dl_open (
    file=0x7f7adc3aa3d0 "/usr/lib/x86_64-linux-gnu/beignet//libgbe.so", 
    mode=-2147483647, caller_dlopen=0x7f7adc37e5d8, nsid=-2, 
    argc=<optimized out>, argv=<optimized out>, env=0x7fff8e25d098)
    at dl-open.c:660
#7  0x00007f7b2423ef09 in dlopen_doit (a=a@entry=0x7fff8e25b320) at dlopen.c:66
#8  0x00007f7b2d5be574 in _dl_catch_error (objname=0x1da33b0, 
---Type <return> to continue, or q <return> to quit---
    8, mallocedp=0x1da33a8, operate=0x7f7b2423eeb0 <dlopen_doit>, args=0x7fff8e25b320) at dl-error.c:187
#9  0x00007f7b2423f571 in _dlerror_run (operate=operate@entry=0x7f7b2423eeb0 <dlopen_doit>, args=args@entry=0x7fff8e25b320) at dlerror.c:163
#10 0x00007f7b2423efa1 in __dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:87
#11 0x00007f7adc37e5d8 in ?? () from /usr/lib/x86_64-linux-gnu/beignet//libcl.so
#12 0x00007f7b2d5be6ca in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fff8e25d088, env=env@entry=0x7fff8e25d098) at dl-init.c:72
#13 0x00007f7b2d5be7db in call_init (env=0x7fff8e25d098, argv=0x7fff8e25d088, argc=1, l=<optimized out>) at dl-init.c:30
#14 _dl_init (main_map=main_map@entry=0x1e11640, argc=1, argv=0x7fff8e25d088, env=0x7fff8e25d098) at dl-init.c:120
#15 0x00007f7b2d5c38f2 in dl_open_worker (a=a@entry=0x7fff8e25b630) at dl-open.c:575
#16 0x00007f7b2d5be574 in _dl_catch_error (objname=objname@entry=0x7fff8e25b620, errstring=errstring@entry=0x7fff8e25b628, mallocedp=mallocedp@entry=0x7fff8e25b61f, 
    operate=operate@entry=0x7f7b2d5c34e0 <dl_open_worker>, args=args@entry=0x7fff8e25b630) at dl-error.c:187
#17 0x00007f7b2d5c2db9 in _dl_open (file=0x20b9490 "/usr/lib/x86_64-linux-gnu/beignet//libcl.so", mode=-2147483647, caller_dlopen=0x7f7adc5ea2fb, nsid=-2, 
    argc=<optimized out>, argv=<optimized out>, env=0x7fff8e25d098) at dl-open.c:660
#18 0x00007f7b2423ef09 in dlopen_doit (a=a@entry=0x7fff8e25b860) at dlopen.c:66
#19 0x00007f7b2d5be574 in _dl_catch_error (objname=0x1da33b0, errstring=0x1da33b8, mallocedp=0x1da33a8, operate=0x7f7b2423eeb0 <dlopen_doit>, args=0x7fff8e25b860)
    at dl-error.c:187
#20 0x00007f7b2423f571 in _dlerror_run (operate=operate@entry=0x7f7b2423eeb0 <dlopen_doit>, args=args@entry=0x7fff8e25b860) at dlerror.c:163
#21 0x00007f7b2423efa1 in __dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:87
#22 0x00007f7adc5ea2fb in ?? () from /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
#23 0x00007f7adc5ea459 in ?? () from /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
#24 0x00007f7adc5ea9c2 in ?? () from /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
#25 0x00007f7adc5eb233 in clGetPlatformIDs () from /usr/lib/x86_64-linux-gnu/libOpenCL.so.1
#26 0x00007f7b26d0c577 in ?? () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3
#27 0x00007f7b26d64a72 in cv::ocl::haveOpenCL() () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3
#28 0x00007f7b26d802c8 in cv::ocl::useOpenCL() () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3
#29 0x00007f7b26dcb309 in cv::UMat::getStdAllocator() () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3
#30 0x00007f7b26dccb66 in cv::Mat::getUMat(int, cv::UMatUsageFlags) const () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3
#31 0x00007f7b2a815924 in cv::dnn::experimental_dnn_v2::ConvolutionLayer::create(cv::dnn::experimental_dnn_v2::LayerParams const&) ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3
#32 0x00007f7b2a77de10 in ?? () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3
#33 0x00007f7b2a75cce2 in cv::dnn::experimental_dnn_v2::LayerFactory::createLayerInstance(cv::String const&, cv::dnn::experimental_dnn_v2::LayerParams&) ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3
#34 0x00007f7b2a75ce85 in ?? () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3
#35 0x00007f7b2a76bbf7 in ?? () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3
#36 0x00007f7b2a77a9a0 in ?? () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3
#37 0x00007f7b2a77afb8 in ?? () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3
#38 0x00007f7b2a77b299 in cv::dnn::experimental_dnn_v2::Net::forward(cv::String const&) () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3
#39 0x000000000041d551 in main ()

Please help me. Thank you very much!

MartyG-RealSense, 2020-05-16:
This can be a very hard error to diagnose unfortunately, as it typically just means that the computer tried to access an area of its memory storage that it was not allowed to.  It can apparently suggest that there may be a broken module on your Ubuntu installation (hence why it may work on one Ubuntu PC but not another).

You could either do a fresh installation of Ubuntu to see if it corrects the problem, or try to repair the broken files with a guide like the one in the link below.

https://blog.opstree.com/2019/04/02/resolving-segmentation-fault-core-dumped-in-ubuntu/

fsc1023, 2020-05-16:
@MartyG-RealSense Thank you very much!
I have ran the debgu again, error occured when  it comes to the setp 80:

80	        Mat detection = net.forward("detection_out"); //compute output
(gdb) 
[ INFO:0] Initialize OpenCL runtime...

Thread 1 "rs-dnn" received signal SIGSEGV, Segmentation fault.
0x00007fff86076de3 in ?? ()
   from /usr/lib/x86_64-linux-gnu/libLLVM-3.7.so.1

I had tried the first method in the link you posed， unfortunately, it is not work also.
ask for help again, thank you!

MartyG-RealSense, 2020-05-16:
I think that all I can do is link the rs-dnn example program's creator @maloel to this discussion to see if they can suggest a way to fix the SIGSEV error.

Until then, I wonder if you could try the other OpenCV dnn sample program in the SDK and see if that works for you.

https://github.com/twMr7/rscvdnn

ev-mp, 2020-05-17:
@fsc1023 , the error indicates an issue within OpenCl 3rd party library:
>#37 0x00007f7b2a77afb8 in ?? () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3
#38 0x00007f7b2a77b299 in cv::dnn::experimental_dnn_v2::Net::forward(cv::String const&) () from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3

The call stack display is interrupted short of reaching the LRS code, so the only place to check in Librealsense is the loading of DNN models - please verify that those models are present and the files are not corrupted (e.g. zero size).

We'll keep the thread open for sometime to allow  for community members to respond and provide  insights w.r.t. OpenCv.

fsc1023, 2020-05-17:
@MartyG-RealSense  Thank  you very much.  the code (https://github.com/twMr7/rscvdnn ) rscdnn have no CMakeLists.txt, I have not find a solution to compile and run it, cmake or g++?

fsc1023, 2020-05-17:
@ev-mp Thank you, I reviewed the files MobileNetSSD_deploy.caffemodel (23.1M) and MobileNetSSD_deploy.prototxt (29.4kB), both of them have no problem. Maybe I have to reinstall the Ubuntu OS, but it is annoying.

fsc1023, 2020-05-20:
Thank you! @ev-mp @MartyG-RealSense I have solved the error. Maybe it was caused by the environment of my PC. I reinstalled the OpenCL, then, the error disappeared magically.

MartyG-RealSense, 2020-05-20:
@fsc1023   That's great news - thanks so much for the update   :)

MartyG-RealSense, 2020-05-30:
Case closed due to the solving of the problem and no further comments received.

closed 2020-05-30 by MartyG-RealSense