realsenseai/librealsense #6406: RsCamera D435i cannot be recognizable on Samsung A51.

hunterman1, 2020-05-17:
RsCamera D435i cannot be recognizable on Samsung A51.

After connecting it via cable it cannot be visibible under DeviceList
Example: Capture ->MainActivity.java
try(DeviceList dl = mRsContext.queryDevices(

This camera also don't work on RsCamera application available on Google Play store: https://play.google.com/store/apps/details?id=com.intel.realsense.camera&pageId=none

On Samsung Galaxy J7 it works.

For more info you can contact me directly via email: rrobert.kawecki@gmail.com (double R in the beginning)
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D434i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Android studio  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {Android }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-05-17:
Apparently the Samsung A51 runs Android 10.  This has been known to cause some problems with the RealSense Android wrapper.  A fix that has been successful for Android 10 users is to set **targetSdkVersion** to use an earlier target version, such as targetSdkVersion 27.

RealSenseCustomerSupport, 2020-05-18:
----------------------------------------------

please ensure you use USB OTG cable

hunterman1, 2020-05-22:
Hello, Downgrading to targetSdkVersion 27 version solved the problem. Still I'm facing issues with camera. This time I'm observing that the view from the camera is stuttering. This could be observed on Android 9(Samsung Galaxy A50) and on Android 10(Samsung Galaxy A51). I'm using examples from the github source code..

I tried to install also application from Google Play store(https://play.google.com/store/apps/details?id=com.intel.realsense.camera&pageId=none). On Samsung Galaxy A50 everything works as expected - view from the camera is good, without any stutters or delays.

Is it possible to get source code for this application available on your Google Play store? Maybe some changes were implemented there which allows to run this application without problems.

King regards,
Robert

MartyG-RealSense, 2020-05-22:
There is a personal email contact for the app developer on the bottom of the store listing.  That may be the best way to get in contact with them to ask them about the app.

[image]

I'll also tag the developer by their GitHub name in case they see it.  @matkatz

hunterman1, 2020-05-22:
I've just wrote an email on this email address. Nevertheless, the problem with Android example code (Capture example) still exists. In my opinion it stutters, both on A51 and A50 smartphones(Android 10 and Android 9).
Below I've attached an example Android project that I implemented directly basing on your example. Could you please try to check what is wrong there?

In my opinion something stutters the image from the camera- both just from camera view, and depth image view.

MartyG-RealSense, 2020-05-22:
I do not currently have an Android RealSense development environment set up to test your example project myself, unfortunately.  Hopefully one of the RealSense developer team members will be able to provide guidance on your https://github.com/IntelRealSense/librealsense/issues/6442 post (I am on the Support team).

hunterman1, 2020-05-24:
Ok, thanks for your support and being interested in monitoring #6442 ticket. I hope that registered issue will be found and fixed.
Kind regards,
Robert

MartyG-RealSense, 2020-05-24:
@hunterman1   Thanks very much.   As you have contacted the Android developer by email, I will close this case so that the problem can be handled through https://github.com/IntelRealSense/librealsense/issues/6442 - thanks very much!

closed 2020-05-24 by MartyG-RealSense