realsenseai/librealsense #6407: realse sense depth camera d435 not detecting

jamesjamal, 2020-05-17:
| | |
|---|---|
|**librealsense**|2.34.0 RELEASE|
|**OS**|Linux|

 17/05 17:06:10,543 INFO [139799137572608] (backend-v4l2.cpp:454) Cannot access /sys/class/video4linux

ev-mp, 2020-05-17:
@jamesjamal, the error indicates that the Linux UVC driver was not loaded.
Kernel logs is required to further analyze this, but in the meanwhile - please specify the OS/Distro type/ Kernel version and whether you've tried to apply kernel patches/DKMS.

MartyG-RealSense, 2020-05-17:
Thanks @ev-mp !

@jamesjamal  In addition to the excellent advice above, you could also check whether your video module is working by using this command:

**lsmod | grep video**

jamesjamal, 2020-05-17:
[image]
@ev-mp @MartyG-RealSense 
I am using ubuntu 18 with kernel 5*  Ros(melodic)

T265 working fine with ubuntu machine and ROS and windows
D435 device not working with  in both Ros and Realsense-viewer in ubuntu and with windows works fine.
I need to set it up in ROS. in ubuntu.

jamesjamal, 2020-05-17:
[image]

MartyG-RealSense, 2020-05-17:
If you have both a T265 and a D435 plugged in at the same time, it could be a conflict between the two as this is a well documented phenomenon with the RealSense ROS wrapper.  If you have both cameras attached, could you try unplugging the T265 and see if the D435 is detected after roslaunch?

jamesjamal, 2020-05-17:
No i checked once a while.

MartyG-RealSense, 2020-05-17:
Is the D435 able to be detected in other programs such as the RealSense Viewer, rs_enumerate_devices or SDK sample programs, or is it just in ROS that it is not detected?

jamesjamal, 2020-05-17:
@MartyG-RealSense 
i can able to connect d435 in windows it works properly.
in Ubuntu my d435 its not working in both Ros and Realsense_viewer
But my T265 is working fine with both in ubuntu
is this related to UVC driver ?

MartyG-RealSense, 2020-05-17:
The UVC driver remains the likely cause.

Could you tell me please which kernel version you are using.  You listed it as 5*.  Librealsense supports kernel versions 5 and 5.3.

jamesjamal, 2020-05-17:
[image]

MartyG-RealSense, 2020-05-17:
Thank you.   And have you applied kernel patches as suggested by @ev-mp ?

MartyG-RealSense, 2020-05-30:
Do you require further assistance please, or can this case be closed?   Thanks!

jamesjamal, 2020-05-30:
Thanks for the guidance I downgraded to 4.6 kernel in my Ubuntu machine it works perfectly.

MartyG-RealSense, 2020-05-30:
Thanks for the update!  I'm very pleased that you found a solution.   I will close the case now.

closed 2020-05-30 by MartyG-RealSense