realsenseai/librealsense #6408: no device connected

zhengzhe97, 2020-05-17:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | v2.32.1| 
| Operating System & Version |   Ubuntu 18.04.3 LTS  | 
| Kernel Version (Linux Only)    |  4.9.14                                         | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            | 2.32.1                           | 
| Language                            |  C ++                        | 

### Issue Description
I install sdk on jetson nano following [](https://www.jetsonhacks.com/2019/05/16/jetson-nano-realsense-depth-camera/). I can run realsense viewer, and it also can detect my realsense D435i and output the image. In order to test on C++, I write a C++ program to show the color image and the demo doesn't any problem. when I try to use the same code in my project and compile it by cmake, 'no device connected' occurs.

MartyG-RealSense, 2020-05-17:
It sounds as though you are trying to build a RealSense project of your own that is similar to the existing **rs-capture** sample program.  Except that instead of showing both a color and deoth stream, you only want to display the color stream.  Is that correct, please?

https://github.com/IntelRealSense/librealsense/tree/master/examples/capture

If the problem is that your project does not have linkages to the RealSense files that govern the camera, the information in the link below about connecting your project to those files with CMake may be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/4277#issuecomment-505242118

You may also like to look at a simple Getting Started project template that can be used as a basis for learning how to link projects to the Librealsense library.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

zhengzhe97, 2020-05-18:
Yes, I just want to display the color stream. I try to use the simple Getting Started project, it still returns no device connected. 
```
RealSense error calling re2_pipeline_start(pipe:0x557a8eac70):
No device connected
```
I write a demo by QT, just CMakeLists.txt and main.cpp. It can run successfully on QT, but if I use cmake it , it returns 'no device connneted'.
CMakeLists.txt
```
cmake_minimum_required(VERSION 3.10)
project(untitled)

set(OpenCV_DIR "/home/zhengzhe/Desktop/opencv/opencv-3.4.8/release")
find_package(OpenCV REQUIRED)
find_package(realsense2 REQUIRED)
add_executable(${PROJECT_NAME} "main.cpp")

target_link_libraries(${PROJECT_NAME} ${OpenCV_LIBS})
target_link_libraries(${PROJECT_NAME} ${realsense2_LIBRARY})
```
main.cpp
```
#include <iostream>
#include<vector>
#include<opencv2/opencv.hpp>
#include<librealsense2/rs.hpp>

using namespace std;

static const int INPUT_H = 678;
static const int INPUT_W = 1024;
int main()
{

    rs2::pipeline p;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 960, 540, RS2_FORMAT_BGR8, 30);
    p.start(cfg);

    while (true)
    {
        rs2::frameset frames = p.wait_for_frames();
        rs2::video_frame img = frames.get_color_frame();
        cv::Mat image(cv::Size(960, 540), CV_8UC3, (void*)img.get_data(), cv::Mat::AUTO_STEP);
        cv::Mat resizedImage = cv::Mat::zeros(INPUT_H, INPUT_W, CV_32FC3);
        cv::resize(image, resizedImage, cv::Size(INPUT_W, INPUT_H));
        cv::imshow("image",resizedImage);
        cv::waitKey(1);
    }
}

```

duongbaoquy, 2020-05-18:
This is my code to show color_stream - maybe it will help you find out something
`
#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include <opencv2/opencv.hpp>   // Include OpenCV API

int main(int argc, char * argv[]) try
{
    rs2::colorizer color_map;
    rs2::pipeline pipe;
    pipe.start();

    using namespace cv;
    const auto window_name = "Display Image";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
    {
        rs2::frameset frameset = pipe.wait_for_frames();
        rs2::frame color = frameset.get_color_frame().apply_filter(color_map);

        const int w = color.as<rs2::video_frame>().get_width();
        const int h = color.as<rs2::video_frame>().get_height();

        // Create OpenCV matrix of size (w,h) from the colorized depth data
        Mat image(Size(w, h), CV_8UC3, (void*)color.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        Mat img2;
        cvtColor(image, img2, cv::COLOR_RGB2BGR);
        imshow(window_name, img2);
    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
`

zhengzhe97, 2020-05-18:
@duongbaoquy 
Thanks for your code, but it couldn't solve my problem.

duongbaoquy, 2020-05-18:
Can you provide more error images, I think your problem is not cmakelist

zhengzhe97, 2020-05-18:
The following image is the same code on QT and compiles by CMake.
on QT it can detect my camera and display the color stream.
[image]
compiles by CMake it throws 'no device connected'
[image]
realsense_viewer
[image]

MartyG-RealSense, 2020-05-18:
@duongbaoquy   Thanks so much for the help that you are kindly providing for @zhengzhe97 - it's great when the community helps each other!

zhengzhe97, 2020-05-18:
@MartyG-RealSense Is there a way to solve this problem. I am weird about the situation.

MartyG-RealSense, 2020-05-18:
@zhengzhe97   Have you added the Librealsense include path and library to your project, please?

For .pro project file:
http://doc.qt.io/qt-5/qmake-project-files.html

zhengzhe97, 2020-05-18:
I am sure the path and library in CMakeList. I use cmake to compile the project instead of qmake, so it just a C++ project instead of a QT project. I compile the project by the button('Run CMake' and 'Run' on QT menu, the project display the color stream.
I compile the same project in terminal use 
```
mkdir build
cd build
cmake ../
make
```
it returns 'no device connected'

MartyG-RealSense, 2020-05-18:
Please compare the steps you are taking to the SDK's beginner "Getting Started" project for building a basic project with CMake, as it likely has a similar structure to your own project.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

MartyG-RealSense, 2020-05-30:
Do you require further assistance please, or can this case be closed?   Thanks!

duongbaoquy, 2020-05-30:
I dont know if this can help you. I got same error with board ultra96v2, i dont know why but when i changed the camera from D435i to D415, its work for me.

MartyG-RealSense, 2020-06-05:
@zhengzhe97   Do you still require assistance with this case please, or can the case be closed?  Thanks!

closed 2020-06-05 by zhengzhe97

MartyG-RealSense, 2020-06-05:
@zhengzhe97   Thanks very much!