realsenseai/librealsense #6409: Feature request: offer an option to download non-shifted IR image even Disparity Shift > 0 on D400

robertxwu, 2020-05-17:
As stated in "Best-Known-Methods for Tuning Intel® RealSense™D400 Depth Cameras for Best Performance", one approach to make D400 series camera works in a close range is to increase Disparity Shift. For instance, setting Disparity Shift to 145 works well for MinZ = 180mm. 

However, as discussed in the following thread, when Disparity Shift > 0, the right IR images are shifted to right _S_ pixels, and _S_ = Disparity Shift.
https://github.com/IntelRealSense/librealsense/issues/6311#issuecomment-629846762

It will be very useful if an option is provided that non-shifted IR can be retrieved even Disparity Shift > 0.

Thanks in advice!

MartyG-RealSense, 2020-05-17:
Thanks so much for your Feature Request.  I have labelled it so that the RealSense team can officially track it.

MartyG-RealSense, 2020-07-14:
Hi @robertxwu   Thank you very much for your patience regarding your feature request for an option to download a non-shifted IR image.

Intel carefully considered the proposal and decided not to proceed with it.  The reason given was that if you are looking at the right image and seeing a shift, this is exactly what should be happening.  It was recommended that you should apply your own shift of the right image if you want to, or alternatively you can unshift and grab images.

As the Change Request process has concluded, this case will now be closed.  Feel free to add a comment below though, please.

closed 2020-07-14 by MartyG-RealSense