realsenseai/librealsense #6410: Windows T265 Detection Issues

sidxrs, 2020-05-18:
I have a T265. I am running v2.34.0 on Windows 10 (PC). I cannot run any of the samples, as it says it cannot detect the device, even though it is plugged into a USB 3.0 Port. When I open device manager, it shows it as "Intel(R) RealSense(TM) Tracking Camera T265", but under USB Devices,  and not imaging devices. I have tried reinstalling everything, but I cannot get it to work. What do I do?

rafaelspring, 2020-05-19:
@sidxrs Does the T265 show up in the Realsense Viewer app?

sidxrs, 2020-05-19:
It does show up in the app.

jakemin, 2020-05-27:
The first time you connect the T265 to your system, Device Manager will recognize it as a Movidius MA2X5X on a universal serial bus device.
Then launch RealSense Viewer and it will be recognized again as T265 by Universal Serial Bus device in Device Manager.
When another camera is connected, it may not appear in the RealSense Viewer.
You can add T265 by clicking "Add Source" on the top left.
[image]
[image]

closed 2021-01-25 by RealSenseSupport