realsenseai/librealsense #6412: D415 did not work on latest Jetson Xavier NX

7oud, 2020-05-18:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (05.12.03.00) | 
| Operating System & Version |   { Ubuntu 18.04 for Tegra  | 
| Kernel Version (Linux Only)    |  (Linux nx 4.9.140-tegra)                                         | 
| Platform                                 | NVIDIA Jetson Xavier NX, JetPack 4.4 DP  |
| SDK Version                            |  { 2.<32>.<1> }                          | 

### Issue Description
I installed realsense SDK on the latest NVIDIA Jetson Xavier NX, as [installation_jetson.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md)

MartyG-RealSense, 2020-05-18:
Hi @7oud   The Xavier NX is new enough that at this point, the only guides available that I know of are for the Xavier AGX model.

The **JetsonHacks website has a guide for doing a simple install with a RealSense Debian repository that is compatible with Jetson AGX Xavier and does not involve patching of modules and kernels.

https://github.com/jetsonhacks/installRealSenseSDK

It may be worth trying this with your NX, as because it is a simple process, you will not have lost much time if it does not work for NX.

JetSonHacks also have an earlier AGX Xavier installation guide involving patching that is more complex.

https://github.com/jetsonhacks/buildLibrealsense2Xavier

7oud, 2020-05-19:
@MartyG-RealSense Thanks!  The installation from Debian in **JetsonHacks** repo is the same as here. So only two methods can be tried, building from source and earlier AGX Xavier installation guide

MartyG-RealSense, 2020-05-19:
Thank you very much @7oud  - please let me know whether either of these two methods works for you.  Good luck!

7oud, 2020-05-19:
@MartyG-RealSense I have test the 1st method, just build the latest librealsense of master branch, as [installRealSenseSDK](https://github.com/jetsonhacks/installRealSenseSDK) described. However it didn't work. Most of time realsense view could not show the images.
When plug out then plug in D415, viewer will show the image, if stop the stream and reopen it, "No frame received". Even the viewer show images after the 1st time plug in the camera, some error info was given "get_device_time_ms() took too long (more then 2 mSecs)"

MartyG-RealSense, 2020-05-19:
If you have built the latest Librealsense branch (currently 2.34.0) then the error **get_device_time_ms() took too long (more then 2 mSecs)** is not likely related to the NX.  This is a problem that has been occurring in the RealSense Viewer on Windows, Ubuntu and MacOS with various users' installation of 2.34.0, including my own.  So for now I would recommend disregarding that issue as it likely has no relation to your problems with NX specifically.

The error "double free or corruption (out)" that you experienced when closing the Viewer is apparently caused during freeing up memory, which makes sense as applications try to free up memory for re-use when they are being closed.

This situation has characteristics of a case from November 2019 that involved Jetson.

https://github.com/IntelRealSense/librealsense/issues/5315

7oud, 2020-05-19:
@MartyG-RealSense yes, there are some similarities to #5315
1. When unplugin and plugin D415 camera, the viewer works, even though there are some error info **get_device_time_ms() took too long (more then 2 mSecs)**. And at this time, the video devices are connected.
```
crw-rw-rw-+ 1 root plugdev 81, 0 5月  19 17:16 video0
crw-rw-rw-+ 1 root plugdev 81, 1 5月  19 17:16 video1
crw-rw-rw-+ 1 root plugdev 81, 2 5月  19 17:16 video2
```
2. When stop the streaming and restart it, "No frames received" for both color and depth images. And at this time, video devices are disappeared.
```
ls: cannot access 'vid*': No such file or directory
```
3. my user account is in the following groups, including **video**
```
zxxx adm cdrom sudo audio dip video plugdev i2c lpadmin gdm sambashare weston-launch trusty crypto gpio
```

MartyG-RealSense, 2020-05-19:
When you stop streams and restart them, are you stopping and starting the same stream or alternating between the streams please (e.g depth on RGB off, or RGB on and depth off).  Sometimes the streams in the Viewer can get confused when this is done.  Here is an example case:

https://github.com/IntelRealSense/librealsense/issues/2057

Do you experience the "No frames received!" error if you start the Viewer and only toggle one stream (the same stream) on and off please?

7oud, 2020-05-19:
1. rgb off rgb on, rgb not work
2. depth off depth on, depth not work
once a stream is off, it cannot be reopen

MartyG-RealSense, 2020-05-19:
I wonder if the pipeline is remaining open (not destroyed) when the stream is stopped the first time, preventing the pipeline from being restarted (as its status to be that it is closed before it can be opened).  Unplugging and replugging the camera would reset it, and the pipeline along with it.

7oud, 2020-05-19:
Could you give some c++ code to verify your assumption, I will test it lately. And whether reopening an unclosed stream, will video devices shut down(gone)?

MartyG-RealSense, 2020-05-19:
Approaches to C++ scripting for checking if the pipeline is closed are discussed in the link below with script examples.

https://github.com/IntelRealSense/librealsense/issues/2240

ancorasir, 2020-05-20:
@7oud I have met exactly the same problem described above except I'm using D435i. I also use the https://github.com/jetsonhacks/installRealSenseSDK/buildLibrealsense.sh to install the latest librealsense. The realsense-viewer only works for the first time after the camera is connected. If I turn off the stream and turn on the same stream again, I got "No frame received".

7oud, 2020-05-20:
@MartyG-RealSense I tested [callback sample](https://github.com/IntelRealSense/librealsense/tree/master/examples/callback) with some log.
```
    rs2::log_to_console(RS2_LOG_SEVERITY_INFO);

    std::map<int, int> counters;
    std::map<int, std::string> stream_names;
    std::mutex mutex;

    auto callback = [&](const rs2::frame& frame)
    {
        std::cout << "come into callback... " << std::endl;

        std::lock_guard<std::mutex> lock(mutex);
        if (rs2::frameset fs = frame.as<rs2::frameset>())
        {
            for (const rs2::frame& f : fs)
            {
                counters[f.get_profile().unique_id()]++;
                std::cout << "counters = " << counters[f.get_profile().unique_id()] << std::endl;
            }
        }
    };
```
1. When plugin the camera at first time, video devices are listed.
```
zxxx@nx:/dev$ ls video* -al
crw-rw-rw-+ 1 root plugdev 81, 0 5月  20 13:31 video0
crw-rw-rw-+ 1 root plugdev 81, 1 5月  20 13:31 video1
crw-rw-rw-+ 1 root plugdev 81, 2 5月  20 13:31 video2

```
2. run the callback program, the callback function is invoked, even though there are some error info
```
 20/05 13:32:48,481 INFO [548514643808] (sensor.cpp:1340) Request: Z16 Depth, 
Resolved to: Z16 Depth, 
 20/05 13:32:48,553 INFO [548514643808] (uvc-streamer.cpp:28) endpoint 130 read buffer size: 1844224
 20/05 13:32:48,596 INFO [548514643808] (sensor.cpp:1340) Request: RGB8 Color, 
Resolved to: YUYV Color, 
 20/05 13:32:48,658 INFO [548514643808] (uvc-streamer.cpp:28) endpoint 132 read buffer size: 1844224
RealSense callback sample

 20/05 13:32:48,720 ERROR [548451037632] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:48,744 ERROR [548130419136] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:48,774 ERROR [548130419136] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:48,808 ERROR [548130419136] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:48,838 ERROR [548130419136] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:48,850 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/05 13:32:48,911 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/05 13:32:48,982 ERROR [548451037632] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:48,993 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 13:32:49,064 ERROR [548130419136] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:49,077 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 13:32:49,148 ERROR [548130419136] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:49,159 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 13:32:49,230 ERROR [548451037632] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:49,256 ERROR [548101546432] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:49,267 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 13:32:49,338 ERROR [548130419136] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:49,350 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 13:32:49,410 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 13:32:49,481 ERROR [548101546432] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:49,493 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 13:32:49,564 ERROR [548130419136] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:49,575 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 13:32:49,646 ERROR [548101546432] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:49,660 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 13:32:49,731 ERROR [548451037632] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:49,752 ERROR [548130419136] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:32:49,763 WARNING [548434252224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/05 13:32:49,834 ERROR [548101546432] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
[... 47 lines cut ...]
[... cut ...]
 20/05 13:46:44,984 ERROR [548326310336] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:46:45,106 ERROR [548326310336] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
 20/05 13:46:45,228 ERROR [548326310336] (types.h:307) get_device_time_ms() took too long (more then 2 mSecs)
```
So, as the pipeline is still running for the 1st time, the video devices are already disappeared(seen from another terminal), which leads to the program can not be restart.
Looking forward to fixing this bug, which is important to us

MartyG-RealSense, 2020-05-20:
@7oud  Thanks so much for running the test and generating information for the developers to be able to analyze!

ancorasir, 2020-05-20:
I can reproduce the issue with D435i. Video devices disappeared after the pipline is established for the first time.

MartyG-RealSense, 2020-05-20:
@ancorasir  As the issue is reproducible, I will label this discussion as "Bug" so that it is trackable by the RealSense developers.  Thanks for your report!

shaoxiang, 2020-05-22:
Required Info |  
-- | --
Camera Model | { D400 }
Firmware Version | (05.12.03.00)
Operating System & Version | { Ubuntu 18.04 for Tegra
Kernel Version (Linux Only) | (Linux nx 4.9.140-tegra)
Platform | NVIDIA Jetson  Nano, JetPack 4.4
SDK Version | { 2.34.1 }

I solved this problem by downloading the latest version of v2.34.1, which runs smoothly on the Jetson Nano.  @7oud  @ancorasir You can try.
If you have CUDA dev-kit installed, don't forget to add -DBUILD_WITH_CUDA=true for better performance. So my cmake is:
cmake \
    -DBUILD_EXAMPLES=true \
    -DFORCE_LIBUVC=true \
    -DBUILD_WITH_CUDA=true \
    ..

MartyG-RealSense, 2020-05-22:
@shaoxiang   I'm happy to hear that you had success with your Nano!

7oud, 2020-05-22:
@shaoxiang @MartyG-RealSense Yes, the libRS worked well on Nano from v2.23, I have tested it. However it didn't work on Xavier NX now

MartyG-RealSense, 2020-05-22:
@7oud  Yes, I know NX has a distinct process from that of Nano to work out.  We can be glad for @shaoxiang that their own problem is solved though.   *smiles*

7oud, 2020-05-22:
@shaoxiang so the problem you met on nano was fixed in v2.34.1? is it bcz of jetpack4.4 and what's the problem? when I used nano with d415, I found the conversion from yuyv to rgb with cuda is slower, conversion using cpu-based opencv is much faster

shaoxiang, 2020-05-25:
@7oud 
Yes, fixed by v2.34.1 and this version works well on the nano not found jam or delay at all.
When the version lower than v2.34.1, I have met many problems，like:
get_device_time_ms() took too long (more then 2 mSecs) and Serious frame loss! 
In Ros I use two cameras (d435i and d410) at the same time and get 30FPS Realtime.
[image]
[image]
[image]
I will continue to test to see if there are any other problems.

AndreV84, 2020-05-28:
I can get 435i running on Jetson NX
e.g. with realsense-viewer
However, after closing it will become unavailable until further reboot;
Moreover, It would throw 
```
rs-hello-realsense 
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x557f97e110):
    Frame didn't arrive within 15000
```
I wasn't able also to confirm anything but rs-distance and rs-capture; Trying to run it from ROS.
@shaoxiang Are you using Jetson nano or Jetson NX? With the latter Jetpack 4.4?
How do you install librealsense2 driver and examples?
 How do you construct ROS? ROS Dockerfile?

AndreV84, 2020-05-28:
when the latest librealsense will be added to apt-repository?
```
apt policy librealsense*
librealsense2-dbg:
  Installed: (none)
  Candidate: 2.32.1-5ubuntu5~1.gbp83f952
  Version table:
     2.32.1-5ubuntu5~1.gbp83f952 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.31.0-5ubuntu5~1.gbp95d195 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.30.0-5ubuntu5~1.gbp9f99fa 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.28.1-5ubuntu5~1.gbpc1c97e 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
librealsense2-dev:
  Installed: 2.32.1-5ubuntu5~1.gbp83f952
  Candidate: 2.32.1-5ubuntu5~1.gbp83f952
  Version table:
 *** 2.32.1-5ubuntu5~1.gbp83f952 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
        100 /var/lib/dpkg/status
     2.31.0-5ubuntu5~1.gbp95d195 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.30.0-5ubuntu5~1.gbp9f99fa 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.28.1-5ubuntu5~1.gbpc1c97e 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
librealsense2:
  Installed: 2.32.1-5ubuntu5~1.gbp83f952
  Candidate: 2.32.1-5ubuntu5~1.gbp83f952
  Version table:
 *** 2.32.1-5ubuntu5~1.gbp83f952 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
        100 /var/lib/dpkg/status
     2.31.0-5ubuntu5~1.gbp95d195 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.30.0-5ubuntu5~1.gbp9f99fa 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.28.1-5ubuntu5~1.gbpc1c97e 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
librealsense2-gl-dbg:
  Installed: (none)
  Candidate: 2.32.1-5ubuntu5~1.gbp83f952
  Version table:
     2.32.1-5ubuntu5~1.gbp83f952 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.31.0-5ubuntu5~1.gbp95d195 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.30.0-5ubuntu5~1.gbp9f99fa 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.28.1-5ubuntu5~1.gbpc1c97e 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
librealsense2-gl-dev:
  Installed: (none)
  Candidate: 2.32.1-5ubuntu5~1.gbp83f952
  Version table:
     2.32.1-5ubuntu5~1.gbp83f952 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.31.0-5ubuntu5~1.gbp95d195 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.30.0-5ubuntu5~1.gbp9f99fa 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.28.1-5ubuntu5~1.gbpc1c97e 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
librealsense2-gl:
  Installed: 2.32.1-5ubuntu5~1.gbp83f952
  Candidate: 2.32.1-5ubuntu5~1.gbp83f952
  Version table:
 *** 2.32.1-5ubuntu5~1.gbp83f952 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
        100 /var/lib/dpkg/status
     2.31.0-5ubuntu5~1.gbp95d195 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.30.0-5ubuntu5~1.gbp9f99fa 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.28.1-5ubuntu5~1.gbpc1c97e 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
librealsense2-utils:
  Installed: 2.32.1-5ubuntu5~1.gbp83f952
  Candidate: 2.32.1-5ubuntu5~1.gbp83f952
  Version table:
 *** 2.32.1-5ubuntu5~1.gbp83f952 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
        100 /var/lib/dpkg/status
     2.31.0-5ubuntu5~1.gbp95d195 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.30.0-5ubuntu5~1.gbp9f99fa 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.28.1-5ubuntu5~1.gbpc1c97e 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
librealsense2-udev-rules:
  Installed: 2.32.1-5ubuntu5~1.gbp83f952
  Candidate: 2.32.1-5ubuntu5~1.gbp83f952
  Version table:
 *** 2.32.1-5ubuntu5~1.gbp83f952 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
        100 /var/lib/dpkg/status
     2.31.0-5ubuntu5~1.gbp95d195 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.30.0-5ubuntu5~1.gbp9f99fa 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
     2.28.1-5ubuntu5~1.gbpc1c97e 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages
```

MasonLilly, 2020-06-04:
I'm experiencing similar issues as described throughout this thread. With a D435i, a Xavier NX, and version 2.32 (installed via apt-get), realsense-viewer is able to open the depth and color streams, but individually they stop working after being closed and re-opened, and after this the video devices have disappeared from /dev.

MartyG-RealSense, 2020-06-05:
@MasonLilly   If one stream type is stopped and another type is started and then that is stopped and the previous type was started, after several open-closes of different stream types (instead of doing stop-start on the same stream type), there is a chance that the SDK may become confused.

The latest 400 Series camera firmware **5.12.5** contains a fix for the problem of IR frames no longer arriving after several stop-starts.

MasonLilly, 2020-06-05:
I have this problem even if i toggle a single stream on and off. In my
case, it almost always occurs after the first on/off cycle.

I somewhat doubt that it's a firmware issue, since I've had no problem
running the same camera with the same firmware on Windows and non-Jetson
Ubuntu, but I'll try the different firmware soon and see.

MartyG-RealSense, 2020-06-05:
Please let us know the results of a firmware update, @MasonLilly and if it does not help then I can look at filing an official bug report, since a few people in this discussion seem to be having a similar problem.

MasonLilly, 2020-06-05:
I'll be darned, that seems to have worked! Since updating the firmware I haven't had the above issue, no matter how capriciously I turn streams on and off. Many thanks!

MartyG-RealSense, 2020-06-05:
Great news, I'm very pleased for you   :)    Is it okay to close the case now if you are satisfied with the outcome?   Thanks!

7oud, 2020-06-06:
Yes, the latest firmware with v2.35 SDK works normally. Thanks!

closed 2020-06-06 by 7oud

jorgemia, 2020-06-14:
Had the same issue of `No Frames Received` from the rgb camera with a D435i (5.12.3) on librealsense 2.33.1 on a Xavier NX with Jetpack 4.4. 

I’ve updated the camera to 5.12.5 and it seems to work fine now. I see a white screen for a second when restarting the RGB stream and then the picture appears. Don’t know what’s causing this but don’t remember it happening with a regular nano. 

Also the `control transfer returned error` message still appears at the start when launching the depth stream.

MartyG-RealSense, 2020-06-14:
@jorgemia  I don't know if you are using ROS, but the opinion of Doronhi the RealSense ROS developer has been that if the **control_transfer returned** warning is only appearing at the start and is not generating continuously then it is not something to be concerned about.  He said:

"The warning message "control_transfer returned error" reflects communication error with the device that is resolved by repeating the attempt. These messages are still under investigation but as long as they appear only now and then, say at the beginning of the run, they can safely be ignored for now".

"If the message "messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11" is continuous, that is appears multiple times per second, then it is not good. It suggest a real communication problem between the code and the device and can have effects".

jorgemia, 2020-06-14:
Yes already knew about it, thanks @MartyG-RealSense !!