realsenseai/librealsense #6413: Enable / disable RGB stream while running

pmirabel, 2020-05-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415  | 
| Firmware Version                   | 5.11.01.100 | 
| Operating System & Version |   Linux (debian 9) | 
| SDK Version                            |  2.32.1                          | 
| Language                            |  C                         | 

### Issue Description

Hello, I want to enable / disable RGB stream after camera initialization, eg "while running" through ROS wrapper so I answered to an issue [on the  ROS related repo](https://github.com/IntelRealSense/realsense-ros/issues/1195).
As it seems to be more an SDK question, I'm now here to ask you : 
Is there any way to achieve that ? 

The quickiest way to explain my goal would be " I want to mimic the realsense-viewer comportment when I click on the toggle button _RGB Camera "on / OFF"._

MartyG-RealSense, 2020-05-18:
I think comportment means 'behaviour'.  So you want to create an on / off toggle for the RGB stream in your program that behaves like the RGB on / off toggle control in options side-panel of the RealSense Viewer, please?

If you want the RGB stream to stop completely (not pause) and then start again, I would think that you would just use something like **pipeline.start()** to start the RGB stream and **pipeline.stop()** to stop it.

In the link below is an example script for allowing depth and color streams to be started and stopped independently during runtime.

https://github.com/IntelRealSense/librealsense/issues/1984

dorodnic, 2020-05-19:
If you want to mimic what viewer is doing exactly, please check out sensor-control example. 
Pipeline should also work, as long as two processes do not try streaming from same sensor at the same time. This is also more problematic when using LIBUVC backend

MartyG-RealSense, 2020-05-30:
@pmirabel  Do you require further assistance please, or can this case be closed?   Thanks!

pmirabel, 2020-06-02:
Hi @MartyG-RealSense  & thank you. I will continue this discussion on the realsense-ros related issue, as it has to deal specifically with this implementation (and still have to see if it's really usefull or not).

closed 2020-06-02 by pmirabel

MartyG-RealSense, 2020-06-02:
Thanks so much for the update @pmirabel