realsenseai/librealsense #6414: issues using D435 with Dell Precision 5540 vendor kernel

sinback, 2020-05-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (4.15.0-1081-oem (provided by Dell, for the Precision 5540 laptop))                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2 }                          | 
| Language                            |  not applicable - using realsense utilities                          | 
| Segment			|  robot                          | 

### Issue Description
I'm not sure if this camera is supposed to work with a vendor kernel, but I have a whole bunch of issues using the Realsense with this kernel... I am loath to switch to a vanilla kernel because I'll lose vendor-provided hardware support for the rest of the machine if I do that, but I'll give it a shot if support with vendor kernels isn't supposed to exist.

realsense-viewer reports every connection to the camera as USB2.1, not USB3, regardless of cable type (I am sure I have used a cable which has USB-C 3.0 support - there is a Thunderbolt 3, USB3.1-supporting Intel USB-C controller in the laptop).

I saw rs-color work a couple times, but it usually reports "Frame didn't arrive within 15000".

rs-depth doesn't work - sometimes says, "The connected device doesn't support depth streaming!", sometimes just also gets no frame within 15 seconds. A couple times I have seen it just spit garbage out on stdout, which might mean it's getting frames? it looks like a bunch of random ASCII letters and whitespace.

After running rs-color or rs-depth and not getting a frame, the camera usually stops being recognized and I have to plug it in again.

Sometimes I get permission denied on its video devices (`/dev/video1, /dev/video2, /dev/video3` - there's a `/dev/video0` that I think is the builtin webcam) but sometimes I don't. I'm using the udev rule which comes with the debian package, which has no diff with the one checked into the repo right now. I'm in the plugdev group and the video devices are owned by root/plugdev group.

Secure boot is disabled.

Last time I did an apt upgrade I saw that dpkg built the dkms module against my Dell OEM kernel so it's certainly trying.

I don't think I could use a kernel module built against a different kernel but if somebody gave me one built against a vanilla 4.15.0-1081 kernel I could give it a shot, I think if you change the version string the kernel module is supposed to run with it sometimes will run even though it's not supposed to.

MartyG-RealSense, 2020-05-18:
Hi @sinback   If you build Librealsense with the 'backend' installation method then you should hopefully be able to sidestep the kernel situation entirely.  This method - which requires an internet connection - does not depend on Linux versions, kernel versions of patches.

If you would like to use this installation method, please go to the Build folder (Librealsense > Build) and run the CMake build instruction below:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

sinback, 2020-05-20:
@MartyG-RealSense Thanks for letting me know about the libusb backend! It sounds like a smart move for this project in terms of supporting more hardware in the future. :)

I built and installed it with the exact flags you suggested and had no issues building it, but when I run `rs-enumerate-devices` I now get:

```
20/05 13:30:01,293 ERROR [140360282916608] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
Could not create device - failed to set power state . Check SDK logs for details
```

I am not sure where to find the SDK logs but I'm happy to send those along too if you know where to find them

sinback, 2020-05-20:
I reinstalled librealsense2-dkms to see if that helped with the uvcvideo problems and it did get rid of the messages in dmesg about unknown video format, but it didn't get rid of the message about unable to create debugfs directory, and it didn't change the behavior of rs-enumerate-devices either. With librealsense2-dkms installed dmesg looks like

```
[ 2399.590108] usb 1-1: new high-speed USB device number 56 using xhci_hcd
[ 2399.746847] usb 1-1: New USB device found, idVendor=8086, idProduct=0b07
[ 2399.746848] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2399.746850] usb 1-1: Product: Intel(R) RealSense(TM) Depth Camera 435 
[ 2399.746851] usb 1-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
[ 2399.751085] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[ 2399.755962] uvcvideo: Unable to create debugfs 1-56 directory.
[ 2399.756003] uvcvideo 1-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 2399.756004] uvcvideo 1-1:1.0: Entity type for entity Processing 2 was not initialized!
[ 2399.756005] uvcvideo 1-1:1.0: Entity type for entity Camera 1 was not initialized!
[ 2399.756054] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/input/input66
[ 2399.756513] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[ 2399.759067] uvcvideo: Unable to create debugfs 1-56 directory.
[ 2399.759148] uvcvideo 1-1:1.3: Entity type for entity Processing 7 was not initialized!
[ 2399.759149] uvcvideo 1-1:1.3: Entity type for entity Extension 8 was not initialized!
[ 2399.759150] uvcvideo 1-1:1.3: Entity type for entity Camera 6 was not initialized!
```

MartyG-RealSense, 2020-05-20:
Does reapplying the udev rules help please?

https://github.com/IntelRealSense/librealsense/issues/3263

sinback, 2020-05-20:
No, it didn't help, sadly.

I borrowed my roommate's desktop and this particular camera's rgb stream acts much happier with that but the depth stream still seems to be totally broken. We tried both Ubuntu 18.04's debian release and using Windows 10. Especially because of the corruption on the output of rs-depth I saw before (and still now), now I am thinking it may have a hardware issue with the camera and we are both wasting our time :( but as far as I know, it's never really been used before except when I tried to get it to work this week, so I wonder how I could have broken it.

I think I can pick up a different unit from a coworker, how about we close this issue and I reopen it if I still have issues with that one?

MartyG-RealSense, 2020-05-20:
It is very, very rare for a camera to have a hardware fault from the factory that renders it unusable no matter what is tried, but it can happen.

If you have access to the RealSense Viewer program then you could try going to the More option at the top of the options side-panel and then selecting 'On Chip Calibration' and going through the on-screen process to see if it improves the broken depth image.

sinback, 2020-05-20:
I tried doing that just now, on the desktop, which does act a little better than my laptop, I think. I got a couple depth frames out of it but then it froze and the calibration progress bar in the top right stopped progressing. I waited for a few minutes without it progressing before giving up. I also tried the Very Fast calibration option but it didn't make any progress either.

I will let you know how the other camera performs.

sinback, 2020-05-21:
Hey, an update - today I switched from NVIDIA graphics card graphics on my laptop to Intel integrated graphics, and switched from the libusb backend back to the apt repository backend, and saw a massive improvement in the D435. Both streams work fine now  It still thinks it's a USB 2 device and works super poorly with my USB-C controller, but it works alright with the normal USB 3 controller. It works alright with my python librealsense application too :) I tried a D415 as well and it wasn't working great either when I used the other GPU but did fine when I switched.

Based on this, I am fairly confident this was some kind of a power issue all along. I am not sure why it was also misbehaving with my desktop, which has never had USB power issues even with a bunch of peripherals, but I'm ok with closing this issue.

MartyG-RealSense, 2020-05-21:
I'm glad you found a workable solution!  Thanks for the permission to close, and good luck with your work   :)

closed 2020-05-21 by MartyG-RealSense