realsenseai/librealsense #6418: kinfu constant reset

Inffzy99, 2020-05-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version | Win 10 | 
| GPU | 2080ti | 
| Platform                                 | PC |
| SDK Version                            | 2.0 | 
| Language                            | C++ | 
| Segment			| AR | 

### Issue Description
kinfu fails to capture pointclouds due to `if (!kf->update(frame))` on line 287 in rs-kinfu.cpp always staying false.

[image]
[image]

dorodnic, 2020-05-19:
Hi @Inffzy99 
Sorry, kinfu algorithm and implementation is somewhat outside our control. The code in this example can be a starting point for feeding data from RealSense into Kinfu algorithm.
I find [open3D](http://www.open3d.org/docs/release/tutorial/ReconstructionSystem/capture_your_own_dataset.html) to be better alternative for 3D reconstruction. You should also consider [infinitam](http://www.robots.ox.ac.uk/~victor/infinitam/) as a more modern approach compared to Kinfu, but I have not tested it myself.

RealSenseSupport, 2020-05-26:
Hi @Inffzy99
Do you need further support on this issue?
If we don't hear from you in 7 days, this issue will be closed.

closed 2020-06-03 by RealSenseSupport