realsenseai/librealsense #6419: Jetson TX2::RGB camera doesn't work in RealSense Viewer

wookiekim, 2020-05-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D450i| 
| Firmware Version                   | 05.12.03.00  | 
| Operating System & Version | Ubuntu 16.04.6 LTS   | 
| Kernel Version (Linux Only)    | 4.4.38-tegra                                         | 
| Platform                                 | NVIDIA Jetson TX2  |
| SDK Version                            |  2.24.0 ~ 2.28.1                         | 
| Language                            |  C++/C   | 
| Segment			|  others                          | 

### Issue Description

I'm trying to utilize the data (depth, colour...) obtained through the D450i camera for research purposes.

I have followed the instructions [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) given in the docs for librealsense installation, **except that ** the fourth step didn't work with the following message:

> 404  Not Found [IP: 91.189.88.152 80], E:Some index files failed to download. They have been ignored, or old ones used instead.

However, skipping the fourth step did not pose any problems with the fifth step, where I could install `librealsense2-utils` and `librealsense2-dev`.

Initiating `realsense-viewer` works fine, but whenever I try to stream colour by toggling **RGB camera**, it throws the following error message:

> terminate called after throwing an instance of 'std::runtime_error' 

what() : something illegible

Just to check if only the realsense viewer was the problem, and other colour-related functions work fine, I checked with `rs-capture`, which terminates with the following message:

> RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x981ba0):
    Frame didn't arrived within 15000

### What I have tried
* Searching through the docs, issues for anything similar
* Installation of higher version of cmake
* Removing and reinstalling, which introduced dependency errors and now I'm using aptitude to install the `librealsense2-*` packages without much trouble.
  * I can only install SDK 2.24.0 ~ 2.28.1 

###  What I haven't tried yet
* Installing older/newer versions of librealsense
  * Should I build from source? how? I've tried this once, but it has always introduced new errors for me.
* Doing something with cuda
  * I've seen that RGB works on Jetson from SDK 2.13+ **with CUDA**, how do I check if I have the correct configurations?

Thank you for your support. Any help will be appreciated :)

dorodnic, 2020-05-19:
Hi @wookiekim 
Is the camera connected directly to the TX2 board or via a hub? 
I have not encountered this type of problem, and as far as I can tell you are doing everything correctly
@ev-mp - maybe something related to permissions for the video device, what do you think?
@wookiekim - you write `what() : (something illegible)`, do you have the full error message?

wookiekim, 2020-05-19:
Thanks for the reply!
The camera is connected via a hub, so that I can use a set of wireless keyboard and mouse as well.

I have written `something illegible`, because it is just shown as a short, dense set of squares. 
I have tried downgrading the firmware to the 'recommended firmware' as suggested by realsenseviewer, but the problem remains.

wookiekim, 2020-05-20:
A direct copy and paste of the error message is as follows:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  �
```

wookiekim, 2020-05-20:
The same problem arises when using a new D435 camera (Not D435i)

wookiekim, 2020-05-21:
Using gdb yields the following error messages upon termination:

```
Thread 33 "realsense-viewe" received signal SIGABRT, Aborted.
[Switching to Thread 0x7faafff1a0 (LWP 9256)]
0x0000007fb65b2528 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
```

Any help would be appreciated.

RealSenseSupport, 2020-05-26:
Hi @wookiekim ,

Is USB hub that you use unpowered? Can you please run rs-enumerate-devices and attach the output to your comment?

Thank you!

wookiekim, 2020-05-27:
@RealSenseSupport the USB hub is directly connected to the Jetson TX2, and the realsense camera to the board.

Below is the output of rs-enumerate-devices:

```
Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	935322070552
    Firmware Version              : 	05.11.11.100
    Recommended Firmware Version  : 	05.11.11.100
    Physical Port                 : 	
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	001423025783

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 2	  1280x800	@ 30Hz	   Y8
    Infrared 1	  1280x800	@ 30Hz	   Y8
    Infrared 2	  1280x800	@ 25Hz	   Y16
    Infrared 1	  1280x800	@ 25Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y8
    Infrared 2	  1280x800	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared 2	  1280x720	@ 15Hz	   Y8
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 6Hz	   Y8
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 2	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared 2	  848x480	@ 60Hz	   Y8
    Infrared 2	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 15Hz	   Y8
    Infrared 2	  848x480	@ 15Hz	   Y8
    Infrared 2	  848x480	@ 6Hz	   Y8
    Infrared 1	  848x480	@ 6Hz	   Y8
    Infrared 2	  848x100	@ 100Hz	   Y8
    Infrared 1	  848x100	@ 100Hz	   Y8
    Infrared 1	  640x480	@ 90Hz	   Y8
    Infrared 2	  640x480	@ 90Hz	   Y8
    Infrared 2	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 2	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 2	  640x480	@ 15Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 2	  640x480	@ 6Hz	   Y8
    Infrared 2	  640x400	@ 25Hz	   Y16
    Infrared 1	  640x400	@ 25Hz	   Y16
    Infrared 1	  640x400	@ 15Hz	   Y16
    Infrared 2	  640x400	@ 15Hz	   Y16
    Infrared 2	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 60Hz	   Y8
    Infrared 2	  640x360	@ 60Hz	   Y8
    Infrared 1	  640x360	@ 30Hz	   Y8
    Infrared 2	  640x360	@ 30Hz	   Y8
    Infrared 1	  640x360	@ 15Hz	   Y8
    Infrared 2	  640x360	@ 15Hz	   Y8
[... 223 lines cut ...]
    Color	  320x240	@ 6Hz	   RGB8
    Color	  320x240	@ 6Hz	   Y16
    Color	  320x240	@ 6Hz	   BGRA8
    Color	  320x240	@ 6Hz	   RGBA8
    Color	  320x240	@ 6Hz	   BGR8
    Color	  320x240	@ 6Hz	   YUYV
    Color	  320x180	@ 60Hz	   RGB8
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 6Hz	   RGB8
    Color	  320x180	@ 6Hz	   Y16
    Color	  320x180	@ 6Hz	   BGRA8
    Color	  320x180	@ 6Hz	   RGBA8
    Color	  320x180	@ 6Hz	   BGR8
    Color	  320x180	@ 6Hz	   YUYV
```

RealSenseSupport, 2020-06-03:
@wookiekim ,
Thank you for providing the rs-enumerate-devices output.
We still don't have exact details about USB Hub. What kind of USB hub do you use : unpowered or powered? It's also not clear from your answers

> The camera is connected via a hub, so that I can use a set of wireless keyboard and mouse as well. 

> the USB hub is directly connected to the Jetson TX2, and the realsense camera to the board

how camera is connected : via USB hub or directly to the board?
Please don't edit already published comments, but rather provide us with the requested information by posting a new comment.

Thank you!

RealSenseSupport, 2020-06-08:
@wookiekim ,

Do you still need help with this question?

wookiekim, 2020-06-08:
@RealSenseSupport  Sorry for the late reply. 
The camera is connected via a USB hub, NOT directly to the board. 
Sorry for the confusion. 
I'm not sure how to distinguish between powered and non-powered hubs, but since my hub is only connected to the JetsonTX2 board, I guess my hub is unpowered. 

Thank you!

wookiekim, 2020-06-08:
@RealSenseSupport Tried with a powered hub, but the same problem arises :(

RealSenseSupport, 2020-06-16:
@wookiekim ,

At that stage, we cannot rule out this possibility. Can you please try to install from scratch L4T 32.2.3 on your board?
Please elaborate on what do you mean by saying "I can only install SDK 2.24.0 ~ 2.28.1".
Regardless of the version, if you still have a problem with step 4 of the installation process, please 
check your network connection. I have checked that step and there is no problem
to add the server, mentioned in installation guide, to list of repositories.

wookiekim, 2020-06-17:
> Please elaborate on what do you mean by saying "I can only install SDK 2.24.0 ~ 2.28.1".

I was referring to the installable version of `librealsense2-util` and `libtrealsense2-dev`.

I do not think I have a problem with the installation process. At the moment, I am obliged to use L4T28.1.0, so I've just enabled the depth and one of the IR streams only to make it work. It's just the RGB (and the IR camera near the RGB) that doesn't work. 

If it's a matter of possibilities, I guess I'll cope with just the depth and IR streams for now. Thank you for your help!

closed 2020-06-17 by wookiekim